home-assistant/operating-system #2307: 9.4 version doesn't work on odroid-xu4 sdcard

hzphreak, 2023-01-17:
### Describe the issue you are experiencing

Writing the 9.4 image to the SDCARD and booting it on odroid-xu4 the screen throws errors "ext4 with invalid superblock checksum" and the system enters in recovery mode with a minimal prompt. Using version 9.3 solves the problem.

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. BalenaEcher with https://github.com/home-assistant/operating-system/releases/download/9.4/haos_odroid-xu4-9.4.img.xz
2. Boot SDCARD in odroid-xu4
3. Errors on screen (ext4 with invalid superblock checksum and docker daemon doesn't start)

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

agners, 2023-01-19:
With 9.4 only a stable kernel update came in which could affect this, it seems a bit unlikely that this causes ext4 issues.

Are you sure that the SD card write was successful when you tried with 9.4? Can you maybe retry (maybe with a different SD card even?)

agners, 2023-01-25:
@hzphreak can you check [Home Assistant OS 9.5](https://github.com/home-assistant/operating-system/releases/tag/9.5).