home-assistant/operating-system #2318: Trying to update to 9.5, error 404

tjorim, 2023-01-24:
### Describe the issue you are experiencing

Saw the release of 9.5 as beta and tried to update via command line.
Got error response `Error raised from OTA Webserver: 404`.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Go to the terminal and type `ha os update --version 9.5`.

### Anything in the Supervisor logs that might be useful for us?

```txt
23-01-24 21:24:11 INFO (MainThread) [supervisor.api.middleware.security] /os/update access from a0d7b954_ssh
23-01-24 21:24:11 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/9.5/haos_yellow-9.5.raucb
23-01-24 21:24:11 ERROR (MainThread) [supervisor.os.manager] Error raised from OTA Webserver: 404
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

### System information

## System Information

version | core-2023.1.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.76-v8
arch | aarch64
timezone | Europe/Brussels
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4996
Installed Version | 1.30.0
Stage | running
Available Repositories | 1197
Downloaded Repositories | 54
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.4
-- | --
update_channel | beta
supervisor_version | supervisor-2023.01.1
agent_version | 1.4.1
docker_version | 20.10.19
disk_total | 458.4 GB
disk_used | 22.8 GB
healthy | true
supported | true
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | AirCast (3.5.1), Spotify Connect (0.12.2), SSH & Web Terminal (13.0.2), File editor (5.5.0), Node-RED (14.0.1), Studio Code Server (5.5.2), Zigbee2MQTT (1.29.2-1), Mosquitto broker (6.1.3), Home Assistant Google Drive Backup (0.110.1), MariaDB (2.5.1), phpMyAdmin (0.8.3), JupyterLab (0.9.3), Matrix (0.10.0), The Lounge (0.17.0), Glances (0.17.2), Tailscale (0.7.0), ZeroTier One (0.16.0), AdGuard Home (4.8.0), Nginx Proxy Manager (0.12.3), Uptime Kuma (0.5.2), WireGuard (0.7.1), CUPS Print Server (2.0.2), ESPHome (2022.12.6)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 29
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 11, 2022 at 7:18 AM
-- | --
current_recorder_run | January 23, 2023 at 8:45 PM
estimated_db_size | 5040.02 MiB
database_engine | mysql
database_version | 10.6.8

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

Went digging and found failed release action: https://github.com/home-assistant/operating-system/actions/runs/3999068543

agners, 2023-01-24:
The 9.5 release is currently still building: https://github.com/home-assistant/operating-system/actions/runs/3999068543

closed 2023-01-24 by agners

tjorim, 2023-01-24:
I see it was restarted, will give it a while.
\

agners, 2023-01-24:
Yeah GitHub or my runner had a hiccup, not sure. Did run through now.