home-assistant/operating-system #2322: when selecting the edit with YAML option on any page of lovlance it does not present the code as it should

Dazzlly, 2023-01-27:
### Describe the issue you are experiencing

when selecting the edit with YAML option on any page of lovlance it does not present the code as it should
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.go to edit panel from any lovlance view page or control panel as you prefer to call it
2.select the pencil (edit button) on the page you want to edit
3.when selecting the edit with YAML option in the 3 stacked squares in the upper right corner, it does not present the code as it should
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING[0m
[32m23-01-27 11:49:47 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m23-01-27 11:49:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository[0m
[32m23-01-27 11:49:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/brenner-tobias/ha-addons repository[0m
[32m23-01-27 11:49:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository[0m
[32m23-01-27 11:49:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/CoolKit-Technologies/ha-addon.git repository[0m
[32m23-01-27 11:49:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository[0m
[32m23-01-27 11:50:00 INFO (MainThread) [supervisor.store] Loading add-ons from store: 70 all - 0 new - 0 remove[0m
[32m23-01-27 11:50:00 INFO (MainThread) [supervisor.store] Loading add-ons from store: 70 all - 0 new - 0 remove[0m
[32m23-01-27 11:50:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-01-27 12:16:35 INFO (MainThread) [supervisor.host.info] Updating local host information[0m
[32m23-01-27 12:16:36 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m23-01-27 12:16:36 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m23-01-27 12:16:36 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information[0m
[32m23-01-27 12:16:36 INFO (MainThread) [supervisor.host.manager] Host information reload completed[0m
[32m23-01-27 12:20:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-01-27 12:49:40 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON[0m
[32m23-01-27 12:49:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER[0m
[32m23-01-27 12:49:48 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-01-27 12:49:48 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING[0m
[32m23-01-27 12:49:48 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-01-27 12:49:48 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING[0m
[32m23-01-27 12:49:48 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m23-01-27 12:50:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-01-27 13:20:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[... 221 lines cut ...]
estimated_db_size | 25.50 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Sonoff</summary>

version | 3.3.1 (b20e33c)
-- | --
cloud_online | 3 / 4
local_online | 2 / 2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

[Bug_Lovlance-YAML.zip](https://github.com/home-assistant/operating-system/files/10521347/Bug_Lovlance-YAML.zip)

Dazzlly, 2023-01-27:
in the additional information I added a windows step recorder zip file in case you want to see how to reproduce the bug in steps

agners, 2023-01-27:
I don't think that the pencil should show the YAML. To show the YAML there is the tripple dot menu "Raw configuration editor".

In any case, this is not a Home Assitant OS issue. If anything, this belongs to the [frontend repository](https://github.com/home-assistant/frontend/).

closed 2023-01-27 by agners

Dazzlly, 2023-01-27:
> Não acho que o lápis deva mostrar o YAML. Para mostrar o YAML, existe o menu de ponto triplo "Editor de configuração bruta".
> 
> Em qualquer caso, este não é um problema do Home Assitant OS. De qualquer forma, isso pertence ao [repositório frontend](https://github.com/home-assistant/frontend/) .

Thanks, I already made the issue in the correct place
https://github.com/home-assistant/frontend/issues/15237

Dazzlly, 2023-01-28:
apparently after installing the latest update of
Operating System Host Home Assistant OS 9.5
made today, the problem has been fixed, here is a screenshot with the demonstration

[image]

if it was a Home Assistant OS update that fixed it... then it wasn't a FrontEnd thing?