home-assistant/operating-system #2323: 9.5 shows up even though it's in beta

tangix, 2023-01-28:
### Describe the issue you are experiencing

I haven't signed up for beta-testing but 9.5 still shows up in Home Assistant 2023.1.7.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Check the update notification in HA

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

ludeeus, 2023-01-28:
9.5 is declared as stable <https://version.home-assistant.io/stable.json>.
<https://github.com/home-assistant/version/pull/270>

tangix, 2023-01-28:
Maybe update the Release Notes?
[image]

bcutter, 2023-01-29:
> Maybe update the Release Notes?
> [image]
> 

Exactly. Heard about a folk on the forums having issues after the latest HA OS update (likely a false positive). Was wondering why a stable one has a beta note - everything a bit confusing even it is just a text. I dont't install beta releases on my productive systems.

Well, at least people are reading release notes before pressing the update button - thats a good thing right :-)

GaTechThomas, 2023-01-29:
This means that every time a Home Assistant update comes out, then users will have to go search for whether it's stable or not. As a system matures, users must be able to trust the system more and more. If this sort of avoidable inconsistency is not resolved then I will stop being a paid user, and I suspect that I'm not the only one.

agners, 2023-01-29:
The ultimate source of truth is the [version repository](https://github.com/home-assistant/version/). I've started adding the hint in the release notes out of courtesy, because I got the reverse complaints: "HAOS 9.x is out, why don't I see my update".

> If this sort of avoidable inconsistency is not resolved then I will stop being a paid user, and I suspect that I'm not the only one.

Yeah, automated way to remove the hint in the release notes would be nice. PR welcome.

closed 2023-01-29 by agners

bcutter, 2023-01-29:
I completely don't share the "hammer method" mentioned before as this is open source and minor mistakes (like this one) just happen - I am so happy to have experts like @agners  contributing to this great project.

Usually (seen many times) as a quick fix simply a new release will be provided (without a beta note) without any content changes - overriding the former one in the update section.

This gives confidence to all unsure users not having updated yet and only adds little extra work (download/bandwidth, another install/reboot) for all users having already installed the former one. 

So a 9.6 which is a rebranded 9.5. Just a thought.