home-assistant/operating-system #2324: 9.5  Home assistant Blue (Odroid. Samba is working, but no access with a browser or companion app...

jeanbart82, 2023-01-28:
### Describe the issue you are experiencing

Can't access... not sure how to roll back as I only have Samba access now.
Rebooting is not working... (unplugging power etc.)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.just upgraded from 9.4 to 9.5 OS
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
not accessibly I think...
```

### Anything in the Host logs that might be useful for us?

```txt
nothing?
```

wermerb, 2023-01-28:
RPI4 cannot startup after this release

agners, 2023-01-29:
Can you ping the device (e.g. `ping homeassistant.local`)? If so, can you access the observer (http://homeassistant.local:4357/)?

dan-drews, 2023-01-29:
Good morning! I'm experiencing this as well. I can ping my instance.

I can also access my observer and I receive the following:

Supervisor: | Connected
-- | --
Supported: | Unsupported
Healthy: | Unhealthy

dan-drews, 2023-01-29:
I wish I could provide more useful steps for how I got it working, but I did solve my problem. Hopefully we have some similarities in our setup that could point you in the right direction.

First, some background that might be applicable to you. I use AdGuard within home assistant, and in the past I had issues where home assistant would use my router for DNS, then my router would point to AdGuard by default, leading to issues where my home assistant OS couldn't reach the internet after reboots. I ended up solving this within my Unifi router by setting up a different network config for home assistant so that it would reach out to 1.1.1.1 by default.

I don't know exactly if it's related to the 9.5 update, but I do see this in the release notes:
> Use same class B network for Docker as Supervisor (https://github.com/home-assistant/operating-system/pull/2246) (https://github.com/home-assistant/operating-system/pull/2259) @agners

Anyway, somehow my Home Assistant server was unassigned from the specific network I setup in my unifi router. So when I managed to SSH in (No idea why it would let me SSH in sporadically) I wasn't able to nslookup any websites, but I could ping IP addresses. This made me check my unifi setup and discover that the network for that port was unassigned. So I updated the network, rebooted and everything works fine.

jeanbart82, 2023-01-30:
I ended up spending the whole evening flashing my blue again and restoring a snapshot. At this point it's running again with 9.4 I don't dare to do an update. 

**I have Adguard installed.  So you say I should try change my (TP-link X60) router back to default dns settings like 1.1.1.1 and do the update again?** 

I might try it again on a different day when I have some more time for restoring. so much is depending now on my blue. Maybe setting up an back up on the NAS or something.

I didn't expect it to do these things anymore...

agners, 2023-01-31:
I have two ODROID-N2+ on OS 9.5, so there isn't a generic problem with ODROID-N2+ and OS 9.5. Since Adguard sometimes causes this weird circular dependencies it could be the culprit, but hard to tell at this point. I'd suggest to take a full backup (and download it) before trying to update again, then you should be able to restore quite quickly in case something fails.

dnedeltchev, 2023-02-02:
Did anyone get to the bottom of this issue? I'm also on an N2+, OS 9.4 and AdGuard. However my N2+ is configured with a static IP and pointing directly to the router for DNS so no dependency on AdGuard. But I'm still scared to try and upgrade.
😁

jeanbart82, 2023-02-03:
I was trying to update again. Now mad sure I had a way of SSH into HA... So I could possible use 
`ha host update --version 9.4` to roll back... but even that did not help... and now even in 9.4 after restarts... I am not getting back in. 
Not sure how to stop AdGuard trough SSH... or maybe even NGINX proxy manager... (not sure if that is doing something...)

feel a bit out of options... not sure where to look.

dnedeltchev, 2023-02-03:
Is your AdGuard your DHCP as well? And is your HA on a static IP?

jeanbart82, 2023-02-03:
No DHCP is done at the router.

At least I managed to roll back a backup with SSH. 
`ha backups restore c1a07617 --password 'my secret password'` or `ha backups restore [slug] --password 'my secret password'`

So I might disable an add-on like adguard first before trying to update again.

jeanbart82, 2023-02-03:
OK. It seems to be fixed for now. Rolling back with SSH to 9.4 didn't fix it. I just got the same problem. Then I tried to restore the backup. That managed to get back in with the browser... see my last post....
So I wanted to stop the add-on for AdGuard and NGINX proxy manager... but the add-ons where acting strange... they seemed to be stopped before I could stop it, but the logs said they were running... stopping did not work. So I just uninstalled them both. 

I did the update to 9.5 again... had to try to access some variations with the local ip adres again... http:// did not work, but the https:// with some warnings got it running again... everything seemed to be fine and normal again with the Add-ons... SO Installed NGINX proxy manager again... did the setup and everything was fine. adguard. same thing. 

Not sure what was wrong... could be add-ons for adguard, nginx or maybe I missed something with the https//

anyways. this can be closed... maybe to much legacy issues... who knows...

closed 2023-02-03 by jeanbart82