home-assistant/operating-system #2325: Upgrade to 9.5 breaks Homekit?

dannycab, 2023-01-28:
### Describe the issue you are experiencing

I recently upgraded to 9.5 and every single device and entity that HomeKit was making available is now indicating 'No Response.' To be clear everything in HA is working fine, automations, controls, etc. It is simply that everything seems to have become completely disconnected from the HomeKit controller.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgraded OS
2. Devices become unavailable in iOS 16.3 across the board

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/7f3debb3/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/59763550/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/bd9ec4c2/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/ccb8a787/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/06d3d90a/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/d50e75c3/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/686b61c3/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/f43ec75b/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/c8f17784/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/96b57b1b/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/7f8c72ca/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/0089d8fb/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/df0cfebd/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/3dc42c77/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/ba15eac0/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/047649af/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/9201b755/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/3637eac7/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/7e0624c3/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/2b402e4d/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/87470f07/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-01-28 13:47:36 INFO (MainThread) [supervisor.api.middleware.security] /backups/92424c03/info access from cebe7a76_hassio_google_drive_backup[0m
[31m23-01-28 13:47:38 ERROR (MainThread) [supervisor.services.modules.mqtt] There is already a MQTT service in use from core_mosquitto[0m
[32m23-01-28 13:47:38 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on ghcr.io/esphome/esphome-hassio with version 2022.12.8[0m
[32m23-01-28 13:47:38 INFO (MainThread) [supervisor.addons.addon] Restore config for addon core_configurator[0m
[32m23-01-28 13:47:38 INFO (MainThread) [supervisor.addons.addon] Restoring data for addon core_configurator[0m
[32m23-01-28 13:47:38 INFO (SyncWorker_0) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-configurator with version 5.5.0[0m
[32m23-01-28 13:47:38 INFO (MainThread) [supervisor.backups.manager] Restore 96b57b1b wait for Home-Assistant[0m
[32m23-01-28 13:47:38 INFO (SyncWorker_2) [supervisor.docker.interface] Starting homeassistant[0m
[32m23-01-28 13:47:38 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready[0m
[32m23-01-28 13:47:41 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.FREEZE[0m
[32m23-01-28 13:47:41 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-01-28 13:47:42 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json[0m
[32m23-01-28 13:47:42 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository[0m
[32m23-01-28 13:47:42 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository[0m
[32m23-01-28 13:47:42 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository[0m
[32m23-01-28 13:47:42 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository[0m
[32m23-01-28 13:47:42 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository[0m
[32m23-01-28 13:47:42 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository[0m
[32m23-01-28 13:47:43 INFO (MainThread) [supervisor.store] Loading add-ons from store: 73 all - 0 new - 0 remove[0m
[32m23-01-28 13:47:43 INFO (MainThread) [supervisor.store] Loading add-ons from store: 73 all - 0 new - 0 remove[0m
[31m23-01-28 13:47:43 ERROR (SyncWorker_9) [supervisor.docker.interface] Container addon_15d21743_samba_backup is not running[0m
[32m23-01-28 13:48:16 INFO (MainThread) [supervisor.backups.manager] Found 35 backup files[0m
[32m23-01-28 13:48:16 INFO (MainThread) [supervisor.backups.manager] Found 35 backup files[0m
[32m23-01-28 13:49:04 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance[0m
[32m23-01-28 13:49:04 INFO (MainThread) [supervisor.backups.manager] Full-Restore 96b57b1b done[0m
[... 136 lines cut ...]
Jan 28 18:57:16 homeassistant kernel: audit: type=1334 audit(1674932236.781:545): prog-id=139 op=LOAD
Jan 28 18:57:16 homeassistant kernel: audit: type=1334 audit(1674932236.781:546): prog-id=140 op=LOAD
Jan 28 18:57:16 homeassistant systemd[1]: Starting Hostname Service...
Jan 28 18:57:16 homeassistant systemd[1]: Started Hostname Service.
Jan 28 18:57:16 homeassistant kernel: audit: type=1334 audit(1674932236.916:547): prog-id=141 op=LOAD
Jan 28 18:57:16 homeassistant kernel: audit: type=1334 audit(1674932236.917:548): prog-id=142 op=LOAD
Jan 28 18:57:16 homeassistant kernel: audit: type=1334 audit(1674932236.917:549): prog-id=143 op=LOAD
Jan 28 18:57:16 homeassistant systemd[1]: Starting Time & Date Service...
Jan 28 18:57:17 homeassistant systemd[1]: Started Time & Date Service.
Jan 28 18:57:46 homeassistant systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Jan 28 18:57:46 homeassistant kernel: audit: type=1334 audit(1674932266.945:550): prog-id=140 op=UNLOAD
Jan 28 18:57:46 homeassistant kernel: audit: type=1334 audit(1674932266.945:551): prog-id=139 op=UNLOAD
Jan 28 18:57:46 homeassistant kernel: audit: type=1334 audit(1674932266.945:552): prog-id=138 op=UNLOAD
Jan 28 18:57:47 homeassistant systemd[1]: systemd-timedated.service: Deactivated successfully.
[... cut ...]
Jan 28 19:02:19 homeassistant kernel: audit: type=1334 audit(1674932539.509:556): prog-id=144 op=LOAD
Jan 28 19:02:19 homeassistant systemd[1]: Started Journal Gateway Service.
Jan 28 19:02:19 homeassistant systemd-journal-gatewayd[12043]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
```

### Additional information

Is it possible to downgrade the OS to test? I can't seem to find documentation to do so. I restored from last night's backup successfully, but that didn't downgrade the OS.

bcutter, 2023-01-29:
I only had one HK device integrated (testing purposes) and on every HA OS reboot (no update required) I had to manually re-add that device as it was unavailable like forever.

As I likely won't continue working with that HK ~~crap~~ stuff I nearly don't care, just wanted to say: the problem might not be the 9.5 update.

agners, 2023-01-29:
Can you try downgrading to see if that indeed resolves your problem? From the terminal, use:

```
ha os update --version 9.4
```

dannycab, 2023-01-29:
> Can you try downgrading to see if that indeed resolves your problem? From the terminal, use:
> 
> ```
> ha os update --version 9.4
> ```

I was able to do this. After executing, I restarted ha and it is prompting to upgrade to 9.5. However, this did not fix the issue with HomeKit. I am going to try rolling back core to 2023.1.5 as I think that was the last working core version.

dannycab, 2023-01-29:
> I was able to do this. After executing, I restarted ha and it is prompting to upgrade to 9.5. However, this did not fix the issue with HomeKit. I am going to try rolling back core to 2023.1.5 as I [...]

Yep, still no luck.

dannycab, 2023-01-29:
With the os at 9.4 and core at 2023.1.5, I am unable to create a working HomeKit Bridge. HA lets me create it as usual, but it fails to be added to iOS. The error my phone (iOS 16.3) produces is 'Accessory Not Found'...make sure plugged in etc...

agners, 2023-01-31:
Thanks for the updates. My conclusion here  is that it's not HAOS 9.5 related. There seems to be a problem with communicating with HomeKit devices in general. Probably community forums or Discord is the best place to discuss.

closed 2023-01-31 by agners