home-assistant/operating-system #2327: HA9.5 wifi issues

pan-sattan, 2023-01-29:
### Describe the issue you are experiencing

On my new fresh installation of HA I cam see, that the system cannot recognize any wifi networks which has channel bandwidth 20MHz. Only homeassistant is not capable to recognize it. I tried different systemes (piOS, kali) and it works well. Please check this.
My rpi4 is already updated (eeprom, wlan, boot manager). 

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

all the time

### Anything in the Supervisor logs that might be useful for us?

```txt
none
```

### Anything in the Host logs that might be useful for us?

```txt
none
```

agners, 2023-01-29:
Hm, I am using 20MHz wide channels on my 802.11ac network, and i can browse it. Are you sure it is related to the channel bandwidth? What channel are you using?

pan-sattan, 2023-01-29:
yes, when i change it to 40mhz (channel 13) 2,4g it works.. even 5g... but not 20mhz... I am using n protokol

agners, 2023-01-31:
So you are using Channel 13 20MHz, and that doesn't work? Very curious.

pan-sattan, 2023-02-08:
so.. I find, that the problem is with channels below channel 10 on 2,4Ghz wifi network it works well. Problem was, that my router had auto channel function turned ON. But duno why only HA is not supporting all channels?

agners, 2023-02-08:
> But duno why only HA is not supporting all channels?

Currently HAOS always runs in "global domain", as in we support channels which are globally accepted (see #1530). Afaik, as soon as the device is associated with an AP, it should inherit the country settings of that AP and support all the channels. However, if the AP is at startup on a channel which is not supported by the global domain, then you won't be able to connect.

Since this seems to be related to channel and WiFi regulatory domain, I'd consider this as handled by #1530.

closed 2023-02-08 by agners