home-assistant/operating-system #2328: HDMI CEC not working on Home Assistant OS (RPi4)

istvnurbn, 2023-01-29:
### Describe the issue you are experiencing

HDMI CEC is not working in a fresh version of Home Assistant OS installed on a Raspberry Pi 4. TV is connected directly to the Raspberry Pi's HDMI0 port.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Log in via SSH
2. Run `echo scan | cec-client -s -d 1`
3. Get the error `autodetect FAILED`

### Anything in the Supervisor logs that might be useful for us?

```txt
1.

Logger: pycec.cec
Source: /usr/local/lib/python3.10/site-packages/pycec/cec.py:81
First occurred: 6:50:15 PM (8 occurrences)
Last logged: 7:04:15 PM

No adapters found

2.

Logger: pycec
Source: /usr/local/lib/python3.10/site-packages/pycec/network.py:422
First occurred: 6:50:15 PM (899 occurrences)
Last logged: 7:05:17 PM

Not initialized. Waiting for init.
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

## System Information

version | core-2023.1.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.84-v8
arch | aarch64
timezone | Europe/Budapest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4964
Installed Version | 1.30.1
Stage | running
Available Repositories | 1193
Downloaded Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.01.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 28.5 GB
disk_used | 3.5 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (13.0.2), Studio Code Server (5.5.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 29, 2023 at 4:33 PM
-- | --
current_recorder_run | January 29, 2023 at 6:50 PM
estimated_db_size | 0.36 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

wbarber69, 2023-01-29:
did you try rebooting then after reboot try restarting the pi-cec addon.  ive had a similar issue after every major update.  this usually fixes it.

istvnurbn, 2023-01-30:
Hi!

I've rebooted the RPi itself many times, powecycled my TV, restarted the add-on itself, plugged the cable to a different port on my RPi but I still have these 2 warnings in my logs:

1.

`Logger: pycec
Source: /usr/local/lib/python3.10/site-packages/pycec/network.py:422
First occurred: 10:22:47 (51 occurrences)
Last logged: 10:23:37

Not initialized. Waiting for init.`

2.

`Logger: pycec.cec
Source: /usr/local/lib/python3.10/site-packages/pycec/cec.py:81
First occurred: 10:22:47 (1 occurrences)
Last logged: 10:22:47

No adapters found`

wbarber69, 2023-01-30:
Are you trying to use cec-scanner or pi-cec?  Because built in cec support was removed from ha many versions ago.  You would need to be running the third party ad-on pi-cec in order to return cec functionality to your ha installation.  Running cec-scanner along side pi-cec will give you errors.

closed 2023-01-30 by istvnurbn

wbarber69, 2023-01-30:
Oh well it would have been nice to see what was causing your issues.  I’ve been struggling with cec ever since they deprecated built in support and had to move to pi-cec.  Pi-cec works just fine, but for some reason my ha server (rpi4) constantly tells my receiver to change inputs for no reason that I can tell other than pi-cec may be scanning the network from time to time.

Closed #2328<https://github.com/home-assistant/operating-system/issues/2328> as completed.