home-assistant/operating-system #2329: Missing WiFi Support

jamesawaite, 2023-01-29:
### Describe the issue you are experiencing

Home Assistant OS appears to be missing wifi support for on-board wireless card.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Boot

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2023.1.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.01.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 28.6 GB
disk_used | 9.8 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS UI (1.6.0), Glances (0.17.2), SSH & Web Terminal (13.0.2), InfluxDB (4.5.0), Studio Code Server (5.5.2), Grafana (8.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 20, 2023 at 5:29 PM
-- | --
current_recorder_run | January 29, 2023 at 2:08 PM
estimated_db_size | 260.59 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

Gigabyte GB-BXBT-2807 spec sheet lists AZUREWAVE AW-NB159H 802.11 b/g/n Wi-Fi / Bluetooth 4.0

Bluetooth seems to work
[    1.475799] usb 1-2: New USB device found, idVendor=13d3, idProduct=3410, bcdDevice= 2.00
[    1.478823] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.481810] usb 1-2: Product: Bluetooth Radio
[    1.484776] usb 1-2: Manufacturer: Realtek
[    1.487725] usb 1-2: SerialNumber: 00e04c000001

agners, 2023-01-30:
Hm, the WiFi part seems to be hooked up through PCIe according to http://en.techinfodepot.shoutwiki.com/wiki/AzureWave_AW-NB159H. It should be supported by https://linux-hardware.org/?id=pci:10ec-b723-1a3b-2a59. I'll enable the `CONFIG_RTL8723BE` driver for generic-x86-64.

AvocadoPanic, 2023-04-30:
This was resolved with the inclusion of the driver. :)

closed 2023-04-30 by AvocadoPanic