home-assistant/operating-system #2330: HomeAssistant not loading in VirtualBox

TwisterMc, 2023-01-29:
### Describe the issue you are experiencing

I updated VirtualBox today and now HomeAssistant won't load. It's been working fine for months so I'm not sure what's going on. It's stuck on a screen that says "Booting Slot A" and there is nothing I can do.

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Latest

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install latest version of VirtualBox on Mac.
2. Boot HomeAssistant

:shrug

### Anything in the Supervisor logs that might be useful for us?

```txt
00:00:03.250131 VMEmt: Halt method global1 (5) not available in driverless mode, using method1 (4) instead
00:00:03.250283 Changing the VM state from 'CREATING' to 'CREATED'
00:00:03.257737 Changing the VM state from 'CREATED' to 'POWERING_ON'
00:00:03.257960 Changing the VM state from 'POWERING_ON' to 'RUNNING'
00:00:03.257992 Console: Machine state changed to 'Running'
00:00:03.264664 EFI: debug point SEC_PREMEM
00:00:03.282592 GUI: UIMachineViewNormal::resendSizeHint: Restoring guest size-hint for screen 0 to 800x600
00:00:03.287819 GUI: HID LEDs sync is enabled
00:00:03.318540 GUI: UIMachineLogicNormal::sltCheckForRequestedVisualStateType: Requested-state=1, Machine-state=6
00:00:03.321464 ERROR [COM]: aRC=VBOX_E_VM_ERROR (0x80bb0003) aIID={4680b2de-8690-11e9-b83d-5719e53cf1de} aComponent={DisplayWrap} aText={Could not take a screenshot (VERR_NOT_SUPPORTED)}, preserve=false aResultDetail=-37
00:00:03.409250 ERROR [COM]: aRC=VBOX_E_VM_ERROR (0x80bb0003) aIID={4680b2de-8690-11e9-b83d-5719e53cf1de} aComponent={DisplayWrap} aText={Could not take a screenshot (VERR_NOT_SUPPORTED)}, preserve=false aResultDetail=-37
00:00:03.579241 EFI: VBoxDbg> loadimage64 'SecMain.efi' 0xfffcc094 LB 0x0
00:00:03.581599 EFI: VBoxDbg> loadimage64 'PeiCore.efi' 0x2020120 LB 0x0
00:00:03.584308 EFI: VBoxDbg> loadimage64 'PcdPeim.efi' 0x2026fe0 LB 0x2580
00:00:03.586565 EFI: VBoxDbg> loadimage64 'ReportStatusCodeRouterPei.efi' 0x2029620 LB 0xa00
00:00:03.587262 EFI: VBoxDbg> loadimage64 'StatusCodeHandlerPei.efi' 0x202a0a0 LB 0x700
00:00:03.587837 EFI: VBoxDbg> loadimage64 'PlatformPei.efi' 0x202a8a0 LB 0x5480
00:00:03.662326 EFI: debug point SEC_POSTMEM
00:00:03.663409 EFI: VBoxDbg> loadimage64 'PeiCore.efi' 0x7f364000 LB 0x6cc0
00:00:03.665953 EFI: VBoxDbg> loadimage64 'DxeIpl.efi' 0x7f361000 LB 0x2160
00:00:03.668119 EFI: VBoxDbg> loadimage64 'S3Resume2Pei.efi' 0x7f35f000 LB 0x1680
00:00:03.669141 EFI: VBoxDbg> loadimage64 'CpuMpPei.efi' 0x7f354000 LB 0xa8c0
00:00:03.713096 EFI: VBoxDbg> loadimage64 'TpmMmioSevDecryptPei.efi' 0x7f342000 LB 0xf00
00:00:03.713769 EFI: VBoxDbg> loadimage64 'Tcg2ConfigPei.efi' 0x7f340000 LB 0x1220
00:00:03.714641 EFI: VBoxDbg> loadimage64 'TcgPei.efi' 0x7f33d000 LB 0x26c0
00:00:03.715590 EFI: VBoxDbg> loadimage64 'Tcg2Pei.efi' 0x7f334000 LB 0x8520
00:00:03.717005 EFI: VBoxDbg> loadimage64 'DxeCore.efi' 0x7f318000 LB 0x1b100
00:00:03.720441 EFI: debug point DXE_CORE
00:00:03.723329 EFI: VBoxDbg> loadimage64 'DxeCore.efi' 0x7f318000 LB 0x0
00:00:03.747218 EFI: VBoxDbg> loadimage64 'DevicePathDxe.efi' 0x7e920000 LB 0x9e00
00:00:03.747857 EFI: VBoxDbg> loadimage64 'PcdDxe.efi' 0x7e92d000 LB 0x2f00
00:00:03.748623 EFI: VBoxDbg> loadimage64 'FvbServicesRuntimeDxe.efi' 0x7eeec000 LB 0x2100
00:00:03.752051 EFI: VBoxDbg> loadimage64 'ReportStatusCodeRouterRuntimeDxe.efi' 0x7eeea000 LB 0x1060
00:00:03.752774 EFI: VBoxDbg> loadimage64 'RuntimeDxe.efi' 0x7eee8000 LB 0x13c0
00:00:03.759055 EFI: VBoxDbg> loadimage64 'SecurityStubDxe.efi' 0x7e818000 LB 0x83940
00:00:03.759730 EFI: VBoxDbg> loadimage64 'EbcDxe.efi' 0x7e916000 LB 0x4120
00:00:03.760545 EFI: VBoxDbg> loadimage64 'Legacy8259.efi' 0x7e91d000 LB 0xb00
00:00:03.761804 EFI: VBoxDbg> loadimage64 'CpuIo2Dxe.efi' 0x7e91b000 LB 0xb00
00:00:03.762368 EFI: VBoxDbg> loadimage64 'CpuDxe.efi' 0x7e8fa000 LB 0xdd20
00:00:03.776988 EFI: debug point DXE_AP
00:00:03.779340 EFI: VBoxDbg> loadimage64 'IncompatiblePciDeviceSupportDxe.efi' 0x7e909000 LB 0x840
00:00:03.780036 EFI: VBoxDbg> loadimage64 'PciHotPlugInitDxe.efi' 0x7e8e2000 LB 0x2120
[... 83 lines cut ...]
00:00:03.918959 EFI: VBoxDbg> loadimage64 'IScsiDxe.efi' 0x7df6d000 LB 0x1a1a0
00:00:03.922276 EFI: VBoxDbg> loadimage64 'VirtioNetDxe.efi' 0x7dfd9000 LB 0x3240
00:00:03.922921 EFI: VBoxDbg> loadimage64 'E1kNetDxe.efi' 0x7e007000 LB 0x2ce0
00:00:03.923588 EFI: VBoxDbg> loadimage64 'UhciDxe.efi' 0x7dfbe000 LB 0x4220
00:00:03.924260 EFI: VBoxDbg> loadimage64 'EhciDxe.efi' 0x7dfb9000 LB 0x4cc0
00:00:03.924933 EFI: VBoxDbg> loadimage64 'XhciDxe.efi' 0x7df9b000 LB 0x76e0
00:00:03.925622 EFI: VBoxDbg> loadimage64 'UsbBusDxe.efi' 0x7dfb4000 LB 0x4d40
00:00:03.926302 EFI: VBoxDbg> loadimage64 'UsbKbDxe.efi' 0x7dfd1000 LB 0x3e00
00:00:03.927058 EFI: VBoxDbg> loadimage64 'UsbMassStorageDxe.efi' 0x7df97000 LB 0x3100
00:00:04.368077 ERROR [COM]: aRC=VBOX_E_VM_ERROR (0x80bb0003) aIID={4680b2de-8690-11e9-b83d-5719e53cf1de} aComponent={DisplayWrap} aText={Could not take a screenshot (VERR_NOT_SUPPORTED)}, preserve=false aResultDetail=-37
00:00:04.626700 Display::i_handleDisplayResize: uScreenId=0 pvVRAM=00007fcccb906000 w=1024 h=768 bpp=32 cbLine=0x1000 flags=0x0 origin=0,0
00:00:04.836072 PS2K: Selected scan set 2
00:00:04.935367 ATA: LUN#0: EXECUTE DEVICE DIAGNOSTIC, status 50
00:00:04.938787 ATA: LUN#0: INITIALIZE DEVICE PARAMETERS: 63 sectors per track, 16 heads
00:00:04.940575 ATA: LUN#2: EXECUTE DEVICE DIAGNOSTIC, status 00
00:00:04.964313 GUI: UIMediumEnumerator: Medium-enumeration finished!
00:00:05.384704 EFI: VBoxDbg> loadimage64 '.efi' 0x7d804000 LB 0x9a000
00:00:06.601149 EFI: debug point DXE_AP
```

### Anything in the Host logs that might be useful for us?

```txt
The only logs I know of are above.
```

agners, 2023-01-30:
It seems you are using a rather old machine configuration with IDE disk drive emulation. Newer setups as per [installation guide](https://www.home-assistant.io/installation/windows/) should be using SCSI or SATA. Can you try a setup from scratch and compare your storage configuration to that of a new one?

TwisterMc, 2023-01-31:
I ran though https://www.home-assistant.io/installation/macos/ and there's no change with the old image.

The new image gives me
[image]

asinla, 2023-02-07:
Did it get sorted? I was about to update my VirtualBox on my Mac.

TwisterMc, 2023-02-08:
No fix yet. I did rollback VirtualBox with no success.

Turns out most of my stuff was TP Link / Kasa so I just bought [TPL Bridge](https://www.brooker.mobi/tpl-bridge/) and that works like a charm. I'll re-do home assistant some day.

walshtj, 2023-02-08:
I had the same issue.  I updated Virtualbox from 6.1 to 7 and it 'broke' the VDI.
Long story short - I don't know what happened.  

I quickly gave up trying to fix it as I needed to get the house working again. So I downgraded to Virtualbox 6.1, reinstalled HA in a VM and used a (nightly) backup.

However, after that I started playing around again on a different (same spec) laptop. I created another HA instance and I've upgraded from VirtualBox from 6.1 to 7.  Then back to 6.1 then back to 7 - all without any problems. The reason I was doing this was to try to get Bluetooth to work with HA in a VM on a Mac (no success).

In summary, I think I was just unlucky the first time and something odd happened with the VDI which corrupted it

asinla, 2023-02-08:
Yeah I've had mixed results too. 
I bought the Sonoff Zigbee 3.0 dongle and set everything up with Z2M eventually but it took very specific steps in VirtualBox for the dongle to show up. Once it did though it's been pretty stable. 
When you were trying to get Bluetooth working was it with a USB Bluetooth adapter or Mac built in? I believe there is no way to get Mac built in to work with VM's. I was going to try an external adapter.

My main issue right now is figuring out the networking stuff through Virtualbox. Can't cast anything to Google devices and my Sonos cannot reach it's port!

> I had the same issue. I updated Virtualbox from 6.1 to 7 and it 'broke' the VDI. Long story short - I don't know what happened.
> 
> I quickly gave up trying to fix it as I needed to get the house working again. So I downgraded to Virtualbox 6.1, reinstalled HA in a VM and used a (nightly) backup.
> 
> However, after that I started playing around again on a different (same spec) laptop. I created another HA instance and I've upgraded from VirtualBox from 6.1 to 7. Then back to 6.1 then back to 7 - all without any problems. The reason I was doing this was to try to get Bluetooth to work with HA in a VM on a Mac (no success).
> 
> In summary, I think I was just unlucky the first time and something odd happened with the VDI which corrupted it
> 
> EDIT: I find USB devices on Virtualbox on the Mac a total lottery in terms of getting them to work. On my test machine I've a spare Conbee II USB which VirtualBox 6.1 can find and use but Virtualbox 7 can't (some sharing violation). On my 'main' HA VM instance I don't use USB stick. Instead I have the original Raspberry Pi 3 running all the 'Deconz' stuff including the Conbee II stick. Ideally I'd like to integrate all the parts together on to a single old MacBook but its not easy.

walshtj, 2023-02-08:
I bought one of those new SkyConnect USBs to test also (it's en route). However ultimately the issue is USB on VirtualBox on a Mac and I don't know if its Apple or Oracle's fault. If I ever got it all working I'd be afraid to update the Mac OS or the Virtualbox.

Regarding the Bluetooth I tried internal and a cheap external one (TFUFR Bluetooth 5.0 Dongle Adapter on Amazon.co.uk) - neither worked and after 2 days of following tutorials I just gave up again. If you get it working with a different Bluetooth adapter please tell us here.

closed 2023-05-22 by github-actions[bot]

sk31337, 2025-03-13:
> I ran though https://www.home-assistant.io/installation/macos/ and there's no change with the old image.
> 
> The new image gives me [image]

i have the exact same boot screen. Anyone knows how to fix this?

thegrimester, 2025-05-13:
I also had this problem. I was installing HA on a N150 mini PC using VirtualBox running on Ubuntu. I followed the Linux instructions and downloaded the Linux vdi file. 

This did not work and the supervisor was crashing and I could not get to HA via port 8123.

The solution was to download the **Intel** vdi file from the windows installation instructions page. The files look identical and are the same size but there is obviously some difference for intel hardware vs Pi. 

if you are running in intel hardware give this a go.

Probably the instructions should be changed on HAs website