home-assistant/operating-system #2332: After upgrading vom OS 9.4 to 9.4 I see "[WARN] Home Assistant CLI not starting! Jump into emergency console"

klacol, 2023-01-30:
### Describe the issue you are experiencing

I have updates the OS on my Rasberry PI4 from 9.4. to 9.5 via the web interface. After that the CLI does not start.  I have checked the following:
* I can connect a keyboard and run commands
* The LAN-Interface has an IP and can be pinged
* I see the message "[WARN] Home Assistant CLI not starting! Jump into emergency console" in the console
* I have restartet the pi, same issue
* I have disconnected the LAN and reconnected, same issue

Has anybody an advice for me?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I can not read the Log; since the web interface is not accesable.
```

### Anything in the Host logs that might be useful for us?

```txt
I can not read the Log; since the web interface is not accesable.
```

klacol, 2023-02-01:
I can not access the web frontend. I have used ``docker ps -a`` and i see two containers. One is started and one is not started:

* ``homeassistant`` is not started
* ``hassio_obeserver`` is started at port 80/tcp

When I try to start the container ``homeassistant``, I get this error:
``error creating overlay mount to mnt/data/overlay2/a_long_guid/merged``

markwmuller, 2023-02-01:
Not sure related, but I've had something very similar happening. I updated through the web interface, and then the system stopped being responsive (I forget the exact details). I unplugged the Raspberry Pi 4, and I think the SSD I was running fromhad failed. I tried to reflash OS 9.5 onto two different SD cards (4GB, and 16GB), and couldn't get things working (each time, it just booted into the emergency mode). 

I even different Raspberry Pis (both 4). Each time, same behaviour (also tried 32 bit OS, instead of 64). 

I seem to have gotten it working now by flashing onto an old USB HDD (512GB). Could it be that there's a minimum size requirement on the storage?

agners, 2023-02-06:
> When I try to start the container `homeassistant`, I get this error:
> `error creating overlay mount to mnt/data/overlay2/a_long_guid/merged`

This sounds like a corruption of some kind. What type of boot media are you using? If it is an SD card, I'd consider replacing it.

closed 2023-02-06 by klacol