home-assistant/operating-system #2338: eth0 does not work out of the box on Raspberry Pi 400 - device has no carrier

tomranttom, 2023-02-01:
### Describe the issue you are experiencing

On Home Assistant 9.4 and 9.5 ethernet port, eth0, does not work out of the box with Raspberri Pi 400. Main issue seems to be `Error: Can’t update config on eth0: Connection ‘Supervisor eth0’ is not available on device eth0 because device has no carrier`.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Get the latest, in this case 9.5, version of Home Assistant and flash it to SD-card with Balena Etcher
2. Start RPi 400 with the, now finished, SD card.
3. Wait for RPi to start. After 1m 30s you'll get an error `(FAILED] Failed to start Wait Until Kernel Time Synchronized.`
4. From this point forward device is not detected within your router nor can it be connected to from another device.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't acces system logs as there's no way to connect to RPi, even with local connection from device A to RPi.
```

### Anything in the Host logs that might be useful for us?

```txt
Can't acces host logs as there's no way to connect to RPi, even with local connection from device A to RPi.
```

### Additional information

Issue has been narrowed down to be an issue with Home Assistant as the port works just fine on Raspbian and the issue goes away when using USB-A to Ethernet adapter.

`% ping homeassistant.local
ping: cannot resolve homeassistant.local: Unknown host`

Logging in from cli and checking connection prints the following.

`% login
% nmcli con show
% nmcli
eth0 unavailable…`

tomranttom, 2023-02-05:
A little bit more information from Home Assistant web-UI.

```
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:124
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 5:34:18 PM (1 occurrences)
Last logged: 5:34:18 PM

Failed to to call /network/interface/eth0/update - Can't create config and activate eth0: Connection 'Supervisor eth0' is not available on device eth0 because device has no carrier
```

agners, 2023-02-06:
Missing carrier typically means a low level issue (cable?). Did you try different cables or connecting to a different switch?

> On Home Assistant 9.4 and 9.5 ethernet port

Does that mean this worked in releases prior to 9.4 or did you just test those two releases?

tomranttom, 2023-02-06:
Hi,

Tried different cables and none of them worked on eth0. One of the same
cables worked just fine with USB-Ethernet adapter.

Tried this on both HA 9.4 and 9.5. and the connection did not work on
either of those. Don’t know about releases prior to those.

DLKH6538, 2023-02-28:
this issue happened to me as well.

greenstone02, 2023-02-28:
Hi everybody,
I have the same issue on RPI3B+ with haos_rpi3-64-9.5 and haos_rpi3-64-9.4.

The first time the problem happens was during upgrade from 9.4 to 9.5. The upgrade finished with the problem described above after the reboot.

Then I try to install it fresh via haos images witten with dd to SD-card and SSD.

The SD-Cards with image 9.4 and 9.5 (two different SD-card which were used before with different software) were not be able to start. In gparted I get the error, that the disk parameter containing an error. I checked both: 1. no repair via gparted => only black screen; 2. repair via gparted => only black screen.

Then I used a SSD which was succesfully used before updrade haos. Some problem after starting gparted: disk containg an error. 1. no repair via gparted => only black screen; 2. repair via gparted => boot possible, but then I run in the error described above.

I changed the boot parameter for the fat32 partition from msftres to boot,esp

luddement, 2023-03-10:
Same problem here with a new Raspberry pi 400. Can't get anywhere with eth0 ethernet. Wifi works, and everything is updated over it.

From `supervisor logs`:

`[supervisor.host.network] Can´t create config and activate eth0: Connection ´Supervisor eth0´is not available on device eth0 because device has no carrier`

mdbruinnl, 2023-04-03:
same here raspbarry pi 3b, upgrade from 9.4 to 9.5 did not complete succesfully on eth0, after disconnecting the cable and continuing in wifi the upgrade finished, but eth0 is not working anymore with:
[supervisor.host.network] Can´t create config and activate eth0: Connection ´Supervisor eth0´is not available on device eth0 because device has no carrier

working on wifi for now

MarchosiasM, 2023-04-17:
This is an incredibly frustrating issue to not have fixed by now.

agners, 2023-04-18:
Can you all check Home Assistant OS 10?

I am testing Raspberry Pi 3 and 4 regularly, and just tested Raspberry Pi 400 with Home Assistant OS 10. Ethernet works just fine here. So this must be some combination of Ethernet switch and/or software missconfiguration.

tomranttom, 2023-04-30:
Unfortunately updating to Home Assistant 10 and to the latest core the issue still persists.

tomranttom, 2023-04-30:
Commented from phone and accidentally closed this. So no, this is not fixed.

tomranttom, 2023-04-30:
Adding some more information. My network-setup consists of Unifi Dream Machine and after that there's Unifi Switch. I've also tried to connect to HA from directly UDM, as in skipping the swich, still no dice. I'll add this information to bug description as well.

Connection works just fine Raspbian.

canonone99, 2023-06-15:
Just came across this issue while attempting to install HA. I have 2 RPI400, one was purchased around 18 months ago, one earlier this year. 
The older RPI 400 installs HA successfully using Ethernet port. 
The newer model cannot access the Ethernet port with the message as described above.
Same switch cables etc. Used HAOS 10.2 and core 2023.6.1
Both PI’s ports were working perfectly using latest raspberry pi os.

Steph1116, 2023-06-26:
Has anyone found a working solution to this issue? I too had to go in and install HA via wifi to get my system up and running on the rpi400. Is the issue on home assistant end or the rpi400? Do I go out and buy a rpi4 and just run my HA on that or is this going to get sorted?

benjaminrau, 2023-06-26:
Same issue here. Homeassistant was connected via eth0 until today. Stopped working suddenly in Mid of May. (been away some time). Same error:
Can't create config and activate eth0: Connection 'Supervisor eth0' is not available on device eth0 because device has no carrier

Wifi connects well. Updated until most recent version now but issue is still there.

Using a Rasperry PI 4

LukasBeckers, 2023-08-06:
Got the same issue with Raspberry pi 4 and Homeassistant os 10.3

Pro-Control, 2023-09-06:
Same issue here
Rasperry PI 4

dalleyd1, 2023-09-08:
I have the same issue on a recently purchased pi4.  Would be great to see this fixed.

minul08, 2023-09-15:
I have the same issue! I have a pi4!

cntzr, 2023-09-29:
Same for me on OS 10.5. Running on PI4 and got it absolutely unexpected. No changes or updates.

tomranttom, 2023-12-31:
Last of update of they year, quite literally, from me. Tried again to boot the latest Home Assistant, 2023.12, from USB and so far the issue has not been fixed. I still could still not get ip to the RPi400. My USB to Ethernet adapter also stopped working at one point on HA and as such, natively booting to HA is not an option to me anymore on RPi400. I do wonder if there was a minor change of what ethernet chips were used at one point, that would at the very least confirm why @canonone99 is having issues with a newer device but not with the older one. And why also, seemingly, certain RPi4s are having the issue.

I've since moved to installing HA to a docker container as ethernet does work under Raspbian. Do note though if you go with this route that you lose Supervised unless you install HA with other means. I can, however, still test this every now and then as long as I have the RPi400, if need be. Well, at least until I move to the HA Green.

FinemechanicPub, 2024-01-11:
> Just for the record, I got this error when I try to configure ethernet settings without the actual ethernet cable plugged into my NUC. Perhaps some people here are having cable/physical issues, worth a check maybe.

I find this ridiculous. Why can't I change Ethernet IP address and other settings while the cable is disconnected? I would consider it a bug.

My problem is that I can't get IP properly configured after I moved from one router to another. I have managed to connect by WiFi, but now when I attach an Ethernet cable the device gets the same IP assigned as with WiFi and everything stops working. And now I can't change IP without plugging the cable and can't connect when it is plugged in. :facepalm:

tomranttom, 2024-01-14:
Some more information from latest Home Assistant from boot.

```
[OK] Reached target System Time Set.
[FAILED] Failed to start Wait Until Kernel Time Synchronized.
See 'systemctl status systemd-time-wait-sync.service' for details
```

Output from aforementioned systemctl.

```
x systemd-time-wait-sync.service - Wait Until Kernel Time Synchronized
Loaded: loaded (/usr/lib/systemd/systen/systemd-time-wait-sync.service: enabled; preset: enabled)
Drop-In: /etc/systemd/system/systemd-time-wait-sync.service.d
network-online.conf, timeout.conf
Active: failed (Result: timeout) since Tue 2023-04-04 11:17:01 UTC; 2min 25s ago
Docs: man:systemd-time-wait-sync.service(8)
Process: 524 ExecStart=/usr/lib/systemd/systemd-time-wait-sync (code=exited, status=0/SUCCESS)
Main PID: 524 (code=exited, status=0/SUCCESS)
CPU: 16ms

Apr 01 10:55:33 honeassistant systemd[1]: Starting Wait Until Kernel Time Synchronized...
Apr 04 10:55:33 homeassistant sustend-time-wait-sync[524]: Failed to add a watch for /run/systemd/timesync: No such file or directory
Apr 04 10:55:33 homeassistant systemd-time-vait-sync[524]: adjtime state 5 status 40 time Tue 2023-04-04 10:55:33.594891 UTC
Apr 01 11:15:31 homeassistant systemd-time-vait-sync[524]: adjtime state 5 status 40 time Tue 2023-04-04 11:15:31.728256 UIC
Apr 01 11:17:01 homeassistant systemd[1]: systemd-time-wait-sync.service: start operation timed out. Terminating.
Apr 01 11:17:01 homeassistant systemd-time-wait-syncl529): Exit without adjtimex synchronized.
Apr 04 11:17:01 homeassistant systemd[1]: systemd-time-wait-sync.service: Failed with result 'timeout'.
Apr 01 11:17:01 honeassistant systemd[1]: Failed to start Wait Until Kernel Time Synchronized.
```

outofsight, 2024-02-05:
I'm facing now the same issue on Pi4 with HA OS 10.5.
Suddenly no more eth connection. 
IPv4 disabled and can't be enabled.
Any solution?

giankaleo, 2024-02-05:
stesso identico problema su nuc intel, con adatattore usb/lan nesusn problema

reodel, 2024-05-13:
I face the same issue, just installed the latest software version haos_rpi4-64-12.3.img on a brand new Rpi4 64 that i purchased directly from the link provided in the Instructions by Home Assistant. Very frustrating.

closed 2024-08-21 by github-actions[bot]

AlcubierreFTL, 2025-02-02:
Raspberry 5 fresh HA installation and just noticed this.

My router shows an Ethernet connection and IP address yet Home Assistant shows this No Carrier error when trying to enable it.

Have checked the cable although I will try a different cable tomorrow.

Did anyone have any luck?