home-assistant/operating-system #2347: HAOS compile failed with git dubious ownership

minecraft2048, 2023-02-09:
### Describe the issue you are experiencing

I try to follow the instruction at https://developers.home-assistant.io/docs/operating-system/getting-started, and it failed to build containerd with `fatal: detected dubious ownership in repository at '/build'`

This failed with both podman on my Arch linux machine and docker version 20.10.17 on AWS EC2 Ubuntu 
The last output is this:
```
>>> containerd 1.6.8 Building
cd /build/output/build/containerd-1.6.8; GO111MODULE=on GOFLAGS=-mod=vendor GOROOT="/build/output/host/lib/go" GOPATH="/build/output/host/usr/share/go-path" GOCACHE="/build/output/host/usr/share/go-cache" GOMODCACHE="/build/output/host/usr/share/go-path/pkg/mod" GOPROXY=off PATH="/build/output/host/bin:/build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" GOBIN= CGO_ENABLED=1 GOOS="linux" GOARCH=arm64 CC="/build/output/host/bin/aarch64-buildroot-linux-gnu-gcc" CXX="/build/output/host/bin/aarch64-buildroot-linux-gnu-g++" CGO_CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O2 -g0 -D_FORTIFY_SOURCE=1" CGO_CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O2 -g0 -D_FORTIFY_SOURCE=1" CGO_LDFLAGS="" GOTOOLDIR="/build/output/host/lib/go/pkg/tool/linux_arm64"  /build/output/host/bin/go build -v -ldflags "-X github.com/containerd/containerd/version.Version=1.6.8" -modcacherw -tags "apparmor no_btrfs" -trimpath -p 33 -o /build/output/build/containerd-1.6.8/bin/containerd github.com/containerd/containerd/cmd/containerd
# cd /build; git status --porcelain
fatal: detected dubious ownership in repository at '/build'
To add an exception for this directory, call:

        git config --global --add safe.directory /build
error obtaining VCS status: exit status 128
        Use -buildvcs=false to disable VCS stamping.
make[1]: *** [package/pkg-generic.mk:293: /build/output/build/containerd-1.6.8/.stamp_built] Error 1
make[1]: Leaving directory '/build/buildroot'
make: *** [Makefile:39: rpi4_64] Error 2
```

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

dev

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Follow the instruction at https://developers.home-assistant.io/docs/operating-system/getting-started#prepare-development-environment to clone HAOS
2. Follow the instruction at https://developers.home-assistant.io/docs/operating-system/getting-started#build-images-using-build-container
3. Let it run for a while
4. Build failed with `fatal: detected dubious ownership in repository at '/build'`
...

### Anything in the Supervisor logs that might be useful for us?

```txt
There's no supervisor log, as I can't even compile the OS
```

### Anything in the Host logs that might be useful for us?

```txt
There's no host log, as I can't even compile the OS
```

agners, 2023-02-09:
Hm, I haven't come across this so far. It seems that the problem appears when Buildroot builds `containerd`. Since this is running inside the container, the build environment really should be the same :thinking:  I am using Docker 20.10.23 on ArchLinux.

minecraft2048, 2023-02-09:
Is it possible that your docker image is using old git binary from cached layers? There is a CVE for dubious ownership: https://nvd.nist.gov/vuln/detail/cve-2022-24765 and it seems like Debian backported the git security update that fixes that CVE to bullseye

There's this: https://www.kenmuse.com/blog/avoiding-dubious-ownership-in-dev-containers/ that we can try

baflo, 2023-02-13:
I got the same error

agners, 2023-02-13:
> Is it possible that your docker image is using old git binary from cached layers?

Good guess, and I was already certain that should be it. But: I've removed the image, and made sure git got reinstalled. From the logs I can see that the image gets rebuilt and git gets reinstalled. However, I am still not able to reproduce. Even after blowing away my whole build and build cache.

```
>>> containerd 1.6.8 Building                                                                                                                                                                                                                                                                                                 
cd /build/output_ova/build/containerd-1.6.8; GO111MODULE=on GOFLAGS=-mod=vendor GOROOT="/build/output_ova/host/lib/go" GOPATH="/build/output_ova/host/usr/share/go-path" GOCACHE="/build/output_ova/host/usr/share/go-cache" GOMODCACHE="/build/output_ova/host/usr/share/go-path/pkg/mod" GOPROXY=off PATH="/build/output_ova/host/bin:/build/output_ova/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" GOBIN= CGO_ENABLED=1 GOOS="linux" GOARCH=amd64 CC="/build/output_ova/host/bin/x86_64-buildroot-linux-gnu-gcc" CXX="/build/output_ova/host/bin/x86_64-buildroot-linux-gnu-g++" CGO_CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O2 -g0 -D_FORTIFY_SOURCE=1" CGO_CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O2 -g0 -D_FORTIFY_SOURCE=1" CGO_LDFLAGS="" GOTOOLDIR="/build/output_ova/host/lib/go/pkg/tool/linux_amd64"  /build/output_ova/host/bin/go build -v -ldflags "-X github.com/containerd/containerd/version.Version=1.6.8" -modcacherw -tags "apparmor no_btrfs" -trimpath -p 33 -o /build/output_ova/build/containerd-1.6.8/bin/containerd github.com/containerd/containerd/cmd/containerd
cd /build/output_ova/build/containerd-1.6.8; GO111MODULE=on GOFLAGS=-mod=vendor GOROOT="/build/output_ova/host/lib/go" GOPATH="/build/output_ova/host/usr/share/go-path" GOCACHE="/build/output_ova/host/usr/share/go-cache" GOMODCACHE="/build/output_ova/host/usr/share/go-path/pkg/mod" GOPROXY=off PATH="/build/output_ova/host/bin:/build/output_ova/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" GOBIN= CGO_ENABLED=1 GOOS="linux" GOARCH=amd64 CC="/build/output_ova/host/bin/x86_64-buildroot-linux-gnu-gcc" CXX="/build/output_ova/host/bin/x86_64-buildroot-linux-gnu-g++" CGO_CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O2 -g0 -D_FORTIFY_SOURCE=1" CGO_CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O2 -g0 -D_FORTIFY_SOURCE=1" CGO_LDFLAGS="" GOTOOLDIR="/build/output_ova/host/lib/go/pkg/tool/linux_amd64"  /build/output_ova/host/bin/go build -v -ldflags "-X github.com/containerd/containerd/version.Version=1.6.8" -modcacherw -tags "apparmor no_btrfs" -trimpath -p 33 -o /build/output_ova/build/containerd-1.6.8/bin/containerd-shim github.com/containerd/containerd/cmd/containerd-shim
...
```

Maybe something got updated again? Seems the [last update to the git package](https://tracker.debian.org/pkg/git) is two weeks ago already. I got `Setting up git (1:2.30.2-1+deb11u1)` ...

> RUN git config --global --add safe.directory /build

This works as a work around, but ideally the unterlaying git call from the wrong directory should be addressed.

millin, 2023-02-13:
Running `scripts/enter.sh make ...` without `sudo` solved the problem for me

agners, 2023-02-13:
@millin thanks, that hint allowed me to reproduce this problem: With `sudo` I too see the error reported by @minecraft2048.

The script runs the whole build within the container as root if it has been started with user id 0. That leads to the git error message.

It seems that `git` gets started by the go build process to include version information. This can be disabled by using `-buildvcs=false` (see https://tip.golang.org/doc/go1.18#go-version). Since buildroot separates the version control system from the build process this flag probably should be set globally when building go packages.

agners, 2023-02-14:
Fixed by #2347

closed 2023-02-14 by agners