home-assistant/operating-system #2349: Google Coral USB not detected after system reboot

kyle4269, 2023-02-13:
### Describe the issue you are experiencing

The coral works as intended until I do a system reboot. After the reboot it's no longer detected and requires me to unplug it and plug it back in.  

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. plug coral in
2. reboot the system
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2023-02-13 00:49:10.606917518  [2023-02-12 19:49:10] frigate.detectors.plugins.edgetpu_tfl INFO    : Attempting to load TPU as usb
2023-02-13 00:49:10.612901704  [2023-02-12 19:49:10] frigate.app                    INFO    : Capture process started for FrontYard: 234
2023-02-13 00:49:10.623864294  [2023-02-12 19:49:10] frigate.app                    INFO    : Capture process started for Driveway: 242
2023-02-13 00:49:10.633343294  [2023-02-12 19:49:10] frigate.detectors.plugins.edgetpu_tfl ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet, you must configure CPU detectors.
2023-02-13 00:49:10.635057288  Process detector:coral:
2023-02-13 00:49:10.638031366  [2023-02-12 19:49:10] frigate.app                    INFO    : Capture process started for SideGarage: 245
2023-02-13 00:49:10.643715536  Traceback (most recent call last):
2023-02-13 00:49:10.643722186    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
2023-02-13 00:49:10.643723710      delegate = Delegate(library, options)
2023-02-13 00:49:10.643725576    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
2023-02-13 00:49:10.643731769      raise ValueError(capture.message)
2023-02-13 00:49:10.643747730  ValueError
2023-02-13 00:49:10.643749113  
2023-02-13 00:49:10.643751117  During handling of the above exception, another exception occurred:
2023-02-13 00:49:10.643752929  
2023-02-13 00:49:10.643754236  Traceback (most recent call last):
2023-02-13 00:49:10.643756006    File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
2023-02-13 00:49:10.643757331      self.run()
2023-02-13 00:49:10.643758971    File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
2023-02-13 00:49:10.643868546      self._target(*self._args, **self._kwargs)
2023-02-13 00:49:10.643870956    File "/opt/frigate/frigate/object_detection.py", line 98, in run_detector
2023-02-13 00:49:10.643873421      object_detector = LocalObjectDetector(detector_config=detector_config)
2023-02-13 00:49:10.643875046    File "/opt/frigate/frigate/object_detection.py", line 52, in __init__
2023-02-13 00:49:10.643887787      self.detect_api = create_detector(detector_config)
2023-02-13 00:49:10.643890071    File "/opt/frigate/frigate/detectors/__init__.py", line 24, in create_detector
2023-02-13 00:49:10.643891560      return api(detector_config)
2023-02-13 00:49:10.643893302    File "/opt/frigate/frigate/detectors/plugins/edgetpu_tfl.py", line 34, in __init__
2023-02-13 00:49:10.643894940      edge_tpu_delegate = load_delegate("libedgetpu.so.1.0", device_config)
2023-02-13 00:49:10.643908365    File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 162, in load_delegate
2023-02-13 00:49:10.643910521      raise ValueError('Failed to load delegate from {}\n{}'.format(
2023-02-13 00:49:10.643912050  ValueError: Failed to load delegate from libedgetpu.so.1.0
2023-02-13 00:49:10.643913286  
2023-02-13 00:49:10.657515322  [2023-02-12 19:49:10] frigate.app                    INFO    : Capture process started for Backyard: 249
2023-02-13 00:49:27.788552202  172.30.32.1 - - [12/Feb/2023:19:49:27 -0500] "GET /api/stats HTTP/1.1" 200 3254 "-" "HomeAssistant/2023.2.3 aiohttp/3.8.1 Python/3.10" "-"
2023-02-13 00:49:27.796725897  172.30.32.1 - - [12/Feb/2023:19:49:27 -0500] "GET /api/version HTTP/1.1" 200 14 "-" "HomeAssistant/2023.2.3 aiohttp/3.8.1 Python/3.10" "-"
2023-02-13 00:49:27.812904697  172.30.32.1 - - [12/Feb/2023:19:49:27 -0500] "GET /api/config HTTP/1.1" 200 19019 "-" "HomeAssistant/2023.2.3 aiohttp/3.8.1 Python/3.10" "-"
2023-02-13 00:49:31.161926352  [2023-02-12 19:49:31] frigate.watchdog               INFO    : Detection appears to have stopped. Exiting Frigate...
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service nginx: stopping
2023-02-13 00:49:31.206156728  [2023-02-12 19:49:31] frigate.record                 ERROR   : Error occurred when attempting to maintain recording cache
2023-02-13 00:49:31.206164374  [2023-02-12 19:49:31] frigate.record                 ERROR   : list index out of range
Service NGINX exited with code 0 (by signal 0)
s6-rc: info: service nginx successfully stopped
s6-rc: info: service nginx-log: stopping
s6-rc: info: service frigate: stopping
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### System information

## System Information

version | core-2023.2.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4955
Installed Version | 1.30.1
Stage | running
Available Repositories | 1267
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.01.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 113.9 GB
disk_used | 8.8 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.5.0), Home Assistant Google Drive Backup (0.110.1), Lutron Certificate (0.8.0), Mosquitto broker (6.1.3), SSH & Web Terminal (13.0.2), Samba share (10.0.0), Studio Code Server (5.5.2), deCONZ (6.18.0), Z-Wave JS (0.1.75), Node-RED (14.0.2), Frigate (Full Access) Beta (0.12.0) (0.12.0-beta8)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 2
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 3, 2023 at 4:09 AM
-- | --
current_recorder_run | February 12, 2023 at 7:47 PM
estimated_db_size | 175.54 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2023-02-13:
I wonder if this is a low level USB detection issue (hardware/Linux kernel) or something on our end. Can you provide the host log after the reboot, and after unplugging/replugging?

After a reboot, does an add-on restart help (instead of phyiscally unplugging/replugging)?

closed 2023-02-13 by kyle4269

chinswain, 2023-09-24:
Interesting - I'll try swapping ports.