home-assistant/operating-system #2352: HAOS not listed in Boot menu selection

Infraviored, 2023-02-13:
### Describe the issue you are experiencing

After HAOS installation according to the guide, the system does not show up as boot option.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Intel NUC 5 I3RYH MP / Core i3.

1. Boot Ubuntu from an USB Stick, Install Balena Etcher
2. etch the Generic x86-64 image to the nvme (internal hard drive) - got a warning (care, this is your harddrive!) and proceeded.
3. reboot
4. HAOS NOT shown in boot selection
5. Boot Ubuntu, 
sudo efibootmgr --create --disk /dev/<drivename> --part 1 --label "HAOS" \
   --loader '\EFI\BOOT\bootx64.efi'
6. HAOS still not shown.

Secureboot: Disabled
UEFI Boot: Enabled.

Ubuntu USB stick shows up in the boot options.

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

agners, 2023-02-13:
```
sudo efibootmgr --create --disk /dev/ --part 1 --label "HAOS" --loader '\EFI\BOOT\bootx64.efi'
```

Using `efibootmgr` definitely should make it show up. `--disk` should be your NVMe without partition, e.g. `/dev/nvme0n1`

agners, 2023-02-13:
Wild guess: Maybe that device's BIOS does not support booting from NVMe? Does it work with other operating systems?

Infraviored, 2023-02-13:
i used

sudo efibootmgr --create --disk /dev/nvme0n1 --part 1 --label "HAOS" --loader '\EFI\BOOT\bootx64.efi'

(forgot the hard drive name here in the post)

I will check if I can boot from NVME by installing ubuntu.

agners, 2023-02-13:
> #422
> 
> apparently this is an issue that multiple people have.
> 
> In the last post, a 'solution' is presented but I do not understand it.

This is a very very old issue, where we used another bootloader and back when NVMe support wasn't complete. Today, lots of people are using HAOS on NVMe's without problems.

The most likely cause is enabled Secure Boot. Did you make sure that Secure Boot is disabled in your BIOS?

Infraviored, 2023-02-13:
> Wild guess: Maybe that device's BIOS does not support booting from NVMe? Does it work with other operating systems? Also make sure that Secure Boot is disabled in the BIOS.

That indeed was the problem!

I could also not boot ubuntu. 
The only thing left was a BIOS update. Which I did. And now I can boot Ubuntu!

I am now reinstalling HAOS, probably with success in the end. 

Thanks so much for the wild guess!

closed 2023-02-14 by agners