home-assistant/operating-system #2353: Python3 "out of memory" error on ODriod 4GB version

jlandgrave, 2023-02-13:
### Describe the issue you are experiencing

Since mid-December I have been getting an "Out of memory: Killed process xxxxx" (python3)" error about every 15 minutes on my HA server.  I keep hoping that an update will fix it but it appears to be getting worse over time.
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Reboot.  Wait 15 minutes.

### Anything in the Supervisor logs that might be useful for us?

```txt
23-02-13 12:45:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:01 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:01 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:02 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10489821 exceeds limit 4194304') is not str
23-02-13 12:45:02 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:07 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:07 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:07 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:07 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10489994 exceeds limit 4194304') is not str
23-02-13 12:45:07 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:13 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:13 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:13 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:13 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10489844 exceeds limit 4194304') is not str
23-02-13 12:45:13 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:18 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:18 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:18 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10489874 exceeds limit 4194304') is not str
23-02-13 12:45:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:24 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10490412 exceeds limit 4194304') is not str
23-02-13 12:45:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:29 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:29 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:29 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:30 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10489974 exceeds limit 4194304') is not str
23-02-13 12:45:30 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:35 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:35 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:35 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:35 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10489919 exceeds limit 4194304') is not str
23-02-13 12:45:35 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:40 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:40 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:40 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:41 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10489974 exceeds limit 4194304') is not str
23-02-13 12:45:41 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:46 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-02-13 12:45:46 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-02-13 12:45:46 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-02-13 12:45:47 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 258:WebSocketError(<WSCloseCode.MESSAGE_TOO_BIG: 1009>, 'Message size 10490235 exceeds limit 4194304') is not str
23-02-13 12:45:47 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-02-13 12:45:52 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
[... 230 lines cut ...]
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 2, 2023 at 2:54 PM
-- | --
current_recorder_run | February 13, 2023 at 12:17 PM
estimated_db_size | 4831.70 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Xiaomi MIoT</summary>

logged_in | true
-- | --
can_reach_micloud_server | ok
accounts_count | 2
account_devices_count | 1
added_devices | 1

</details>

agners, 2023-02-13:
Hm, I wonder which Python process that is. Can you run the following command in the shell to get a list of Python process and which HA part that is:

```
ha> login
# ps xfa | grep python
```

healeydave, 2023-02-25:
I'm having the same problem.
Since updating Home Assistant and HAOS to the latest versions in the last couple of weeks, I am randomly finding my HA unresponsive. Today I plugged a monitor in and found this on the console:

"Out of Memory: killed process 6942 (python3)"

My HA has been running on a Dell 3050 USSF PC for over a year, rock solid and I haven't changed anything on my system other than updating HA and Frigate.

Answer-1, 2023-03-01:
Also having a similar issue, not out of memory yet, but the hass python process is slowly eating all the memory (running core on RPI).
I think I have narrowed it down to about the time I went to python 3.10, but not 100% sure.
Restarting HA gets the memory usage to about zero, and I'm good for 2-3 weeks before it get too high to my liking.

closed 2023-06-06 by github-actions[bot]