home-assistant/operating-system #2357: Proxmox Console: Booting Slot A

Code-TT, 2023-02-15:
### Describe the issue you are experiencing

I'm using haos_ova-9.5.qcow2.xz on Proxmox on a NUC. 
If I go on Console I don't get to the HA Promt, I just get 
`Booting 'Slot A (OK=1 Try=1)'` and nothing else. No system output, no promt. The Webinterface is loading normally. 

[image]

[image]

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Boot the Proxmox VM with  haos_ova-9.5.qcow2
2. Click on Console of VM in Proxmox

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-02-15 13:45:16 INFO (MainThread) [__main__] Running Supervisor[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons[0m
[32m23-02-15 13:45:16 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 2 add-ons[0m
[33m23-02-15 13:45:16 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)[0m
[32m23-02-15 13:45:16 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning addon_core_mosquitto application[0m
[33m23-02-15 13:45:16 WARNING (SyncWorker_0) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)[0m
[32m23-02-15 13:45:17 INFO (SyncWorker_0) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mosquitto with version 6.1.3[0m
[32m23-02-15 13:45:17 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_core_dnsmasq application[0m
[32m23-02-15 13:45:18 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-dnsmasq with version 1.5.1[0m
[33m23-02-15 13:45:19 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)[0m
[32m23-02-15 13:45:22 INFO (MainThread) [supervisor.services.modules.mqtt] Set core_mosquitto as service provider for mqtt[0m
[32m23-02-15 13:45:22 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.STARTUP[0m
[32m23-02-15 13:45:22 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/allenporter/stream-addons repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/AlexxIT/hassio-addons repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/blakeblackshear/frigate-hass-addons repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/jakowenko/double-take-hassio-addons repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository[0m
[32m23-02-15 13:45:23 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 7 add-ons[0m
[32m23-02-15 13:45:24 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_core_samba application[0m
[32m23-02-15 13:45:24 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-samba with version 10.0.0[0m
[32m23-02-15 13:45:24 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-02-15 13:45:24 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_a0d7b954_vscode application[0m
[32m23-02-15 13:45:24 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/vscode/amd64 with version 5.5.2[0m
[32m23-02-15 13:45:24 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_core_duckdns application[0m
[32m23-02-15 13:45:24 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_samba[0m
[32m23-02-15 13:45:25 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-duckdns with version 1.15.0[0m
[... 229 lines cut ...]
installed_addons | Samba share (10.0.0), File editor (5.5.0), Mosquitto broker (6.1.3), Studio Code Server (5.5.2), Duck DNS (1.15.0), Dnsmasq (1.5.1), WireGuard (0.7.1), deCONZ (6.18.0), NGINX Home Assistant SSL proxy - Port 80 Version (3.0.2), InfluxDB (4.5.0), Grafana (8.1.0), Home Assistant Google Drive Backup (0.110.1), ESPHome (2022.12.8), SSH & Web Terminal (13.0.2), Frigate (0.11.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 28
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 8. Februar 2023 um 18:55
-- | --
current_recorder_run | 15. Februar 2023 um 13:45
estimated_db_size | 413.64 MiB
database_engine | sqlite
database_version | 3.38.5

</details>
```

agners, 2023-02-16:
What machine configuration are you using? Did you follow an installation guide (e.g. [this one](https://community.home-assistant.io/t/installing-home-assistant-os-using-proxmox-7/201835) from the community forum?)

Code-TT, 2023-02-17:
I used these Video: https://www.youtube.com/watch?v=1Un4zJJWUTE&t=12s

This are my settings of the VM:

```
root@pve:/etc/pve/qemu-server# cat 100.conf 
agent: 1
bios: ovmf
boot: order=scsi0
cores: 4
efidisk0: local-lvm:vm-100-disk-0,efitype=4m,pre-enrolled-keys=1,size=4M
hostpci0: 0000:00:02,x-vga=1
memory: 12288
meta: creation-qemu=7.1.0,ctime=1675952790
name: HomeAssistant
net0: virtio=0E:84:8E:2C:BF:84,bridge=vmbr0,firewall=1
numa: 0
onboot: 1
ostype: l26
scsi0: local-lvm:vm-100-disk-1,iothread=1,size=128G
scsihw: virtio-scsi-single
smbios1: uuid=b53af47b-c454-49d6-96d0-426f93ef3987
sockets: 1
usb0: host=1-4
vga: virtio
vmgenid: c1da5b78-e9bc-4a06-a387-a7d02eceef81
```

tteck, 2023-02-17:
`pre-enrolled-keys=1` is a no no

Code-TT, 2023-02-17:
I deleted pre-enrolled-keys=1 from the config, reboot, no change... 
If there are no other ideas, I will try a reinstall with the script you mentioned above next week.

Code-TT, 2023-03-14:
I have found the bug! 

I use Frigate. To use the hardware acceleration, I had to pass the graphics card (PCI Device). As soon as I do that, there's only "Booting Slot A..." in the console and nothing else. As soon as I remove the graphics card under Hardware, it works again, as it should.

I would now have to decide whether I want the shell or the hardware acceleration for Frigate. At the moment Frigate wins. But it would be nice if the problem could be fixed. :)

[image]

Code-TT, 2023-03-14:
> What machine configuration are you using? Did you follow an installation guide (e.g. [this one](https://community.home-assistant.io/t/installing-home-assistant-os-using-proxmox-7/201835) from the community forum?)

@agners: Yes, I have tried that too. It is due to the passthrough of the graphics card. 
I just wish I had seen this tutorial sooner, these scripts would have saved me a bunch of time ;)

tteck, 2023-03-14:
@Code-TT have you attempted to use the machine type "q35" instead of "i440fx"?

Code-TT, 2023-03-14:
If I change to Q35 now, the web interface is not loading...

tteck, 2023-03-14:
If changing the machine type from i440fx to q35 did not work, you can try creating a new VM with the q35 machine type instead of modifying an existing VM. Sometimes, modifying an existing VM can cause unexpected issues, and creating a new VM can provide a fresh start with the desired machine type.

Code-TT, 2023-03-14:
I have now used the above script to set everything up again with Q35. 
However, Frigate only works again with hardware acceleration when I pass the graphics card as a PCI device. But then the shell shows again "Booting 'Slot A (OK=1 TRY=1)'".

Any ideas?

Here is my new configuration:

[image]
[image]
[image]

twiti7, 2023-04-09:
I hade same issue after I update today proxmox. Same I have on new fresh install of proxmox.

closed 2023-07-15 by github-actions[bot]