home-assistant/operating-system #2366: HAOS wont boot on Dell Optiplex FX160 with Intel Atom 230 Diamondville 1.6ghz CPU

BlackRockCity, 2023-02-22:
### Describe the issue you are experiencing

I was able to install Debian and run Home Assistant supervised. However, when HAOS is installed, HAOS gets to a screen with a blinking underscore and wont boot past that even when left for 10 hours. 

Dell Optiplex FX160 with Intel Atom 230 Diamondville 1.6ghz CPU is a 64bit x86 platform with 4gb of RAM. Its a 2008 PC with an early version of the Atom processor. Running the latest BIOS version A13. There is no option to switch between UEFI and Legacy BIOS... I am assuming it is running UEFI. 

Fast boot is disabled. 
Secure boot is disabled.

This thread leads me to believe that it should run but these people are running a 5 year newer Atom CPU. The fact that I got HA to run fine on top of Debian seems like there is something wrong with the HAOS boot sequence on this platform.

https://github.com/home-assistant/operating-system/issues/1752

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Used balena etcher and URL to burn OS to SSD from a laptop
2. installed SSD in PC
3. booted to blinking cursor
4. repeated the above steps 1 through 3 to try booting from a USB stick instead of the SSD, same result
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Initial boot never succeeded
```

### Anything in the Host logs that might be useful for us?

```txt
Initial boot never succeeded
```

### System information

Dell Optiplex FX160
BIOS A13
RAM 4GB
SSD 128GB
1.6 GHZ Intel Atom 230 Diamondville CPU

agners, 2023-03-01:
Hm, I assume this is a 32-bit UEFI BIOS, but that should be supported with latest HAOS.

Did you try a different SSD, maybe a USB drive?

Debian runs an installer, which also runs `efibootmgr` configure a boot entry. Some people report that certain system require to run that command manually using a live distribution to make HAOS boot, see https://github.com/home-assistant/operating-system/issues/1760#issuecomment-1077716696. Maybe that is the case for your system as well?

Can you try a latest dev build? It comes with a newer GRUB2, maybe that helps:
https://os-builds.home-assistant.io/10.0.dev20230228/

closed 2023-06-06 by github-actions[bot]

thijstriemstra, 2023-06-10:
> However, when HAOS is installed, HAOS gets to a screen with a blinking underscore and wont boot past that even when left for 10 hours.

I'm running into the same issue with plain Debian and Optiplex FX160, so it's not a home assistant specific issue. Trying Ubuntu Core now..