home-assistant/operating-system #2367: Outated NetworkManager causes stale Thread IPv6 routes to get stranded in the kernel routing table

ToastyYak, 2023-02-24:
### Describe the issue you are experiencing

Full details at the NetworkManager issue tracker here: [https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/1232](https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/1232)

In short, NetworkManager 1.34 included with HAOS has a bug that causes it to lose track of routes in the kernel routing table when it received multiple ICMPv6 RA RIOs to the same network.

Pulling in NM 1.40.10 from [upstream buildroot](https://github.com/buildroot/buildroot/tree/2023.02-rc1/package/network-manager) resolve that issue.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install latest HAOS
2. Check routing table via `ip -6 route` or `route -A inet6` to confirm no routes pointing to "dead::/64"
3. Run the following in scapy, changing eth0 if necessary, to emulate a Thread Border Router ICMPv6 RA:
```python
send(IPv6(src="fe80::aaaa", dst='FF02::1')/ICMPv6ND_RA(routerlifetime=0,reachabletime=0)/ICMPv6NDOptSrcLLAddr(lladdr=get_if_hwaddr(xDEV))/ICMPv6NDOptRouteInfo(plen=64, rtlifetime=10, prefix='dead::'))
```
4. Within 10 seconds, run the following in scapy:
```python
send(IPv6(src="fe80::bbbb", dst='FF02::1')/ICMPv6ND_RA(routerlifetime=0,reachabletime=0)/ICMPv6NDOptSrcLLAddr(lladdr=get_if_hwaddr(xDEV))/ICMPv6NDOptRouteInfo(plen=64, rtlifetime=10, prefix='dead::'))
```
5. Note that the second route never expires.

As an aside, the lifetime is managed internally by NM and won't show up in the kernel routing table itself, so `ip` missing that info is expected.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### Additional information

I don't know exactly when between NM 1.34 and 1.40 the issue was resolved, but an NM project owner did say that at the issue on their tracker that IP address handling was heavily worked before 1.36. "Heavily reworked" makes me think a simple patch like the recent DHCPv4 NAK one isn't going to cut it.

Jc2k, 2023-02-24:
We also have https://github.com/home-assistant/operating-system/discussions/2333 where we are talking about NM's handling of these RA's in general.

agners, 2023-03-01:
dev builds [10.0.dev20230228](https://os-builds.home-assistant.io/10.0.dev20230228/) and newer come with NetworkManager 1.40.10. So with HAOS 10.x release this should be addressed.

closed 2023-03-01 by agners

Jc2k, 2023-03-01:
(For anyone finding this in the future, at this point we haven't fixed the route table flapping on every RA, and afaik upstream NM is still buggy here)