home-assistant/operating-system #2370: Not Able to Boot from SSD when other USB Hubs are plugged in.

furley11, 2023-02-26:
### Describe the issue you are experiencing

When I have a power outage, the reboot of my RPi, which runs on an SSD connected by USB, doesn't happen. I have to unplug the two (Zigbee and ZWave) USB hubs and cycle the power to get it to reboot. Then, of course, everything boots up fine, and I plug in the two hubs with no issues.

I assume this is because it is trying to boot from one of the hubs instead of the SSD. While the issue is easy to solve each time, I would like to be able to tell it to boot from the SSD so that during a power outage or flicker, I don't lose all functionality.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

core-2023.2.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

cycle power

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

## System Information

version | core-2023.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.84-v8
arch | aarch64
timezone | America/Chicago
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.30.1
Stage | running
Available Repositories | 1224
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | May 22, 2023 at 7:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.01.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 219.4 GB
disk_used | 10.3 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.5.2), Z-Wave JS (0.1.76), OneDrive Backup (1.41), SSH & Web Terminal (13.0.2), MariaDB (2.5.2), File editor (5.5.0), Samba share (10.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 25, 2023 at 4:53 AM
-- | --
current_recorder_run | February 25, 2023 at 10:23 AM
estimated_db_size | 197.34 MiB
database_engine | mysql
database_version | 10.6.10

</details>

kdelios, 2023-02-26:
Looks like a power issue. Use a powered USB hub or a 3,5A power source for your Rpi

furley11, 2023-02-28:
> Looks like a power issue. Use a powered USB hub or a 3,5A power source for your Rpi

I am already using a sufficient power supply. It runs with both hubs plugged in. I just can't boot with them plugged in.

agners, 2023-02-28:
I agree with @kdelios this looks like a power issue. Running is often not the problem, but [in-rush current](https://en.wikipedia.org/wiki/Inrush_current) can exceed what your power supply can provide. The more devices connected, the higher in-rush current (all those capacitors in your USB hubs need to get charged :cold_sweat: 

Usually, when initializing the SSD later things just work. So I'd recommend booting from SD card, and use the data disk feature.

closed 2023-06-05 by github-actions[bot]