home-assistant/operating-system #2371: Latest HAOS image for RPI3, flashed on 64GB Micro SD, uses not working NTP time.cloudflare.com. Install stuck.

Chris-Meijer, 2023-02-26:
### Describe the issue you are experiencing

1. flash the latest image (haos_rpi3-64-9.5.img.tar) to a micro SD card of 64GB (!) (SanDisk Extreme Pro A2)
2. Connect Network (Cat cable) to RPI3B-64 and insert card
3. Connect power supply (3A)
4. Wait for installation to finish on URL homeassistant.local.8123. It never happens.
5. Checked the log on homeassistant.local.8123, I found an error setting the system time (see attachments).

Strange fact: I think it worked ok when the image is flashed on a 32GB Samsung Evo card. Did not repeat that test.

To solve the problem, I connected a keyboard and a display to RPI3. In HA terminal on the prompt HA>, I entered 'login'.
After a lot of research I used vi to change the /etc/systemd/timesyncd.conf.
In the [ time] section I added NTP=nl.pool.ntp.org
Now the installation continued. Still got a lot of errors, but at least it did finish.

Can you change the image in a way that it does use an NTP-server that is working, please?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

 	core-2023.2.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. flash the latest image (haos_rpi3-64-9.5.img.tar) to a micro SD card of 64GB (SanDisk Extreme Pro U3, V30, A2)
2. Connect Network and Internet (Cat cable) to RPI3B-64 and insert card
3. Connect power supply (3 Amps)
4. Wait for installation to finish on URL homeassistant.local.8123. It never happens.
5. Check the log on homeassistant.local.8123 (click on blue blinking dot).
search for 1st error:
 -----------------------------
22-07-13 13:32:47 CRITICAL (MainThread) [supervisor.core] Fatal error happening on load Task <coroutine object Core._adjust_system_datetime at 0x7fb9ef1150>: Automatic time synchronization is enabled
------------------------------

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-02-26 11:36:59 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-02-26 11:49:29 INFO (MainThread) [supervisor.host.info] Updating local host information[0m
[32m23-02-26 11:49:30 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m23-02-26 11:49:31 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m23-02-26 11:49:31 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information[0m
[32m23-02-26 11:49:31 INFO (MainThread) [supervisor.host.manager] Host information reload completed[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING[0m
[32m23-02-26 12:02:47 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m23-02-26 12:02:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository[0m
[32m23-02-26 12:02:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository[0m
[32m23-02-26 12:02:57 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository[0m
[32m23-02-26 12:03:01 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove[0m
[32m23-02-26 12:03:01 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove[0m
[32m23-02-26 12:07:00 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[... 276 lines cut ...]
version_api | ok
installed_addons | SSH & Web Terminal (13.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21 februari 2023 om 19:40
-- | --
current_recorder_run | 25 februari 2023 om 21:04
estimated_db_size | 18.84 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2023-02-28:
Most likely your provider or firewall blocks NTP usage. Can you check with your provider which NTP server is allowed?

Unfortunately, there is no easy way to change the NTP server at this point. The best option is to provide a custom NTP server through DHCP.

kwassholm, 2023-02-28:
I'm having the same issue where the "Preparing Home Assistant" phase is stuck due to "certificate has expired or is not yet valid" error.
I tried "ntpdate -q time.cloudflare.com" command on another Raspberry Pi that is connected to the same network and it is ok, so I don't think the issue is with ISP blocking NTP usage, at least on my case.

There's also this in the log file:
```
22-07-13 14:32:31 WARNING (MainThread) [supervisor.core] System time/date shift over more than 3 days found!
22-07-13 14:32:31 INFO (MainThread) [supervisor.host.control] Setting new host datetime: 2023-02-27T17:57:33+00:00
22-07-13 14:32:31 CRITICAL (MainThread) [supervisor.core] Fatal error happening on load Task <coroutine object Core._adjust_system_datetime at 0x7f8a2f5150>: Automatic time synchronization is enabled
```

To me, it seems that it did try to set the new datetime but then failed to do so, for some reason. Or is that just a red herring?
According to the internets auto sync needs to be disabled if one tries to manually set the datetime, but I don't know if that's the case here.

closed 2023-06-05 by github-actions[bot]