home-assistant/operating-system #2376: Hyper-V Snapshots Failing

scyto, 2023-03-01:
### Describe the issue you are experiencing

I am trying to take a backup of hassos on Hyper-V using synolgy active backup.
While i can take a non-applications aware snapshot the application aware snapshots are failing.

I followed t[his troubleshooting article](https://kb.synology.com/en-us/DSM/tutorial/What_can_I_do_when_production_checkpoint_cannot_be_created_ABB) and confirmed i cannot take a snapshot in the hyperv ui. This article implied that the services required for checkpoint creation are not installed or running.

I then looked to t[his Microsoft article](https://learn.microsoft.com/en-us/windows-server/virtualization/hyper-v/manage/manage-hyper-v-integration-services#start-and-stop-an-integration-service-from-a-linux-guest) to identify if the services were running.

running `lsmod | grep hv_utils` gives an output as follows:

```
➜  ~ lsmod | grep hv_utils
hv_utils               45056  0 
```
this is different to the example outpu in the MS article (no hv_vmbus) listed

running `ps -ef | grep hv` gives output as follows: 
[image]

This would indicate that the vss daemon (hv_vss_daemon is not running (and the kvp daemon too as just a point of note)

I can't run compgen as recommended in the MS article as that package isn't installed in hassos

tl;dr it seems the VSS daemon is not installed or not running for some reason - i will keep digging through docs but this is the first time i have dug deep on the daemons.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Follow the **METHOD 1** steps in this article to reproduce the error in hyper-v

https://kb.synology.com/en-us/DSM/tutorial/What_can_I_do_when_production_checkpoint_cannot_be_created_ABB

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing of note
```

### Anything in the Host logs that might be useful for us?

```txt
nothing of note, all the logs seem to be about docker containers
```

### System information

## System Information

version | core-2023.2.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.7
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.30.0
Stage | running
Available Repositories | 1241
Downloaded Repositories | 36

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 42

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | beta
supervisor_version | supervisor-2023.01.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 125.3 GB
disk_used | 21.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.5.2), SSH & Web Terminal (13.0.2), MariaDB (2.5.2), Node-RED (14.0.3), TasmoAdmin (0.23.0), Glances (0.17.2), ESPHome (2022.3.1), Network UPS Tools (dev), Spotify Connect (0.12.2), AirCast (3.5.1), Frigate (Full Access) Beta (0.12.0) (0.12.0-beta7), ESPHome (2023.2.4), Samba share (10.0.0), Samba Backup (5.2.0), Network UPS Tools (0.11.0)

</details>

<details><summary>Universal Devices ISY/IoX</summary>

host_reachable | ok
-- | --
device_connected | true
last_heartbeat | February 28, 2023 at 6:08 PM
websocket_status | connected

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 23
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 25, 2023 at 9:57 PM
-- | --
current_recorder_run | February 26, 2023 at 6:43 PM
estimated_db_size | 607.20 MiB
database_engine | mysql
database_version | 10.6.10

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

scyto, 2023-03-01:
ahh, i wonder if this is the issue....

how can I update the version of the intregrations (i still don't understand why i cant see the daemons in the VM console, but whatever)

[image]

agners, 2023-03-01:
> generic-x86-64 (Generic UEFI capable x86-64 systems)

The `generic-x86-64` image is meant for bare-metal x86-64 systems. For virtualization, please use the OVA image.

That said, currently we don't install any user space tools for Hyper-V, so this is just a feature "not implemented"... That is why you don't see the daemons in the console. It is weird that Hyper-V claims that the protocol version is wrong, since we don't have any VSS service enabled, but I guess that is just some Hyper-V weirdness.

We collect feature requests in our Community Forum at: https://community.home-assistant.io/c/feature-requests/13.

closed 2023-03-01 by agners

scyto, 2023-03-02:
Thanks, I did use the official hassos VHDX - i just don't think of VHDX being OVA compliant so didn't pick that - so my bad.

After doing more research it seems the protocol message is spurious (thanks MS).

There are some Hyper-V items installed in the image (see the screenshot below) just not the complete set. Maybe its inherited from the base image used to construct this and accidental? 

Either way it sounds like the answer is the same - use the feature request forum, will do.

[image]

Lastly is there a way for me to assist (i previously did some check-ins for the Roomba integration), i don't immediately see in this repro where the build files are for creating the VHDX / what the process is for building the VHDX?

scyto, 2023-03-02:
feature request logged as directed https://community.home-assistant.io/t/add-hyper-v-guest-services-to-hyper-v-vhdx-image-to-support-standard-and-production-snapshots/542154