home-assistant/operating-system #2378: How to manually install mesa-va-drivers for ffmpeg hardware acceleration ?

DennisFury, 2023-03-01:
### Describe the issue you are experiencing

Operating System 9.5
Home Assistant 2023.2.5
Supervisor 2023.01.1

How can we get mesa-va-drivers package installed to support AMD APU hardware acceleration support for ffmpeg? 

HA OS does not seem to have a package manager, and I'm unsure how to manually build this driver into the OS and HA container.  Can one of the HA OS gurus help provide a guide?

https://github.com/blakeblackshear/frigate/issues/311

This would help us Frigate add-on users who have AMD CPUs. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have HA OS on AMD APU
2. Want to use hadware acceleration for ffmpeg
3. Need driver installed in HA OS, but how?

### Anything in the Supervisor logs that might be useful for us?

```txt
N/a
```

### Anything in the Host logs that might be useful for us?

```txt
N/a
```

closed 2023-03-07 by github-actions[bot]

crookedview, 2023-03-12:
@DennisFury were you ever able to get hardware acceleration working with your AMD CPU?  I have an A10-9700E and trying to get Home Assistant to see the GPU device but haven't had much luck thus far.

> ### Describe the issue you are experiencing
> 
> Operating System 9.5 Home Assistant 2023.2.5 Supervisor 2023.01.1
> 
> How can we get mesa-va-drivers package installed to support AMD APU hardware acceleration for ffmpeg? Video decoding takes up a lot of CPU%.
> 
> HA OS does not seem to have a package manager, and I'm unsure how to manually build this driver into the OS and HA container. Can one of the HA OS gurus help provide a guide?
> 
> [blakeblackshear/frigate#311](https://github.com/blakeblackshear/frigate/issues/311)
> 
> This would help us Frigate add-on users who have AMD CPUs.
> ### What operating system image do you use?
> 
> generic-x86-64 (Generic UEFI capable x86-64 systems)
> ### What version of Home Assistant Operating System is installed?
> 
> 9.5
> ### Did you upgrade the Operating System.
> 
> Yes
> ### Steps to reproduce the issue
> 
>     1. Have HA OS on AMD APU
> 
>     2. Want to use hadware acceleration for ffmpeg
> 
>     3. Need driver installed in HA OS, but how?
> 
> 
> ### Anything in the Supervisor logs that might be useful for us?
> 
> ```
> N/a
> ```
> 
> ### Anything in the Host logs that might be useful for us?
> 
> ```
> N/a
> ```
> 
> ### System information
> 
> _No response_
> ### Additional information
> 
> _No response_

bbccdd, 2023-11-14:
Hi. I'm also interested. Running HAOS on Proxmox on AMD Ryzen 7 5800X CPU. Any pointers perhaps?

tuxevil, 2023-11-23:
hi everyone

i just got a ryzen 5700u to replace my very limited i3 6100t and vaapi is giving me error

for sure now i have plenty of cpu power to use but im looking for scale up from 32 cameras to 96 and i want to optimice resources and power
i have currently 45 cameras and this cpu is a beast, ~30% usage and both corals are around 7ms but there is still a long way to go to my final 96 cameras setup

im researching about it, and saw the fix for container installation, but im running haos

please can someone says the steps to use iGPU from ryzen