home-assistant/operating-system #2384: Time is wrong and inconsistent

SamMousa, 2023-03-05:
### Describe the issue you are experiencing

In the HA frontend I noticed weird history entries saying the light I turned on was turned on (for example) in 90 minutes.
After I turned it off 2 minutes later it noted added an entry saying I turned it off at the same instant.
So not only is my time off by a random amount, it is also not "flowing" consistently...

The output of `ha host info` shows this: 
```sh
agent_version: 1.4.1
apparmor_version: 3.0.3
boot_timestamp: 1676726005246869
broadcast_llmnr: true
broadcast_mdns: true
chassis: vm
cpe: cpe:2.3:o:home-assistant:haos:9.5:*:production:*:*:*:ova:*
deployment: production
disk_free: 14.2
disk_life_time: null
disk_total: 30.8
disk_used: 15.3
dt_synchronized: true
dt_utc: "2023-03-05T20:35:17.979115+00:00"
features:
- reboot
- shutdown
- services
- network
- hostname
- timedate
- os_agent
- haos
- resolved
- journal
hostname: ""
kernel: 5.15.90
llmnr_hostname: homeassistant
operating_system: Home Assistant OS 9.5
startup_time: 1.043058
timezone: Etc/UTC
use_ntp: true
```

The time listed at `dt_utc` hasn't changed for at least 20 minutes. (And it is 23:22 local time now, or 22:22 UTC) 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Not sure

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing related to time synchronization or QEMU guest agent
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing related to time synchronization or QEMU guest agent
```

### System information

## System Information

version | core-2023.3.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4953
Installed Version | 1.30.1
Stage | running
Available Repositories | 1240
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | August 12, 2023, 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.01.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 30.8 GB
disk_used | 15.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.5.2), MariaDB (2.5.2), Terminal & SSH (9.6.1), Grafana (8.1.0), Zigbee2MQTT Proxy (proxy), chrony (2.5.0), Mosquitto broker (6.1.3), ESPHome (2023.2.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 5
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 3, 2023, 07:37
-- | --
current_recorder_run | March 2, 2023, 09:20
estimated_db_size | 1338.45 MiB
database_engine | mysql
database_version | 10.6.10

</details>

### Additional information

I'm guessing this has something to do with time synchronization between host and guest system. I'm running inside a qemu virtual machine.

Interestingly, the `date` command will give me proper output:
```sh
[core-ssh ~]$ date
Sun Mar  5 23:25:34 CET 2023
[core-ssh ~]$ date
Sun Mar  5 23:25:35 CET 2023
[core-ssh ~]$
```

Within HA printing time with a jinja template yields correct results.

Besides a solution I'm also looking for pointers on where to debug this further!

agners, 2023-03-07:
> In the HA frontend I noticed weird history entries saying the light I turned on was turned on (for example) in 90 minutes.
After I turned it off 2 minutes later it noted added an entry saying I turned it off at the same instant.
So not only is my time off by a random amount, it is also not "flowing" consistently...

That sounds very much like a virtualization issue. What type of hypervisor are you using?

Is this still happening later, at random, or was this maybe a one off situation? Did you use pause/freeze feature of your Hypervisor?

SamMousa, 2023-03-07:
> That sounds very much like a virtualization issue. What type of hypervisor are you using?

Im using qemu via virt-manager.

> Is this still happening later, at random, or was this maybe a one off situation?

The time jumps around randomly, so it is not one off. I do not know however if it one thing caused it to behave this way and now it us us simply not recovering. 

> Did you use pause/freeze feature of your Hypervisor?

No, never used these, but turns out timing is a complicated issue in virtualization. What im wondering is why there seem to be 2 sources for time. Since commandline `date` is correct at the very least it could be detected that something is wrong...

agners, 2023-03-07:
> No, never used these, but turns out timing is a complicated issue in virtualization. What im wondering is why there seem to be 2 sources for time. Since commandline `date` is correct at the very least it could be detected that something is wrong...

There is only one source of time, provided by the HAOS kernel. The time*zones* are different, HAOS uses UTC, whereas the containers (e.g. Home Assistant Core) use the configured time zone. However, that appears not to be the issue here.

My best guess is that the time jumped for some reason, sometime after your light turned on according to the log. HAOS uses NTP to synchronize time, on the host shell you can check `journalctl -u systemd-timesyncd.service` to see if it maybe synchronized time right then.

I don't run my production system on KVM, so I can't really speak from expierence, but there are plenty of users in the community forum which might be able to help debug. Also there is [a community guide](https://community.home-assistant.io/t/install-home-assistant-os-with-kvm-on-ubuntu-headless-cli-only/254941) which might have some hints on what might help.

SamMousa, 2023-03-07:
> My best guess is that the time jumped for some reason, sometime after your light turned on according to the log.

That doesn't explain while 10 invocations of `ha host info`, 2 sec apart give the same result. Or why `date` works correctly consistently... ill get host access and come back... already checked community forums as well as generic qemu resources, no answer there either.

agners, 2023-03-07:
So you are saying within these `ha host info` or `date` invocation you see new log book entries out of order/with a different date? That sounds likely a Core issue then, somehow timestamping the events wrong.

I actually too see events in the future in the logbook, however, only a few seconds. This is probably due to time difference of my host system (where my Browser is running) and where HAOS is running.

SamMousa, 2023-03-07:
No, apparently I'm not being clear.
I'm seeing weird logbook entries that I think are caused by the time from `ha host info` not changing.
So if that is stuck at say `23:22` but it is now `22:15` each entry will say `in 1 hour`.

When I run the `date` command in the terminal I see the correct time however. 

So the terminal's `date` works, but `ha host info` doesn't. (I'm assuming that `dt_utc` is supposed to be the current timestamp? Or is it the last time it synced?)

agners, 2023-03-07:
Hm, I see, I can reproduce that on a system running on Yellow:

```
# ha host info
...
dt_utc: "2023-03-07T06:45:15.980173+00:00"
...
```

This is the time read by Supervisor from the host systems systemd-timedate service. [It seems](https://github.com/home-assistant/supervisor/blob/613ef9010aa5eb350b403a1067bd3f452bb1fe7e/supervisor/dbus/timedate.py#L55-L57) the `TimeUSec` property is used. From  what I can tell, this returns the time on every invocation, at least when calling `busctl` manually on the host system:

```
busctl get-property org.freedesktop.timedate1 /org/freedesktop/timedate1 org.freedesktop.timedate1 TimeUSec
```

I am guessing what we are seeing here is just a cached value. Maybe @mdegat01 can comment on that.

However, I don't think that this is related to the issue you are seeing in the logbook.

SamMousa, 2023-03-07:
Hmm, you might be right.
Ran this script for a bit:
```bash
#!/bin/bash 

for ((i=1;i<=100;i++));
do
	echo "$(ha host info | grep dt_utc)  |  date: $(date)   |  curl: $(curl  -X 'GET' 'https://www.timeapi.io/api/Time/current/zone?timeZone=Europe/Amsterdam' -s -H 'accept: application/json' | jq '.dateTime' )"
	sleep 1
done
```

And it shows: 
```
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:09 CET 2023   |  curl: "2023-03-07T12:47:09.8453226"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:10 CET 2023   |  curl: "2023-03-07T12:47:10.9083326"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:11 CET 2023   |  curl: "2023-03-07T12:47:11.9691523"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:12 CET 2023   |  curl: "2023-03-07T12:47:13.0296313"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:14 CET 2023   |  curl: "2023-03-07T12:47:14.090409"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:15 CET 2023   |  curl: "2023-03-07T12:47:15.149943"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:16 CET 2023   |  curl: "2023-03-07T12:47:16.1972684"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:17 CET 2023   |  curl: "2023-03-07T12:47:17.2566704"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:18 CET 2023   |  curl: "2023-03-07T12:47:18.316483"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:19 CET 2023   |  curl: "2023-03-07T12:47:19.3772191"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:20 CET 2023   |  curl: "2023-03-07T12:47:20.438364"
dt_utc: "2023-03-07T10:35:23.834622+00:00"  |  date: Tue Mar  7 12:47:21 CET 2023   |  curl: "2023-03-07T12:47:21.4987975"
```

That said, I'm still left with this strange behavior:
[image]

This is a light that I just toggled. The history bar doesn't show the correct state (it is on). The logbook shows that it was turned on in the future. 
The jinja template `{{ now() }}` shows this:
[image]

Does this seem more like a core issue then? (While researching the problem I found the reference to `dt_utc` from HA OS but if that is just a cached value and considering that the value is not the bad timestamp i'm seeing in the UI, it might be a red herring.

If you think it is a core issue, should I move this to core or create a new one and reference this one?

agners, 2023-03-07:
Is it maybe just a simple time missmatch between the machine HA is running on and the machine your Browser is running on?

If both times are accurate, then it could be a Core or Frontend issue. I'd suggest to open a new issue with the latest findings. The `dt_utc` is just a Supervisor/HA CLI issue and most likely unrelated.

SamMousa, 2023-03-07:
I think it was a database issue; recreating it in MariaDB fixed it. Now hoping to transfer my long term statistics to the new one somehow. Thanks for your time.

closed 2023-03-07 by SamMousa