home-assistant/operating-system #2390: Yellow reboot on CM4 EMMC and powercycle on NVME disk

Antexa, 2023-03-07:
### Describe the issue you are experiencing

I got a strange behavior on my Yellow, if I go with a system reboot, Yellow boot on CM4 EMMC and if I power cycle the Yellow board it boots on NVME drive. 

this behavior appear after a factory reset (I press the two button on the board, the red and the blue according to this : 
https://yellow.home-assistant.io/faq/#can-i-reinstall-home-assistant-os-again-using-the-usb-flash-drive-with-the-home-assistant-os-installer-for-yellow

[image]

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. System reboot go into CM4 EMMC
2. Powercycle the yellow board go into NVME (as expected!) 

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-03-07 09:31:23 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING[0m
[32m23-03-07 09:31:23 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE[0m
[32m23-03-07 09:31:23 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING[0m
[32m23-03-07 09:31:24 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m23-03-07 09:33:36 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-03-07 09:40:19 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-03-07 09:40:19 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m23-03-07 10:03:36 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING[0m
[32m23-03-07 10:31:24 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m23-03-07 10:33:02 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json[0m
[32m23-03-07 10:33:24 INFO (MainThread) [supervisor.host.info] Updating local host information[0m
[32m23-03-07 10:33:26 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m23-03-07 10:33:26 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m23-03-07 10:33:26 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information[0m
[32m23-03-07 10:33:26 INFO (MainThread) [supervisor.host.manager] Host information reload completed[0m
[32m23-03-07 10:33:36 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-03-07 11:03:37 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m23-03-07 11:13:22 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m23-03-07 11:13:22 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m23-03-07 11:31:24 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING[0m
[... 114 lines cut ...]
Mar 07 10:32:28 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7498eee9ec57f20fbbcfc93ccacbc2e118f603cee9e3b288a17a751df5aa7b85-runc.5v9CiJ.mount: Deactivated successfully.
Mar 07 10:32:30 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6ecd3418cf70193863bfa6ae2be1f49603fc3cf1dca550dab6d4252aa3216748-runc.PZ1Pw8.mount: Deactivated successfully.
Mar 07 10:34:09 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-daa63d6d0efaf22cf7608c4e9dd6749d335da9ac44362036749e9178f3cf08c2-runc.hdr3Ke.mount: Deactivated successfully.
Mar 07 10:34:23 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a8704cb2c43955a563a1544ab63dc3907be1f9430937f02068328c478bb18bae-runc.Z8LPVW.mount: Deactivated successfully.
Mar 07 10:34:28 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7498eee9ec57f20fbbcfc93ccacbc2e118f603cee9e3b288a17a751df5aa7b85-runc.AfWg1g.mount: Deactivated successfully.
Mar 07 10:35:01 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6ecd3418cf70193863bfa6ae2be1f49603fc3cf1dca550dab6d4252aa3216748-runc.Z27jIa.mount: Deactivated successfully.
Mar 07 10:35:10 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-daa63d6d0efaf22cf7608c4e9dd6749d335da9ac44362036749e9178f3cf08c2-runc.SmLGy3.mount: Deactivated successfully.
Mar 07 10:35:24 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a8704cb2c43955a563a1544ab63dc3907be1f9430937f02068328c478bb18bae-runc.168jTk.mount: Deactivated successfully.
Mar 07 10:35:29 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7498eee9ec57f20fbbcfc93ccacbc2e118f603cee9e3b288a17a751df5aa7b85-runc.TWPLQ2.mount: Deactivated successfully.
Mar 07 10:35:54 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a8704cb2c43955a563a1544ab63dc3907be1f9430937f02068328c478bb18bae-runc.0HjTw2.mount: Deactivated successfully.
Mar 07 10:35:59 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7498eee9ec57f20fbbcfc93ccacbc2e118f603cee9e3b288a17a751df5aa7b85-runc.xcY7Mp.mount: Deactivated successfully.
Mar 07 10:36:01 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6ecd3418cf70193863bfa6ae2be1f49603fc3cf1dca550dab6d4252aa3216748-runc.hR23CG.mount: Deactivated successfully.
Mar 07 10:36:24 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a8704cb2c43955a563a1544ab63dc3907be1f9430937f02068328c478bb18bae-runc.D8dJIg.mount: Deactivated successfully.
[... cut ...]
Mar 07 10:39:25 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-a8704cb2c43955a563a1544ab63dc3907be1f9430937f02068328c478bb18bae-runc.icGz5H.mount: Deactivated successfully.
Mar 07 10:39:39 homeassistant kernel: audit: type=1334 audit(1678185579.354:690): prog-id=225 op=LOAD
Mar 07 10:39:39 homeassistant systemd[1]: Started Journal Gateway Service.
Mar 07 10:39:39 homeassistant systemd-journal-gatewayd[416144]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
```

agners, 2023-03-07:
How do you know what system it booted?

Most likely this is related to the data disk feature: We use a file system label `hassos-data` to determine which is the data disk. The factory reset created another `hassos-data` partition on the eMMC, so now it randomly mounts either the data disk from eMMC or NVMe.

Antexa, 2023-03-08:
`How do you know what system it booted?
`
With help from the page config/storage
(eMMC 32GB / NVME 1To)
[image]

`Most likely this is related to the data disk feature: We use a file system label hassos-data to determine which is the data disk. The factory reset created another hassos-data partition on the eMMC, so now it randomly mounts either the data disk from eMMC or NVMe.`
Why the factory reset didn't reset this label ? I also noticed that when I performed the factory reset my NVME drive has not been wiped.

agners, 2023-03-08:
> Why the factory reset didn't reset this label ? I also noticed that when I performed the factory reset my NVME drive has not been wiped.

Yeah indeed, the blue and red button resets only the eMMC currently. If data disk is enabled, this is problematic. It should also clear the NVMe, this is a bug.

agners, 2023-03-14:
> Do you think it's possible to delete de hass-os data label ? To have only have the NVMe disk drive at boot

Yes, with debug access to the host OS you can use the `e2label` command to rename it (the built-in data disk feature does the same, it changes the name to `hassos-data-old`.

jov58, 2023-04-18:
I have exactly the same behaviour but the other way around: if I power-cycle, Yellow boots and uses eMMC. If I then restart HA through the System Menu, it boots using NVMe. This happens most of the time: occasionally Yellow boots with NVMe after a power-cycle or with eMMC after a restart. I wonder if the real underlying cause is some hardware timing problem or so.

So, my "normal" state running Yellow is when the system is using NVMe (faster than eMMC).

**Since `deactivate any external data disk device on first boot` now is present in OS V10.0, I am a bit afraid to update to V10.**

What will happen in my case, won't Yellow boot anymore and refuse to use my NVMe? If so, how do I proceed after that and is it possible to restore my complete config (with add-ons and everything) from a full backup of the NVMe?

I also don't understand the preference for eMMC over NVMe. Why not deactivate eMMC when a NVMe is found? NVMe is much, much faster than eMMC and seems to wear out less quickly.

Isn't it possible to wipe the eMMC, flash the Home Assistant Operating System 10.0 to a NVMe and boot with that, without ever using eMMC?

Lot of questions, any help in understanding the boot procedure really appreciated.

agners, 2023-04-20:
> Since deactivate any external data disk device on first boot now is present in OS V10.0, I am a bit afraid to update to V10.

This indeed could be a problem for your case. The system might deactivate the NVMe.

> I also don't understand the preference for eMMC over NVMe. Why not deactivate eMMC when a NVMe is found? NVMe is much, much faster than eMMC and seems to wear out less quickly.

Most people start with eMMC and only then add a NVMe. So this is the natural way.

Unfortunately the CM4 EEPROM firmware seems not to be able to boot from every NVMe (see also https://github.com/home-assistant/operating-system/discussions/2235).. I think most of those issue got resolved with newer firmwares. But I am still a bit hesitant. Also the firmware boots from eMMC first, so booting from eMMC seems a quicker and more reliable option.

Note that Home Assistant OS is a compressed squashfs of ~200MB. Most things are loaded only once and stay in memory/page cache. You can also argue that storing HAOS on eMMC frees up all I/O's of the NVMe for Home Assistant Core, Add-on etc.

> Isn't it possible to wipe the eMMC, flash the Home Assistant Operating System 10.0 to a NVMe and boot with that, without ever using eMMC?

That is possible with the latest Home Assistant OS Installer for Yellow, by pressing the blue button right after the installer startup completed (see description [here](https://yellow.home-assistant.io/power-supply-nvme/#connecting-the-power-supply)). The feature is a bit experimental at this point, so it is a bit hidden.

For your case, I recommend to start with a clean slate:

1. Take full backup in your current installation and download the backup
2. [Reinstall Home Assistant OS using Option 1](https://yellow.home-assistant.io/guides/reinstall-os/). Force installation onto the NVMe [as described in Step 9 of the Software Installation procedure](https://yellow.home-assistant.io/power-supply-nvme/#connecting-the-power-supply).
3. Once the system is in onboarding mode, upload your backup using the "Alternatively you can restore from a previous backup" link at the bottom of the first onboarding step.

jov58, 2023-04-29:
I'm trying to reinstall HAOS, but I just can't get it to work. I follow the instructions from [How to reinstall the operating system](https://yellow.home-assistant.io/guides/reinstall-os/).

As recommended on that page, I first used the [device wipe system](https://yellow.home-assistant.io/faq/#can-i-factory-reset-the-system). It doesn't seem to have any effect, Yellow comes up with my current config.

Option 1 also doesn't seem to have any effect. Yellow keeps booting up with my current config and doesn't boot from the USB Flash drive.

I'm not sure how to proceed now, any help appreciated. Perhaps there is a way to completely reset my CM4 with eMMC to it's pristine out-of-the-box state?

closed 2023-08-04 by github-actions[bot]