home-assistant/operating-system #2395: hanging on Preparing Home Assistant

alarazr, 2023-03-10:
### Describe the issue you are experiencing

Preparing Home Assistant page indicates a 20 minutes wait time. Hanging for hours

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Install Home Assistant OS 9.5 on RPi4 on SSD
2. Access IP address to Preparing Home Assistant page
3. Wait forever 
...

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[23:49:29] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[23:49:29] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
23-03-09 23:49:31 INFO (MainThread) [__main__] Initializing Supervisor setup
23-03-09 23:49:31 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
23-03-09 23:49:31 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2023.03.1
23-03-09 23:49:31 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
23-03-09 23:49:31 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['ghcr.io/home-assistant/aarch64-hassio-supervisor:2022.12.1']
23-03-09 23:49:33 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
23-03-09 23:49:33 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-03-09 23:49:33 INFO (MainThread) [__main__] Setting up Supervisor
23-03-09 23:49:33 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
23-03-09 23:49:33 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
23-03-09 23:49:33 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
23-03-09 23:49:34 INFO (MainThread) [supervisor.host.services] Updating service information
23-03-09 23:49:34 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
23-03-09 23:49:34 INFO (MainThread) [supervisor.host.network] Updating local network information
23-03-09 23:49:34 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
23-03-09 23:49:34 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
23-03-09 23:49:34 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2022.11.0
23-03-09 23:49:34 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2022.04.1
23-03-09 23:49:34 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
23-03-09 23:49:34 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2022.07.0
23-03-09 23:49:34 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0
23-03-09 23:49:34 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0
[... 38 lines cut ...]
23-03-09 23:49:37 INFO (MainThread) [supervisor.host.manager] Host information reload completed
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.module] Create new suggestion SuggestionType.CREATE_FULL_BACKUP - ContextType.SYSTEM / None
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.module] Create new issue IssueType.NO_CURRENT_BACKUP - ContextType.SYSTEM / None
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-03-09 23:49:37 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-03-09 23:49:38 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-03-09 23:49:38 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-03-09 23:49:38 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
```

### Anything in the Host logs that might be useful for us?

```txt
Can't access them
```

### Additional information

Any help is appreciated. Can provide any other information if necessary

jonathanhofman81, 2023-03-13:
I have the same issue with the haos_odroid-n2-9.5.img.xz firmware. Issue #2398 seems to be a comparable problem.

alarazr, 2023-03-13:
I actually just had to wait. Took 1 hour and 20 minutes on a Raspberry Pi 4 8GB

closed 2023-03-13 by alarazr

ml-7, 2024-02-12:
> I actually just had to wait. Took 1 hour and 20 minutes on a Raspberry Pi 4 8GB

I was about 0,5h on my Rpi 3B with one restart, but it was completed.