home-assistant/operating-system #2408: On Yellow enabling UART support for HM_MOD_RPI_PCB in config.txt removes internal Zigbee device /dev/ttyAMA1

traumtheater, 2023-03-14:
### Describe the issue you are experiencing

When trying to use the Momematic RF hat HM_MOD_RPI_PCB in conjunction with the integrated Zigbee module of Yellow it seems that there is a conflict in the UART connection.
When enabling the configuration in config.txt as specified in the documentation of the RaspberryMatic Home-Assistant Add-On ([https://github.com/jens-maus/RaspberryMatic/wiki/Installation-HomeAssistant]), the Homematic RF hat works but the internal Zigbee device is unavailable. The log states:
```
Error setting up entry Yellow for zha
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyAMA1'
```
I learned that a Z-Wave hat which uses the same pinning as the HM_MOD_RPI_PCB works directly out of the box without changing any dtparams.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Attach HM_MOD_RPI_PCB to J11 header of Yellow
2. Edit /mnt/boot/config.txt and uncomment the following three entries:
```
dtparam=i2c_arm=on
dtoverlay=miniuart-bt
dtoverlay=rpi-rf-mod
```
3. Fully reboot the system

Now internal Zigbee is not found anymore and Zigbee is therefore unavailable but HM_MOD_RPI_PCB is usable in RaspberryMatic and the Homematic integration.

### Anything in the Supervisor logs that might be useful for us?

```txt
23-03-14 15:12:00 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
23-03-14 15:12:01 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0 - Host
23-03-14 15:12:01 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-03-14 15:12:01 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/yellow-homeassistant with version 2023.3.4
23-03-14 15:12:01 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 63 Home Assistant secrets
23-03-14 15:12:02 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 9.5 / BootSlot A
23-03-14 15:12:03 WARNING (SyncWorker_0) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of DeskPi Pro Active Cooling
23-03-14 15:12:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/de838cd8 repository
23-03-14 15:12:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
23-03-14 15:12:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/2ad4c73a repository
23-03-14 15:12:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
23-03-14 15:12:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/45df7312 repository
23-03-14 15:12:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/1f3d020e repository
23-03-14 15:12:04 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
23-03-14 15:12:09 WARNING (SyncWorker_1) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of DeskPi Pro Active Cooling
23-03-14 15:12:09 INFO (MainThread) [supervisor.store] Loading add-ons from store: 90 all - 90 new - 0 remove
23-03-14 15:12:09 INFO (MainThread) [supervisor.addons] Found 9 installed add-ons
23-03-14 15:12:09 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to 1f3d020e/aarch64-addon-hassos_ssh_configurator_addon with version 0.9.1a
23-03-14 15:12:09 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.1.3
23-03-14 15:12:09 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/phpmyadmin/aarch64 with version 0.8.4
23-03-14 15:12:09 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mariadb with version 2.5.2
23-03-14 15:12:09 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to ghcr.io/jens-maus/raspberrymatic with version 3.67.10.20230225
23-03-14 15:12:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.6.1
23-03-14 15:12:09 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.5.0
23-03-14 15:12:09 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2023.2.4
23-03-14 15:12:09 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/tailscale/aarch64 with version 0.7.0
23-03-14 15:12:09 INFO (MainThread) [supervisor.backups.manager] Found 23 backup files
23-03-14 15:12:09 INFO (MainThread) [supervisor.discovery] Loaded 2 messages
23-03-14 15:12:09 INFO (MainThread) [supervisor.ingress] Loaded 7 ingress sessions
[... 231 lines cut ...]
healthy | true
supported | true
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2023.2.4), Terminal & SSH (9.6.1), Mosquitto broker (6.1.3), MariaDB (2.5.2), phpMyAdmin (0.8.4), File editor (5.5.0), RaspberryMatic CCU (3.67.10.20230225), Tailscale (0.7.0), HassOS SSH port 22222 Configurator (0.9.1a)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 1
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

error | failed to load: unknown
-- | --

</details>

agners, 2023-03-14:
@jens-maus have you recently used/tested `rpi-rf-mod` on Yellow?

jens-maus, 2023-03-14:
No I haven't actually tested that because I never thought someone would be so keen in using yellow with zigbee and a homematic module in parallel simply because of too high probability of RF interferences.

The issue could easily be the case because the generic `rpi-rf-mod.dts` will make sure to reserve the GPIO-based uart for the GPIO only. See:

https://github.com/jens-maus/RaspberryMatic/blob/57a7174bd55b4c180aefba23d8ba7b4735c0798a/buildroot-external/package/rpi-rf-mod/dts/rpi-rf-mod.dts#L75-L86

So there isn't a specific yellow dts file available yet which will make sure that the other uart will be untouched and working. Or might this be a side effect of using the `miniuart-bt` dtoverlay in parallel?

traumtheater, 2023-03-21:
If I recall it correctly, not using miniuart-bt but the other options results in both Interfaces not working, neither internal Zigbee nor the Homematic hat.

Mar1usW3, 2023-04-11:
i´m curious if this can be fixed or do i need an external zigbee stick?

agners, 2023-04-11:
The overlay is not correct for Yellow, and probably breaks Audio as well.

I'll remove the HomeMatic overlay support for Yellow to avoid issues with the on-board Zigbee.

agners, 2023-04-14:
The overlay won't be part of OS 10, which essentially resolves this issue.

I guess it would be nice to have a device tree overlay which works for `HM_MOD_RPI_PCB`.

> No I haven't actually tested that because I never thought someone would be so keen in using yellow with zigbee and a homematic module in parallel simply because of too high probability of RF interferences.

@jens-maus from what I understand Homeatic is in the 863MHz band, Zigbee is on the 2.4GHz band so these really shouldn't interfere with each other. From a general EMI perspective, we did proper CE certifications and are below the limits both variants, the PoE and Standard variant. And the Yellow Standard variant is even a lot below any limits.

That said, conducted noise from the power supplies etc, can always affect radio performance. We did design the power supplies with radio performance in mind, and at least the Zigbee performance seemed unaffected (as in we reached performance levels Silicon Labs has in their reference implementations).

I'd definitely say it is worth a try :smile:

Chris2834, 2023-04-19:
> No I haven't actually tested that because I never thought someone would be so keen in using yellow with zigbee and a homematic module in parallel simply because of too high probability of RF [...]

@jens-maus : I'm using #yellow and HM_MOD_RPI_PCB by USB connection  succesfull, because I could not get it running with J11 on #Yellow. 
Did not see this issue in time, but can read in your .dts code, that different GPIO's are blocked by standard .dts (and removed for yellow now) than used in #Yellow J11.
Yellow J11 is using GPIO 14/15  as TX/RX port. (See https://yellow.home-assistant.io/hardware/yellow-v1.2-standard/)

If you can give me a hint how to add a correct .dts file, I might check it.

traumtheater, 2023-04-19:
> @jens-maus : I'm using #yellow and HM_MOD_RPI_PCB by USB connection succesfull, because I could not get it running with J11 on #Yellow. Did not see this issue in time, but can read in your .dts [...]

I ordered a HB-RF-USB-2 hoping that I can use the module via USB. Sadly also via this USB adapter my HM-MOD-RPI-PCB is not detected. How did you get it working via USB?

Chris2834, 2023-04-19:
> I ordered a HB-RF-USB-2 hoping that I can use the module via USB. Sadly also via this USB adapter my HM-MOD-RPI-PCB is not detected. How did you get it working via USB?

Yep; I'm using the HB-RF-USB-2 without any HW /SW mod.

But I don't like the "external" wiring and additional box needed. 
Would still prefer the  J11 connection - potentially with add on antenna on the back side - to get a clean setup.

traumtheater, 2023-04-19:
Ok this is a bummer. I really do not know why it is not working on my side. I also do not know how to debug the issue.
Is there a difference which USB port to connect? Or is the jumper setting on the PC important? I see there is a UART/USB setting for example...

smoki3, 2023-04-27:
So does this mean we actually dont have a solution to use HM_MOD_RPI_PCB on the yellow GPIO Pins?

agners, 2023-04-27:
Right now we don't have a tested/verified/working device tree overlay for Yellow, correct. I also don't have the necessary hardware at hand.

smoki3, 2023-04-27:
Do you see a chance to have a short term work around? before we have another solution. (dont want to spent another 30 Euro for the USB variant)

Could it work if we remove all the RPI-RF-MOD stuff from the dts overlay? Because the MOD-RPI-PCB just need the GPIO14 and GPIO15.  Or do we need to use a different uart (for example uart1?)

jens-maus, 2023-04-27:
> Do you see a chance to have a short term work around? before we have another solution. (dont want to spent another 30 Euro for the USB variant)

You mean, I should invest my time for you to save actually 30 EUR?!? ;-)

> Could it work if we remove all the RPI-RF-MOD stuff from the dts overlay? Because the MOD-RPI-PCB just need the GPIO14 and GPIO15. Or do we need to use a different uart (for example uart1?)

This is really IMHO not worth the trouble. Using a HB-RF-USB-2 together with a RPI-RF-MOD or using a HmIP-RFUSB is anyway the better approach than trying to get the obsolete HM-MOD-RPI-PCB running....

Chris2834, 2023-04-27:
I can offer to spent my time on implementation and testing, if I get to know where to modify. I have a second yellow and RF available to check.

I did see the Yellow as best solution to convert my old rPi based Raspberrymatic to more options and reliable hardware with HomeAssistant Yellow - in using the GPIO with the RPI-RF-MOD. I would like to continue that road to go to eliminate as much wiring as possible.

smoki3, 2023-04-27:
> You mean, I should invest my time for you to save actually 30 EUR?!? ;-)

I would rather spent you a coffee instead of buying new hardware and have a cable solution attached to the Yellow. I bought the yellow because this could be a all in one solution if the GPIOs will work. I think it will be very clean box will everythin included. 

> This is really IMHO not worth the trouble. Using a HB-RF-USB-2 together with a RPI-RF-MOD or using a HmIP-RFUSB is anyway the better approach than trying to get the obsolete HM-MOD-RPI-PCB running...

The HB-RF-USB-2 is very big, you need an extra box + cabels in the living room... Even with the HmIP-RFUSB you still have this ugly stick attached to the yellow. 

So the HM-MOD-RPI-PCB looks like the perfect accesory for the yellow. I fully agree with Chis2834. If we have a hint what to test I will do it.

traumtheater, 2023-04-27:
As I have spend the money on the HB-RF-USB-2 and it is still not working I would like to offer my time as well for testing. Even if I somehow get the USB solution to work, I would also prefer a more integrated and clean setup. That is why I wanted to replace my Raspberry 1 with Raspberrymatic in the first place.

@jens-maus Thank you for the amazing support with Raspberrymatic!

jens-maus, 2023-04-27:
> So the HM-MOD-RPI-PCB looks like the perfect accesory for the yellow. I fully agree with Chis2834. If we have a hint what to test I will do it.

Even if the HM-MOD-RPI-PCB looks like a neat solution on a first thought, it actually **is** obsolete/outdated, etc. No newer firmware updates exist for this homematic rf module it and will never get any further firmware updates released by the vendor (eQ3). Currently, the RPI-RF-MOD and HmIP-RFUSB homematic RF modules are the only ones I would recommend these days since they come with the latest 4.4.x firmwares themselves which provide superior homematic/homematicIP support in some areas.

Therefore, I would still suggest to better rely on a RPI-RF-MOD in combination with a HB-RF-USB-2 adapter or use a HmIP-RFUSB for homematic communication and refrain from using the old HM-MOD-RPI-PCB even thought it looks attractive to use it on the GPIO bus.

Chris2834, 2023-04-29:
I hear all the good arguments, but I have still 3 pieces of HM-MOD-RPI-PCB running.
Just want to change to #yellow and my use case is ok with the original FW content of HM-MOD-RPI-PCB.

jens-maus, 2023-05-28:
Let me please allow to re-open this ticket/issue as I am currently working on a HM-MOD-RPI-PCB specific device tree overlay for the Yellow which hopefully will not break Zigbee support.

traumtheater, 2023-06-19:
A quick update from my experiments:
I can happily announce that the issue is fixed with the new HA OS 10.3 release. Now internal Zigbee is working nicely together with the Homematic RF PCB on the J11 header. 
Thanks for everybody and especially @jens-maus for the awesome support! 😊

closed 2023-06-19 by jens-maus

bogosavljevic, 2023-09-22:
Hi All,

Can anyone share some wiring or pictures how to connect HM-MOD-RPI-PCB with yellow on J11 Header.

Thanks