home-assistant/operating-system #2413: No NIC drivers for my PC when installing HAOS

cqsimple, 2023-03-15:
### Describe the issue you are experiencing

I have a OEM2550L2D-MX, Integrated Dual-core Intel Atom processor D2550 4GB ram.  It has two nics.  I have flashed HAOS on a SSD drive and I can boot to the cli.  The network is not initialized.  I cannot get any info regarding the NIC or IP.  I can boot to a debian os and the networking is fine.  I installed HA supervised over debian11 and it is on my network and appears normal.  I would prefer to use HAOS.  Can you help me get this resolved?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.I flashed a version od HA from the url using balenEtcher.
Booted normally but without a network connection
2. Turned off one nic in Bios to see if that affected anything.  Nada
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs
```

### Anything in the Host logs that might be useful for us?

```txt
Again no logs since no network
```

### System information

No gui to get info from

### Additional information

I with I had better info, but I don't

agners, 2023-03-16:
Hm, from what I understand this model uses a Broadcom 57788 NIC. The driver `CONFIG_TIGON3` seems to be the one required in Linux and is enabled :thinking: 

Can you share the output of `lspci`, `lsmod` and `dmesg` in Debian? I wonder if your particular model is indeed using this NIC.

cqsimple, 2023-03-16:
I won't be able to test until next week ; sigh  But I will give it a go.  thanks

agners, 2023-03-23:
Hm, `tg3`, this should be really enabled. Can you try the latest nightly build?
https://os-builds.home-assistant.io/10.0.dev20230323/

bobinoz1, 2023-04-10:
Same problem on old laptop Acer Aspire 4741G i5 64 bit. 
Laptop previously running windows 10 with ethernet (wire and wireless) cards fully operational. Laptop has gpt configured SSD but legacy bios with no UEFI upgrade possible. 
HAOS generic-x86–64 installed using Debian 11 Live on USB drive. Ethernet fully fuctional under Debian 11 allowing installation of HAOS. 
Laptop cards:
Wired:  Ethernet Controller Broadcom Netlink BCM 57780 Gigabit Ethernet PCIe (rev 01)
Firmware/Driver: enp3s0: tg3

Wifi:  Network Controller Broadcom BCM 43225 802.11b/g/n (rev 01)
	Firmware/Driver: wlp5s0b1:brcmsmac
HAOS installs successfully to CLI prompt and cli & nmcli commands function normally but without ethernet functionality so no network ip address
Nmcli commands show no eth0 presence (due to missing drivers under HAOS)
Have tested fresh installs of 9.4, 9.5 and 10.0.dev20230323
https://github.com/home-assistant/operating-system/releases/download/9.4/haos_generic-x86-64-9.4.img.xz    
https://github.com/home-assistant/operating-system/releases/download/9.5/haos_generic-x86-64-9.5.img.xz    
https://os-builds.home-assistant.io/10.0.dev20230323/haos_generic-x86-64-10.0.dev20230323.img.xz

Are you able to add Broadcomm drivers tg3 and brcmsmac please or is there a way to add these manually (which would be painful as would need to re-add these each future upgrade)?

Also without internet access not sure how to capture smesg command output (which runs) or get to relevant logs etc.

agners, 2023-04-12:
> and/or are you able to please check/add Broadcom tigon firmware/drivers tg3 and brcm (for NICs BCM57780 and BCM 43225) and/or

Hm, indeed it seems that the necessary firmware is missing. I've added the firmware to the dev branch with PR #2466 and will backport to OS 10.

closed 2023-04-12 by agners

bobinoz1, 2023-04-12:
OK thanks, will flash fresh OS x86-64 v10 when backported and check/test