home-assistant/operating-system #2414: haos_rpi4-64-9.5.img doesn't even get past "preparing home assistant" on a Raspberry Pi 4B+ 4Gb RAM

alexanderznamensky, 2023-03-16:
### Describe the issue you are experiencing

I'm following official instruction https://www.home-assistant.io/installation/raspberrypi.
Flashing by Balena Etcher image https://github.com/home-assistant/operating-system/releases/download/9.5/haos_rpi4-64-9.5.img.xz.
The HAOS installation is not finishing - always showing "Preparing Home Assistant":
[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Write image by Balena Etcher (https://github.com/home-assistant/operating-system/releases/download/9.5/haos_rpi4-64-9.5.img.xz) to 32Gb eMMC Micro SD card.
2. Started up Raspberry Pi. Find local IP
3. Connect to server on port 8123
4. Nothing happening - always "Preparing Home Assistant"

### Anything in the Supervisor logs that might be useful for us?

```txt
No log
```

### Anything in the Host logs that might be useful for us?

```txt
No log
```

agners, 2023-03-16:
This is not really a HAOS issue, most likely some problem with your network (NTP or DNS filtered/blocked). Click on the blue dot to get additional logs. If you are not able to determine the issue, please check https://www.home-assistant.io/help/.

alexanderznamensky, 2023-03-16:
thanks. but it is strange - I have installed supervised version without any problems.
OK - will use that one