home-assistant/operating-system #2422: rtl8723bs sdio wifi/bluetooth chip support

iLeeeZi, 2023-03-21:
### Describe the issue you are experiencing

Realtek rtl8723bs sdio wifi/bluetooth chip seems to not be enabled on Generic x86-64. Asus Tinker Board has already support for it and the chip is used in many older intel atom powered devices like intel compute stick.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0.dev20230317

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

-

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

agners, 2023-03-25:
What kind of board are you using that you have SDIO on x86-64?

The driver is a staging driver, probably low quality. I'd rather prefer to not enable it unless really necessary.

iLeeeZi, 2023-03-25:
I am using the motherboard from chuwi hi8 air tablet which has intel atom z8350 SoC in it. Previously it had ubuntu and wifi/bluetooth were running fine.

My goal was to have low power consumption as I run it in camper van with battery power, but I can understand if you prefer to not enable it. Usb hub with wifi and bluetooth dongles is always an option.

agners, 2023-05-31:
We have other staging drivers enabled, so if this is popular, I am ok enabling it. Just be aware that this is a staging driver... Expect issues and failures (e.g. as we are seeing with r8188eu, see #2548).

closed 2023-06-01 by agners

Beanow, 2024-09-05:
Hello from the future. Just want to confirm I have indeed gotten issues with the `rtl8723bs` driver.

In particular it fails to reconnect to the access point after the connection fails (like if the AP is reset).
In spite of all the NM settings being proper to retry this indefinitely. It will stay down for hours and need rebooting.

```
Sep 05 19:11:04 your-hostname wpa_supplicant[380]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="SOME_SSID"
Sep 05 19:11:04 your-hostname wpa_supplicant[380]: wlan0: Trying to associate with SOME_BSSID (SSID='SOME_SSID' freq=2427 MHz)
Sep 05 19:11:04 your-hostname wpa_supplicant[380]: wlan0: Association request to the driver failed
Sep 05 19:11:04 your-hostname wpa_supplicant[380]: BSSID SOME_BSSID ignore list count incremented to 2, ignoring for 10 seconds
Sep 05 19:11:04 your-hostname wpa_supplicant[380]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="SOME_SSID" auth_failures=2 duration=20 reason=CONN_FAILED
```

**wlan0: Association request to the driver failed** :eyes:

---

It's a temporary solution for me though, as this device is just waiting for it's LAN cable to be pulled.

PR #2567 (2023-06-01): Add support for RTL8723BS SDIO WiFi modules
Many x86_64 tablets (e.g. Cherry View) use SDIO WiFi modules, this enables the driver for a common one I've come across in the wild.

This module requires firmware from the following, which are already enabled for this platform:

  - BR2_PACKAGE_LINUX_FIRMWARE_RTL_87XX
  - BR2_PACKAGE_LINUX_FIRMWARE_RTL_87XX_BT

fixes #2422
commit 1e67fe4b (2023-06-01): Add support for RTL8723BS SDIO WiFi modules (#2567)

Many x86_64 tablets (e.g. Cherry View) use SDIO WiFi modules, this
enables the driver for a common one I've come across in the wild.

This module requires firmware from the following, which are already
enabled for this platform:

  - BR2_PACKAGE_LINUX_FIRMWARE_RTL_87XX
  - BR2_PACKAGE_LINUX_FIRMWARE_RTL_87XX_BT

fixes #2422
commit fef59845 (2023-06-15): Add support for RTL8723BS SDIO WiFi modules (#2567)

Many x86_64 tablets (e.g. Cherry View) use SDIO WiFi modules, this
enables the driver for a common one I've come across in the wild.

This module requires firmware from the following, which are already
enabled for this platform:

  - BR2_PACKAGE_LINUX_FIRMWARE_RTL_87XX
  - BR2_PACKAGE_LINUX_FIRMWARE_RTL_87XX_BT

fixes #2422