home-assistant/operating-system #2428: Error: Unable to create websockets listener on port 1884

jbouwh, 2023-03-25:
### Describe the issue you are experiencing

The Mosquitto broker add does not start any ,ore after upgrading to OS to 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to OS to 10.0.rc1
2. Disable port 1884 in the addon
3. Restart the addon
...
See log the addon fails to start up. Before the upgrade it ran okay.

### Anything in the Supervisor logs that might be useful for us?

```txt
23-03-25 11:52:23 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Mosquitto broker is stopped, restarting...
23-03-25 11:52:23 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_core_mosquitto application
23-03-25 11:52:25 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-mosquitto with version 6.1.3
23-03-25 11:52:42 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-2289' coro=<Addon.watchdog_container() done, defined at /usr/src/supervisor/supervisor/addons/addon.py:989> exception=AddonsJobError('Rate limit exceeded, more then 10 calls in 0:30:00')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 1003, in watchdog_container
    await self._restart_after_problem(self, event.state)
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 144, in wrapper
    raise on_condition(
supervisor.exceptions.AddonsJobError: Rate limit exceeded, more then 10 calls in 0:30:00
23-03-25 11:52:50 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-03-25 11:52:50 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_appdaemon
23-03-25 11:52:50 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-03-25 11:55:13 WARNING (MainThread) [supervisor.addons.options] Option 'otbr_web_enable' does not exist in the schema for Silicon Labs Multiprotocol (core_silabs_multiprotocol)
23-03-25 12:00:16 WARNING (MainThread) [supervisor.addons.options] Option 'otbr_web_enable' does not exist in the schema for Silicon Labs Multiprotocol (core_silabs_multiprotocol)
23-03-25 12:05:19 WARNING (MainThread) [supervisor.addons.options] Option 'otbr_web_enable' does not exist in the schema for Silicon Labs Multiprotocol (core_silabs_multiprotocol)
23-03-25 12:06:15 WARNING (MainThread) [supervisor.addons.options] Option 'otbr_web_enable' does not exist in the schema for Silicon Labs Multiprotocol (core_silabs_multiprotocol)
23-03-25 12:10:22 WARNING (MainThread) [supervisor.addons.options] Option 'otbr_web_enable' does not exist in the schema for Silicon Labs Multiprotocol (core_silabs_multiprotocol)
23-03-25 12:15:25 WARNING (MainThread) [supervisor.addons.options] Option 'otbr_web_enable' does not exist in the schema for Silicon Labs Multiprotocol (core_silabs_multiprotocol)
23-03-25 12:20:28 WARNING (MainThread) [supervisor.addons.options] Option 'otbr_web_enable' does not exist in the schema for Silicon Labs Multiprotocol (core_silabs_multiprotocol)
23-03-25 12:21:36 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-03-25 12:23:59 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Mosquitto broker is stopped, restarting...
23-03-25 12:23:59 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning addon_core_mosquitto application
23-03-25 12:24:01 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-mosquitto with version 6.1.3
23-03-25 12:24:16 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Mosquitto broker is stopped, restarting...
23-03-25 12:24:16 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning addon_core_mosquitto application
23-03-25 12:24:17 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-mosquitto with version 6.1.3
23-03-25 12:24:33 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Mosquitto broker is stopped, restarting...
23-03-25 12:24:33 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning addon_core_mosquitto application
23-03-25 12:24:34 INFO (SyncWorker_6) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-mosquitto with version 6.1.3
```

### Anything in the Host logs that might be useful for us?

```txt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1808, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1845, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
[... 114 lines cut ...]
services-up: info: copying legacy longrun nginx (no readiness notification)
s6-rc: info: service legacy-services successfully started
[12:35:26] INFO: Starting NGINX for authentication handling...
[12:35:27] INFO: Starting mosquitto MQTT broker...
2023-03-25 12:35:27: Warning: Mosquitto should not be run as root/administrator.
2023-03-25 12:35:27: mosquitto version 2.0.11 starting
2023-03-25 12:35:27: Config loaded from /etc/mosquitto/mosquitto.conf.
2023-03-25 12:35:27: Loading plugin: /usr/share/mosquitto/go-auth.so
2023-03-25 12:35:27:  ├── Username/password checking enabled.
2023-03-25 12:35:27:  ├── TLS-PSK checking enabled.
2023-03-25 12:35:27:  └── Extended authentication not enabled.
2023-03-25 12:35:27: Opening ipv4 listen socket on port 1883.
2023-03-25 12:35:27: Opening ipv6 listen socket on port 1883.
2023-03-25 12:35:27: Opening websockets listen socket on port 1884.
2023-03-25 12:35:27: Error: Unable to create websockets listener on port 1884.
[11:35:27] WARNING: Halt add-on
s6-rc: info: service legacy-services: stopping
[11:35:27] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

owenmcarthur, 2023-03-25:
same problem over here after update.

agners, 2023-03-25:
The culprit seems to be an issue in libwebsockets, and is resolved in newer versions of it, see also https://github.com/home-assistant/addons/issues/2904. There is PR which is fixing this, see https://github.com/home-assistant/addons/pull/2923.

Once Mosquitto add-on has been updated, this should be resolved.

krobit, 2023-03-25:
same problem

airsailor, 2023-03-27:
Same here. This is terrible because 90% of my HA work by mqtt. Now I have nothing. Please someone solve this problem

agners, 2023-03-27:
This is solved with the latest add-on 6.2.0.