home-assistant/operating-system #2430: The NUT addon does not start after upgrade to OS 10.0 - Error: Out of memory

robercy, 2023-03-26:
### Describe the issue you are experiencing

The plugin does not start after upgrade to OS 10.0 - Error: Out of memory

Add-on: Network UPS Tools
Manage battery backup (UPS) devices
Add-on version: 0.11.1
You are running the latest version of this add-on.
System: Home Assistant OS 10.0.rc1 (running Generic x86-64 under Proxmox)
Home Assistant Core: 2023.3.6
Home Assistant Supervisor: 2023.03.2

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start NUT add-on
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[09:20:54] INFO: Setting mode to netserver...
[09:20:54] INFO: Connected USB devices:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 002 Device 002: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[09:20:54] INFO: Generating /etc/nut/upsd.users...
[09:20:54] INFO: Configuring user: admin
[09:20:54] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[09:20:54] INFO: Configuring Device named myups...
[09:20:54] INFO: Starting the UPS drivers...
Using subdriver: APC HID 0.96
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Network UPS Tools - UPS driver controller 2.7.4
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[09:20:55] INFO: Starting the UPS information server...
0.000000 fopen /run/nut/upsd.pid: No such file or directory
0.000329 listening on 0.0.0.0 port 3493
0.000629 Connected to UPS [myups]: usbhid-ups-myups
0.000679 Out of memory: Cannot allocate memory <================================ ERROR
Network UPS Tools upsd 2.7.4
[09:20:55] WARNING: upsd crashed, halting add-on
s6-rc: info: service legacy-services: stopping
[09:20:55] INFO: upsd stopped, restarting...
[09:20:55] INFO: upsmon stopped, restarting...
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stop
```

### Anything in the Host logs that might be useful for us?

```txt
n.a.
```

### System information

## System Information

version | core-2023.3.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.21
arch | x86_64
timezone | America/Fortaleza
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.31.0
Stage | running
Available Repositories | 1317
Downloaded Repositories | 33
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0.rc1
-- | --
update_channel | beta
supervisor_version | supervisor-2023.03.2
agent_version | 1.5.1
docker_version | 23.0.1
disk_total | 30.8 GB
disk_used | 22.5 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.15.0), File editor (5.5.0), Mosquitto broker (6.1.3), Samba share (10.0.0), chrony (2.5.0), Node-RED (14.1.1), SSH & Web Terminal (13.0.3), Network UPS Tools (0.11.1), phpMyAdmin (0.8.5), Studio Code Server (5.5.5), AdGuard Home (4.8.3), motionEye (0.18.0), NGINX Home Assistant SSL proxy (3.2.0), AppDaemon (0.11.0), Let's Encrypt (4.12.8), Nginx Proxy Manager (0.12.3), Home Assistant Google Drive Backup (0.110.3), UniFi Network Application (2.6.0), MariaDB (2.5.2), ZeroTier One (0.16.2), Glances (0.18.0), ESPHome (2023.3.1), JupyterLab (0.10.0), Logspout addon (1.4.3), EMQX (0.2.5), InfluxDB (4.5.0), Uptime Kuma (0.7.0), MQTT Explorer (browser-1.0.1), Grafana (8.2.1), FTP (4.6.3), Matter Server (4.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 20
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 16, 2023 at 22:33
-- | --
current_recorder_run | March 26, 2023 at 09:16
estimated_db_size | 641.95 MiB
database_engine | mysql
database_version | 10.6.10

</details>

GuidoKJ, 2023-03-26:
I can confirm the same issue is going on with OS 10.0 RC1 on generic-x86-64.

agners, 2023-03-27:
This will be fixed in 10.0.rc2 (by #2427).