home-assistant/operating-system #2433: The NUT addon does not start after upgrade to OS 10.0.rc2 - Error: Out of memory

robercy, 2023-03-27:
### Describe the issue you are experiencing

The plugin does not start after upgrade to OS 10.0.rc2- Error: Out of memory

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Operating System 10.0.rc2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start NUT add-on
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[19:15:45] INFO: Setting mode to netserver...
[19:15:45] INFO: Connected USB devices:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 002 Device 002: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[19:15:45] INFO: Generating /etc/nut/upsd.users...
[19:15:45] INFO: Configuring user: admin
[19:15:45] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[19:15:46] INFO: Configuring Device named myups...
[19:15:46] INFO: Starting the UPS drivers...
Using subdriver: APC HID 0.96
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Network UPS Tools - UPS driver controller 2.7.4
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[19:15:46] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000538	listening on 0.0.0.0 port 3493
   0.000897	Connected to UPS [myups]: usbhid-ups-myups
   0.000959	Out of memory: Cannot allocate memory
Network UPS Tools upsd 2.7.4
[19:15:46] WARNING: upsd crashed, halting add-on
s6-rc: info: service legacy-services: stopping
[19:15:46] INFO: upsd stopped, restarting...
[19:15:46] INFO: upsmon stopped, restarting...
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

### Anything in the Host logs that might be useful for us?

```txt
n.a.
```

### System information

## System Information

version | core-2023.3.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.21
arch | x86_64
timezone | America/Fortaleza
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4998
Installed Version | 1.31.0
Stage | running
Available Repositories | 1315
Downloaded Repositories | 33
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0.rc2
-- | --
update_channel | beta
supervisor_version | supervisor-2023.03.3
agent_version | 1.5.1
docker_version | 23.0.1
disk_total | 30.8 GB
disk_used | 26.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.15.0), File editor (5.5.0), Mosquitto broker (6.2.0), Samba share (10.0.0), chrony (2.5.0), Node-RED (14.1.1), SSH & Web Terminal (13.1.0), Network UPS Tools (0.11.1), phpMyAdmin (0.8.5), Studio Code Server (5.5.5), AdGuard Home (4.8.3), motionEye (0.18.0), NGINX Home Assistant SSL proxy (3.2.0), AppDaemon (0.11.0), Let's Encrypt (4.12.8), Nginx Proxy Manager (0.12.3), Home Assistant Google Drive Backup (0.110.3), UniFi Network Application (2.6.0), MariaDB (2.5.2), ZeroTier One (0.16.2), Glances (0.18.1), ESPHome (2023.3.2), JupyterLab (0.10.0), Logspout addon (1.4.3), EMQX (0.2.5), InfluxDB (4.6.0), Uptime Kuma (0.7.1), MQTT Explorer (browser-1.0.1), Grafana (8.2.1), FTP (4.6.3), Matter Server (4.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 20
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 17, 2023 at 09:31
-- | --
current_recorder_run | March 27, 2023 at 19:15
estimated_db_size | 647.92 MiB
database_engine | mysql
database_version | 10.6.10

</details>

agners, 2023-03-27:
Hm, I thought that is the same issue we've been seeing with Node RED (see https://github.com/hassio-addons/addon-node-red/issues/1562#issuecomment-1483750464). However, on the platforms I had that issue (and it did work with HAOS 9.x with zram previously), PR #2427 fixed the problem.

How much memory do you have? Did that add-on work before using HAOS 9.5?

robercy, 2023-03-27:
Hi, agners.
I use nodered too, and it's Ok.
Everything works fine on HAOS 9.x, and the trouble began with the new HAOS 10.0. 
My setup run on top of Proxmox with 10GB of RAM and 8 cpus allocated to homeassistant. 
Unfortunatly, the PR 2427 didn't fix this problem.
👍🏾

agners, 2023-03-27:
> My setup run on top of Proxmox with 10GB of RAM and 8 cpus allocated to homeassistant.

In that case you should be running the `ova` image type.

Also, from the system console, can you run `login`, and then `systemctl status haos-swapfile`?

robercy, 2023-03-27:
> What is `free -h` and `swapon` on you system saying?

root@pve ~# free -h
               total        used        free      shared      buff/cache   available
Mem:            15Gi       7.5Gi       5.6Gi        82Mi       2.3Gi       7.5Gi
Swap:          8.0Gi          0B       8.0Gi
root@pve ~# swapon
NAME      TYPE      SIZE USED PRIO
/dev/dm-0 partition   8G   0B   -2

agners, 2023-03-27:
HAOS 10.x moved from zram to zswap. It seems that zswap creation failed in your cases.

It also seems you are using additional swap through some method?

robercy, 2023-03-27:
Hi. I'm using the standard Proxmox swap configuration. Can you help me to switch to zswap ?
Thanks in advance.

gateriderz, 2023-03-28:
Same here no zswap, only zram
[image]

agners, 2023-03-28:
> Same here no zswap, only zram [image]

That does not seem to be HAOS 10.0.rc2 then? What is `ha os info` saying?

agners, 2023-03-28:
> Hi. I'm using the standard Proxmox swap configuration. Can you help me to switch to zswap ? I'm googling ! Thanks in advance.

Hm, your edited last message does look good though: It seems the system is now using `zswap`. I wonder why NUT is complaining here, you have 8GB of available RAM! :thinking:

agners, 2023-03-28:
My best guess is currently that this is caused by the Docker 23.0.1 update.

To rule out zswap, can you try the development build just before we moved to zswap, `10.0.dev20230317`?

Use the following sequence of commands:

```
ha su options --channel=dev
ha su reload
ha su update
ha os update --version=10.0.dev20230317
```

After that, change back to the beta channel and check if update worked:
```
ha su options --channel=dev
ha su reload
ha os info
```

Now `swapon` should show zram again. Does the add-on start in this configuration?

robercy, 2023-03-28:
Hello.

On both situation, still the same problem.

---------------------------------------------------------------------------------------------------------------
~ # login
a0d7b954-ssh login: ha su options --channel
Login timed out after 60 seconds
~ # ha su options --channel=dev
Command completed successfully.
~ # ha su reload
Command completed successfully.
~ # ha su update
Processing... Done.

Error: No supervisor update available - 2023.03.3
~ # ha os update --version=10.0.dev20230317
Processing... Done.

Command completed successfully.

Starting NUT with log_level=debug set ------------------------------------------------------------------------------------------------

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[06:01:54] INFO: Setting mode to netserver...
[06:01:54] INFO: Connected USB devices:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 002 Device 002: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
[06:01:54] INFO: Generating /etc/nut/upsd.users...
[06:01:54] INFO: Configuring user: admin
[06:01:55] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[06:01:55] INFO: Configuring Device named myups...
[06:01:55] INFO: Starting the UPS drivers...
Using subdriver: APC HID 0.96
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Network UPS Tools - UPS driver controller 2.7.4
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
s6-rc: info: service legacy-services successfully started
[06:01:56] INFO: Starting the UPS information server...
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000339	listening on 0.0.0.0 port 3493
   0.000565	Connected to UPS [myups]: usbhid-ups-myups
   0.000614	Out of memory: Cannot allocate memory
Network UPS Tools upsd 2.7.4
[06:01:56] WARNING: upsd crashed, halting add-on
[06:01:56] INFO: upsd stopped, restarting...
s6-rc: info: service legacy-services: stopping
[06:01:56] INFO: upsmon stopped, restarting...
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
[... 43 lines cut ...]
cont-init: info: /etc/cont-init.d/nut.sh exited 0
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun upsd (no readiness notification)
services-up: info: copying legacy longrun upsmon (no readiness notification)
[06:09:52] INFO: Starting the UPS information server...
s6-rc: info: service legacy-services successfully started
   0.000000	fopen /run/nut/upsd.pid: No such file or directory
   0.000409	listening on 0.0.0.0 port 3493
   0.000655	Connected to UPS [myups]: usbhid-ups-myups
   0.000715	Out of memory: Cannot allocate memory
Network UPS Tools upsd 2.7.4
[06:09:52] WARNING: upsd crashed, halting add-on
s6-rc: info: service legacy-services: stopping
[06:09:52] INFO: upsd stopped, restarting...
[06:09:52] INFO: upsmon stopped, restarting...
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

agners, 2023-03-28:
Can you type `login` and then `sysctl -w vm.overcommit_memory=1` and see if that helps?

It seems that `upsd` allocates an excessive amount of memory. The newer kernel seem to behave a bit different when it comes to memory overcommit heuristic, in that case it seems to be problematic.

agners, 2023-03-28:
> Hi, dear Agners.
> 
> Now, everything is Ok. !! Congratulations !!
> 
> One more question, this command persist between reboot ?
> 
> A thousands Thank YOU !! You rocks !

No, it does not, unfortunately :cry: 

We'll have to figure out what/why upsd is allocating that much memory. I currently don't have a good idea how to do that, it requires quite some hacking inside the add-on, either using strace or gdb to find what is causing this big allocation.

robercy, 2023-03-28:
Just for information, Still not running on HAOS 10.0.rc3.

agners, 2023-03-29:
> Just for information, Still not running on HAOS 10.0.rc3.

10.0.rc3 is not out yet? You mean the latest nightly `11.0.dev20230328`?

Can you share the output of `dmesg` right after a failed start attempt of NUT (run it in the root shell on the connsole)?

agners, 2023-03-29:
I've been able to work around on add-on level, so that particular add-on should be working with the latest release.

I created a new issue to trying to understand what changed in HAOS with #2438.

closed 2023-03-29 by agners