home-assistant/operating-system #2437: CPU frequentie pinned at max upgrade to OS 10.0.rc2

RemCom, 2023-03-28:
### Describe the issue you are experiencing

CPU frequentie of Intel NUC pinned at max after update while before was correcly going between max and lowest frequentie. CPU usage in percentage is still ate the same level as before the update

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0.rc2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Just do the update

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

RemCom, 2023-03-28:
Before vertical red line before update. After redline after update

[image]

agners, 2023-03-29:
Is the CPU load still low or also higher with 10.0.rc2?

I am not aware in any change which specifically could explain this. We use a newer Linux kernel which could change behavior, but such a drastic difference is somewhat unexpected.

RemCom, 2023-03-29:
The load is actually the same. Also system power draw is the same so I can't really explain it. Maybe it's just an issue with monitoring?

agners, 2023-03-30:
> The load is actually the same. Also system power draw is the same so I can't really explain it. Maybe it's just an issue with monitoring?

Hm yeah maybe, or the new kernel's cpufreq governor just decides its not worth down-clocking.

If you have OS shell access, you can check the following files on what is currently configured:

```
# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq
# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq
```

RemCom, 2023-03-31:
```
cat /sys/devices/system/cpu/cpufreq/policy0/scaling_governor: schedutil
cat /sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq: 800000
cat /sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq: 2800000
```

Frequentie values look oke. maybe it default governor state is different are maybe just other behaviour the before

closed 2023-04-03 by agners

nepozs, 2023-04-04:
I've downgraded to HAOS 9.5, but some day it was working using 10.0_rc1 and 10.0_rc2
[image]

Old AMD machine - looks similar, but temperature was affected too, so I'm sure the power draw is higher.

agners, 2023-04-04:
@nepozs your case seems different as the CPU load (Processor uses in %) is higher.

Can you open a new issues with all the system details? Also, can you check what component is causing the higher CPU load (e.g. by comparing what is shown in the Glances add-on between 10.0.rc3 and 9.5).

nepozs, 2023-04-04:
I've found reason - it is motionEye not working properly after system update.
It is just experimental installation, but I don't have enough free time lately, so I've missed problems with motionEye.

agners, 2023-04-04:
If motionEye has some OS dependency which needs update then most likely this issue should go to https://github.com/hassio-addons/addon-motioneye/issues.