home-assistant/operating-system #2445: Out of Memory errors since 10.0.rc2 upgrade

benjaminarthurt, 2023-04-03:
### Describe the issue you are experiencing

OS Runs well after update for about an hour, but starts showing "out of memory: killed process xxx (node)" errors without active use.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0.rc2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade working environment to 10.0.rc2
2. Wait Web console becomes unresponsive after about an hour. "out of memory" errors appear in console

### Anything in the Supervisor logs that might be useful for us?

```txt
No Errors visible, but unable to access logs until after a hard reboot.
```

### Anything in the Host logs that might be useful for us?

```txt
No Errors visible, but unable to access logs until after a hard reboot.
```

### System information

## System Information

version | core-2023.4.0b4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.21
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.31.0
Stage | running
Available Repositories | 1312
Downloaded Repositories | 16
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | June 15, 2023 at 8:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0.rc2
-- | --
update_channel | beta
supervisor_version | supervisor-2023.03.3
agent_version | 1.5.1
docker_version | 23.0.1
disk_total | 109.3 GB
disk_used | 13.2 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Uptime Kuma (0.7.1), ESPHome (2023.3.2), Studio Code Server (5.5.6), motionEye (0.18.0), Terminal & SSH (9.6.1), SSH & Web Terminal (13.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 2
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 22, 2023 at 9:22 PM
-- | --
current_recorder_run | April 2, 2023 at 12:08 PM
estimated_db_size | 493.19 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

Running on an AtomicPi with External SSD, had been stable until 10.0.rc2 update.
Have attempted to disable several addons to identify what may be contributing to this. I noticed other reports of "out of memory" errors for 10.0.rc1 that were reported to be fixed with rc2.

agners, 2023-04-03:
This is most likely related to #2438.

> Wait Web console becomes unresponsive after about an hour. "out of memory" errors appear in console

How much memory does your system have in general?

It is likely related to one of the add-ons. Can you check which add-on stopped after the error? From the error it seems a `node` (NodeJS?) process got killed. Maybe "Uptime Kuma" is the culprit?

agners, 2023-04-03:
> I had been running MotionEye as well when this started and had disabled that already as I suspected it was possibly the cause, however have had one instance of HA going non-responsive since removing that.

The "non-responsive" part is mainly because of thrashing: When the system comes close to be out-of-memory, the kernel tries very hard to free up some memory.

The upcoming 10.0.rc3 will have mitigation for that. I expect it will still cause Out of Memory errors, but resolve them quite quickly.

agners, 2023-04-06:
@benjaminarthurt have you had a chance to test 10.0.rc3 and/or without Uptime add-on?

benjaminarthurt, 2023-04-06:
I have done both at this point and it's been stable as normal. I'm planning on re-enabling motionEye as a test, but I'm fairly confident this was caused by Uptime/2gb memory on my board.

agners, 2023-04-14:
I'll close this issue as it seems to be load related.

closed 2023-04-14 by agners

agners, 2023-04-18:
Most likely related to https://github.com/hassio-addons/addon-motioneye/issues/394.