home-assistant/operating-system #2454: Link-local addresses (APIPA) support

andrein, 2023-04-06:
### Describe the issue you are experiencing

I have this issue that's driving me nuts: whenever there's a power outage, I can no longer open the gate.

The gate is controlled via the the [homekit controller](https://www.home-assistant.io/integrations/homekit_controller/), using a [Nice IT4WIFI](https://www.niceforyou.com/en/it4wifi) interface.

When there's a power outage, my DHCP server, switch and wifi access points shut down, but the gate stays on because it has a battery backup. When the power comes back, the access points usually start up before the DHCP server does, the homekit interface connects to it, but can't get a DHCP lease, so it autoconfigures a link-local address in the 169.254.0.0/16 range. I've confirmed this using a dns-sd:

```
❯ dns-sd -Z _hap._tcp local.
[...]
_hap._tcp                                       PTR     NiceIT4WIFI4A88B4._hap._tcp
NiceIT4WIFI4A88B4._hap._tcp                     SRV     0 0 80 NiceIT4WIFI-hap-4A88B4.local. ; Replace with unicast FQDN of target host
NiceIT4WIFI4A88B4._hap._tcp                     TXT     "id=51:60:8D:1C:F9:1A" "md=Nice Spa - IT4WIFI - 575x" "pv=1.0" "s#=1" "c#=9" "sf=0" "ff=1" "ci=4"

❯ dns-sd -G v4 NiceIT4WIFI-hap-4A88B4.local
DATE: ---Thu 06 Apr 2023---
22:40:39.582  ...STARTING...
Timestamp     A/R  Flags         IF  Hostname                               Address                                      TTL
22:40:39.583  Add  40000002      15  NiceIT4WIFI-hap-4A88B4.local.          169.254.154.223                              255
```

I can access the homekit interface via HTTP from both phone and laptop, using the IP or hostname, but Home Assistant can't.
I wasn't able to confirm this yet, but I believe that Home Assistant OS doesn't have any zeroconf routes set up. 
`ip route` doesn't give any output, `netstat` doesn't understand the `-r` flag, and the `route` command is not installed. (could this also be a bug in `ip route`?)

```
# ip route
# netstat -rn
netstat: invalid option -- 'r'
# route
-sh: route: not found
```

It's been a while since I used linux on a desktop, but I believe the avahi service is responsible with configuring that route, but it looks like it's not enabled in Home Assistant Operating System.

I found some MRs  removing avahi config files (#1279, #1280), but not much context on why avahi was removed in the first place.

Note that this didn't happen when the homekit interface was bound directly to my iPad, it started happening after I started using the Homekit Controller component.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. kill DHCP server
2. connect device to home assistant network
3. ping device from home assistant, notice it doesn't reply.

### Anything in the Supervisor logs that might be useful for us?

```txt
Not really.
```

### Anything in the Host logs that might be useful for us?

```txt
Not really.
```

### System information

## System Information

version | core-2023.4.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.15.84-v8
arch | aarch64
timezone | Europe/Bucharest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 2729
Installed Version | 1.30.1
Stage | running
Available Repositories | 1248
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 28.0 GB
disk_used | 11.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | AirCast (3.5.2), Node-RED (14.1.3), ESPHome (2023.3.2), Terminal & SSH (9.6.1), Home Assistant Google Drive Backup (0.110.3), Let's Encrypt (4.12.8), NGINX Home Assistant SSL proxy (3.2.0), eBUSd (23.1), Mosquitto broker (6.2.0), Grafana (8.2.1), Studio Code Server (5.5.6)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 31 March 2023 at 08:51
-- | --
current_recorder_run | 6 April 2023 at 23:49
estimated_db_size | 213.02 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2023-04-14:
From what I understand it is not common to use [IPv4LL](https://en.wikipedia.org/wiki/Link-local_address#IPv4) (same as APIPA, but the standardized name from what I understand) along with static IP configuration or DHCP assigned addresses (in IPv4 world). Depending on lease time etc, it might be that your gate jumps to IPv4LL while HA does not (yet). Also are you actually using DHCP on your Home Assistant installation?

agners, 2023-04-14:
I've researched a bit into this issue from NetworkManager side: We set `ipv4.method` to `auto` by default, which configures the interface to DHCP. The `ipv4.link-local` setting is left to `default`, which in turn means `auto`, which in turn means that link-local is only enabled if `ipv4.method` is `link-local` (which isn't the case).

You can enable `ipv4.link-local` by using `nmcli` in the console directly using:

```
nmcli con modify "Supervisor eth0" ipv4.link-local enabled
```
(check name of your connection with `nmcli con show`)

I am a bit hesitant to enable this by default, but maybe we could add it as a setting to the IPv4 network settings.

See also: https://networkmanager.dev/docs/libnm/latest/NMSettingIP4Config.html#NM-SETTING-IP4-CONFIG-METHOD-LINK-LOCAL:CAPS

andrein, 2023-04-14:
Hi @agners, thanks for taking the time to look into this.

I'm not suggesting Home Assistant should start using  link local addresses if DHCP fails, that would be super confusing for everyone. What I'm suggesting is allowing home assistant to talk to ~broken~ devices (like my gate controller) that are using link-local addresses for various reasons.

> You can enable `ipv4.link-local` by using `nmcli` in the console directly using:
> 
> ```
> nmcli con modify "Supervisor eth0" ipv4.link-local enabled
> ```
> 
> (check name of your connection with `nmcli con show`)

I've tried this, just to confirm that my laptop can still talk to home assistant after it starts using link-local addressing but it looks like it's not supported.

```
# nmcli con show
NAME             UUID                                  TYPE      DEVICE
Supervisor eth0  cf3cbb2e-cf0c-3056-9cdf-bc2728a27ef8  ethernet  eth0
# nmcli con modify "Supervisor eth0" ipv4.link-local enabled
Error: invalid property 'link-local': 'link-local' not among [method, dns, dns-search, dns-options, dns-priority, addresses, gateway, routes, route-metric, route-table, routing-rules, ignore-auto-routes, ignore-auto-dns, dhcp-client-id, dhcp-iaid, dhcp-timeout, dhcp-send-hostname, dhcp-hostname, dhcp-fqdn, dhcp-hostname-flags, never-default, may-fail, required-timeout, dad-timeout, dhcp-vendor-class-identifier, dhcp-reject-servers].```
```

However, I did notice that HA has link-local IPv6 addresses enabled by default (fe80::98ea:4abd:51ae:1613/64 in this case):

```
# ip addr show dev eth0
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether e4:5f:01:ec:1a:0b brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.250/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 70426sec preferred_lft 70426sec
    inet6 xxxx:xxxx:xxxx:xxxx:d0fd:1c1d:920b:c7d7/64 scope global dynamic noprefixroute
       valid_lft 86255sec preferred_lft 14255sec
    inet6 fe80::98ea:4abd:51ae:1613/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
```

I've checked my MacBook and a linux server, they both have the link-local route set up by default:

MacBook:
```
❯ netstat -rn | grep 169.254
169.254            link#15            UCS               en0      !
169.254.169.254    link#15            UHRLSW            en0      !

❯ ifconfig | grep 169.254 

# notice there's no IPv4LL address assigned, but I can still access the gate http server when it glitches, while Home Assistant can't
```

Linux server (it's not even using network-manager, just plain old init scripts:
```
$ ip route
default via xxx.xxx.xxx.xxx dev eth0
xxx.xxx.xxx.0/24 dev eth0 proto kernel scope link src xxx.xxx.xxx.173
169.254.0.0/16 dev eth0 scope link metric 1002
```

This is a CentOS 7 box, so it was easier to figure out [who's adding the route](https://github.com/fedora-sysv/initscripts/blob/main/network-scripts/ifup-eth#L345-L347)

I'm going to try to make networkmanager add the route in the least invasive way, there's probably some setting somewhere, but the internet isn't very helpful since most searches show people are trying to get rid of that route :)

andrein, 2023-04-14:
I stumbled upon [RFC3927](https://www.rfc-editor.org/rfc/rfc3927) and I believe [section 2.6.2](https://www.rfc-editor.org/rfc/rfc3927#section-2.6.2) describes exactly what needs to be done to fix my issue:

>   In many network stacks, achieving this functionality may be as simple
>   as adding a routing table entry indicating that 169.254/16 is
>   directly reachable on the local link.  This approach will not work
>   for routers or multi-homed hosts.  Refer to [section 3](https://www.rfc-editor.org/rfc/rfc3927#section-3) for more
>   discussion of multi-homed hosts.```

Also, [this debian bug report](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=780872) suggests that avahi-autoipd handles the link-local routes.

I couldn't find the sources, but this is the script that debian uses (note that neither `route`, nor `ip route` work on my home assistant instance, maybe this deserves another bug report?)

```
root@bbf4ca594a94:/# cat /etc/network/if-up.d/avahi-autoipd
#!/bin/sh

set -e

# Description:      Add routes to allow communication between machines which
#                   only have an IPv4LL address assigned and those which only
#                   have a routable address assigned.
#
#                   See http://developer.apple.com/qa/qa2004/qa1357.html for
#                   more information.

[ -x /usr/sbin/avahi-autoipd ] || exit 0

[ "$IFACE" != "lo" ] || exit 0
case "$ADDRFAM" in
        inet) ;;
        *) exit 0 ;;
esac

case "$METHOD" in
        static|dhcp|NetworkManager) ;;
        *) exit 0 ;;
esac

if [ -x /bin/ip ]; then
        # route already present?
        ip route show | grep -q '^169.254.0.0/16[[:space:]]' && exit 0

        /bin/ip route add 169.254.0.0/16 dev $IFACE metric 1000 scope link
elif [ -x /sbin/route ]; then
        # route already present?
        /sbin/route -n | egrep -q "^169.254.0.0[[:space:]]" && exit 0

        /sbin/route add -net 169.254.0.0 netmask 255.255.0.0 dev $IFACE metric 1000
fi
```

andrein, 2023-04-27:
I just had another power outage today.

Revisiting this on Home Assistant OS 10.0, the `ip route` works as expected and adding the `169.254.0.0/16` route allows home assistant to communicate with the gate.

```
# ip route
default via 192.168.1.254 dev eth0 proto dhcp src 192.168.1.250 metric 100
172.30.32.0/23 dev hassio proto kernel scope link src 172.30.32.1
172.30.232.0/23 dev docker0 proto kernel scope link src 172.30.232.1
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.250 metric 100

# ping 169.254.95.56
No response from 169.254.95.56

# ip route add 169.254.0.0/16 dev eth0 metric 1000 scope link

# ip route
default via 192.168.1.254 dev eth0 proto dhcp src 192.168.1.250 metric 100
169.254.0.0/16 dev eth0 scope link metric 1000
172.30.32.0/23 dev hassio proto kernel scope link src 172.30.32.1
172.30.232.0/23 dev docker0 proto kernel scope link src 172.30.232.1
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.250 metric 100

# ping 169.254.95.56
169.254.95.56 is alive!
```

Homeassistant still can't talk to the device because it's trying to reach it on the old IP address. I'll have to look into why it's doing that instead of using the IP available via mDNS (maybe because avahi is missing?)

agners, 2023-04-28:
> I'm going to try to make networkmanager add the route in the least invasive way, there's probably some setting somewhere, but the internet isn't very helpful since most searches show people are trying to get rid of that route :)

Yeah, the typical "I don't understand it so it must be bad" problem :sweat: 

However, I am a bit unsure if simply adding the IPv4 link-local route without having a link-local IPv4 address is a "proper" operation mode. It seems that using the routable address (which I guess the typical private range address which a DHCP server is handing out can be considered routable) as source address is possible and even preferred, if available, see [RFC3927 Section 2.6.2](https://www.rfc-editor.org/rfc/rfc3927#section-2.6.2).

> Homeassistant still can't talk to the device because it's trying to reach it on the old IP address. I'll have to look into why it's doing that instead of using the IP available via mDNS (maybe because avahi is missing?)

Home Assistant Core might be doing its own mDNS lookup, at least there is an zeroconf component. It seems to also filter link-local addresses, see:
https://github.com/home-assistant/core/blob/2023.4.6/homeassistant/components/zeroconf/__init__.py#L571-L573

In the end this might be related to the fact that currently link-local addressing doesn't work with Home Assistant OS.

closed 2023-07-31 by github-actions[bot]