home-assistant/operating-system #2456: Missing firmware rtl_nic/rtl8153a-3.fw

Saoclyph, 2023-04-09:
### Describe the issue you are experiencing

I bought a rtl8153-based usb to ethernet adapter for my odroid-n2 Home Assistant because the integrated NIC is dead...
The network connection is working with this adapter, but the speed is capped to 100Mb instead of 1Gb. 

Checking dmesg, I saw that it was failing to load the firmware `rtl_nic/rtl8153a-3.fw`, which may be the reason for the capped speed.

This firmware is missing from the rootfs, to add it `BR2_PACKAGE_LINUX_FIRMWARE_RTL_815X` need to be added to the [configs files](https://github.com/home-assistant/operating-system/tree/dev/buildroot-external/configs), cf. [buildroot/linux-firmware.mk](https://github.com/buildroot/buildroot/blob/00709af66ec7056ff76a6238f22985cfa0bcc3a6/package/linux-firmware/linux-firmware.mk#L569)

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Plug in a rtl8153-based usb to ethernet adapter
2. Boot the home-assistant machine
3. Check dmesg to see the failure to load the firmware rtl_nic/rtl8153a-3.fw

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
[    4.129158] r8152 2-1.2:1.0: Direct firmware load for rtl_nic/rtl8153a-3.fw failed with error -2
[    4.129168] r8152 2-1.2:1.0: unable to load firmware patch rtl_nic/rtl8153a-3.fw (-2)
```

### Additional information

$ lsmod | grep r8
r8153_ecm              16384  0
cdc_ether              20480  1 r8153_ecm
usbnet                 32768  2 r8153_ecm,cdc_ether
r8152                 106496  1 r8153_ecm

agners, 2023-04-12:
I've added the firmware to the dev branch and will backport to OS 10.

ackpengyao, 2024-06-05:
i have two question: