home-assistant/operating-system #2462: OVA Image Bluetooth support

GamerClassN7, 2023-04-12:
### Describe the issue you are experiencing

Hello i recently switched from prox mox to esxi and after instaling from scratch nev version of HA ova Image i am unable to get on board BT card (Intel Wireless-AC 9560) working even that prewiouselly it was working on proxmox, with OS updated to same version as is to my knowledge in new ova file.

Also sometimes HA Crash right after dialog tahat no adapter was found.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

core-2023.4.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Import OVA file
2. add PCI Device
3. try to add BT integration
...

### Anything in the Supervisor logs that might be useful for us?

```txt
just this error

23-04-12 08:48:30 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-1469' coro=<Addon.watchdog_container() done, defined at /usr/src/supervisor/supervisor/addons/addon.py:989> exception=AddonsJobError('Rate limit exceeded, more then 10 calls in 0:30:00')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/addons/addon.py", line 1003, in watchdog_container
    await self._restart_after_problem(self, event.state)
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 144, in wrapper
    raise on_condition(
supervisor.exceptions.AddonsJobError: Rate limit exceeded, more then 10 calls in 0:30:00
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing i could found.
```

### System information

<html>
<body>
<h2 id="title" class="mdc-dialog__title" style="-webkit-font-smoothing: antialiased; font-family: var(--mdc-typography-headline6-font-family,var(--mdc-typography-font-family,Roboto,sans-serif)); font-size: var(--mdc-typography-headline6-font-size,1.25rem); line-height: var(--mdc-typography-headline6-line-height,2rem); font-weight: var(--mdc-typography-headline6-font-weight,500); letter-spacing: var(--mdc-typography-headline6-letter-spacing,0.0125em); text-decoration: var(--mdc-typography-headline6-text-decoration,inherit); text-transform: var(--mdc-typography-headline6-text-transform,inherit); display: block; position: relative; flex-shrink: 0; box-sizing: border-box; margin: 0px 0px 1px; padding: 24px 24px 0px; color: var(--mdc-dialog-heading-ink-color,rgba(0,0,0,0.87));"><div class="header_title" style="margin-right: 32px; margin-inline-end: 32px; margin-inline-start: initial; direction: var(--direction);">System information</div><ha-icon-button dialogaction="close" class="header_button" style="display: inline-block; outline: 0px; position: absolute; right: 16px; top: 14px; text-decoration: none; color: inherit; inset-inline-start: initial; inset-inline-end: 16px; direction: var(--direction);"><mwc-icon-button title="Close" style="display: inline-block; outline: 0px; --mdc-ripple-color:currentcolor; -webkit-tap-highlight-color: transparent; vertical-align: top; --mdc-theme-on-primary:currentColor; --mdc-theme-text-disabled-on-light:var(--disabled-text-color);"><button class="mdc-icon-button mdc-icon-button--display-flex" aria-label="Close" style="font-size: 24px; width: var(--mdc-icon-button-size,48px); height: var(--mdc-icon-button-size,48px); padding: calc((var(--mdc-icon-button-size,48px) - var(--mdc-icon-size,24px))/ 2); display: inline-flex; position: relative; box-sizing: border-box; border: none; outline: 0px; background-color: transparent; fill: currentcolor; color: inherit; text-decoration: none; cursor: pointer; user-select: none; z-index: 0; overflow: visible; align-items: center; justify-content: center; vertical-align: top;"><span><slot><ha-svg-icon style="display: block; align-items: center; justify-content: center; position: relative; vertical-align: middle; fill: currentcolor; width: var(--mdc-icon-size,24px); height: var(--mdc-icon-size,24px);"><svg preserveAspectRatio="xMidYMid meet" focusable="false" role="img" aria-hidden="true" viewBox="0 0 24 24"><g><path d="M19,6.41L17.59,5L12,10.59L6.41,5L5,6.41L10.59,12L5,17.59L6.41,19L12,13.41L17.59,19L19,17.59L13.41,12L19,6.41Z"></path></g></svg></ha-svg-icon></slot></span></button></mwc-icon-button></ha-icon-button></h2><div id="content" class="mdc-dialog__content" style="-webkit-font-smoothing: antialiased; font-family: var(--mdc-typography-body1-font-family,var(--mdc-typography-font-family,Roboto,sans-serif)); font-size: 14px; line-height: var(--mdc-typography-body1-line-height,1.5rem); font-weight: 400; letter-spacing: var(--mdc-typography-body1-letter-spacing,0.03125em); text-decoration-line: ; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; text-transform: none; flex-grow: 1; box-sizing: border-box; margin: 0px; overflow: auto; color: rgb(225, 225, 225); padding: var(--dialog-content-padding,24px); position: var(--dialog-content-position,relative); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px;"><slot id="contentSlot" style="margin-top: 0px; margin-bottom: 0px;"><div><div>

Version | core-2023.4.2
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.10.10
Operating System Family | Linux
Operating System Version | 5.15.90
CPU Architecture | x86_64
Timezone | Europe/Prague
Configuration Directory | /config

</div></div></slot></div>
</body>
</html>

agners, 2023-04-12:
All ova images get built once, and are then completely the same (we repack it to different format, but the image itself is the same). So this is likely an issue with your hypervisor.

Do you get errors in the kernel logs (Host logs in the frontend?)

agners, 2023-04-12:
> ### What version of Home Assistant Operating System is installed?
> 
> core-2023.4.2

This is the Home Assistant version, not the OS version.

GamerClassN7, 2023-04-13:
> O
Sorry my mistake
[image]

GamerClassN7, 2023-04-13:
> Do you get errors in the kernel logs (Host logs in the frontend?)

No nothing suspicious otherwise i would post it in issue summary as specified in form :)

agners, 2023-04-14:
Can you run a couple of commands on the shell?

```
ha > login
# bluetoothctl list
# dmesg | grep Bluetooth
# dmesg | grep iwlwifi
```

GamerClassN7, 2023-04-16:
Yes tomorow when i will be on site

GamerClassN7, 2023-04-24:
Hi after recent update to 10.0 i am finally able to see on board intel adapter, Sorry for later response :) issue can be closed now :)