home-assistant/operating-system #2476: CPU usage bumped from 2% to 10 after Operating System 10.0 upgrade

elmr91, 2023-04-18:
### Describe the issue you are experiencing

I have juste upgraded my proxmox HAOS VM to OS 10
I immediately noticed CPU usage raising from around 2% to 10% after upgrade.

"docker stats" shows a normal container usage / nearly no load.
``
CONTAINER ID   NAME                      CPU %     MEM USAGE / LIMIT     MEM %     NET I/O           BLOCK I/O        PIDS
4448c00d8d7e   addon_core_configurator   0.03%     30.43MiB / 1.925GiB   1.54%     892kB / 843B      23.9MB / 307kB   8
01cb0b8343ce   addon_core_ssh            0.00%     27.51MiB / 1.925GiB   1.40%     943kB / 31.7kB    25.4MB / 393kB   12
7920e2a0c17a   hassio_multicast          0.20%     848KiB / 1.925GiB     0.04%     0B / 0B           61.4kB / 102kB   4
f7aefac69874   hassio_audio              0.00%     23.89MiB / 1.925GiB   1.21%     911kB / 0B        21.3MB / 315kB   13
1d6b8aefd115   hassio_dns                0.00%     29.41MiB / 1.925GiB   1.49%     935kB / 26.7kB    24.8MB / 106kB   11
a4872c2f1be6   hassio_cli                0.00%     13.92MiB / 1.925GiB   0.71%     917kB / 4.07kB    13.2MB / 283kB   9
04f375c2f3cd   hassio_supervisor         0.00%     106.5MiB / 1.925GiB   5.40%     1.31MB / 1.07MB   61.5MB / 958kB   24
13b89612a676   homeassistant             0.80%     354.9MiB / 1.925GiB   18.00%    0B / 0B           203MB / 116MB    33
47dda87b99d2   hassio_observer           0.00%     13.68MiB / 1.925GiB   0.69%     927kB / 9.84kB    12MB / 106kB     8 
 ``

"top" shows containerd is using a consistent 6-8% CPU (this is the only process using significant CPU load)
``
   349 root      20   0 1394.2m  43.1m   7.2   2.2   5:04.68 S  `- /usr/bin/containerd
``

I rebooted the VM, but CPU load stays the same:  
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Install 9.5 ova image in proxmox
2.Upgrade to Operating System 10
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-04-18 18:22:24 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
23-04-18 18:22:24 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
23-04-18 18:22:24 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
23-04-18 18:22:24 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
23-04-18 18:22:24 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
23-04-18 18:22:24 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
23-04-18 18:22:24 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
23-04-18 18:22:24 INFO (MainThread) [supervisor.host.services] Updating service information
23-04-18 18:22:24 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
23-04-18 18:22:24 INFO (MainThread) [supervisor.host.network] Updating local network information
23-04-18 18:22:24 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
23-04-18 18:22:24 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
23-04-18 18:22:24 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2022.11.0
23-04-18 18:22:24 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
23-04-18 18:22:24 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_cli application
23-04-18 18:22:25 INFO (SyncWorker_1) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2022.11.0 - 172.30.32.5
23-04-18 18:22:25 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2022.04.1
23-04-18 18:22:25 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
23-04-18 18:22:25 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_dns application
23-04-18 18:22:25 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2022.04.1 - 172.30.32.3
23-04-18 18:22:25 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
[... 242 lines cut ...]
version_api | ok
installed_addons | Terminal & SSH (9.6.1), File editor (5.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12 avril 2023 à 19:39
-- | --
current_recorder_run | 18 avril 2023 à 18:22
estimated_db_size | 179.41 MiB
database_engine | sqlite
database_version | 3.38.5

[</details>`](url)

agners, 2023-04-18:
On my machine with KVM hypervisor (which is also used by Proxmox) containerd uses about 0.7% on HAOS 10. But I am running on a rather new AMD CPU.

What type of system are you running on?

Is maybe `journalctl -f -u containerd` showing something?

elmr91, 2023-04-18:
No particular log since boot:
`Apr 18 16:22:27 homeassistant containerd[349]: time="2023-04-18T16:22:27.878453230Z" level=info msg="starting signal loop" namespace=moby path=/run/containerd/io.containerd.runtime.v2.task/moby/585c29bcc7be293d23741540dbb4cc2af1bd3a22b3685bc4f02f04248e9b9f09 pid=2290 runtime=io.containerd.runc.v2
Apr 18 16:22:33 homeassistant containerd[349]: time="2023-04-18T16:22:33.269643073Z" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Apr 18 16:22:33 homeassistant containerd[349]: time="2023-04-18T16:22:33.269729478Z" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Apr 18 16:22:33 homeassistant containerd[349]: time="2023-04-18T16:22:33.269749452Z" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Apr 18 16:22:33 homeassistant containerd[349]: time="2023-04-18T16:22:33.270792200Z" level=info msg="starting signal loop" namespace=moby path=/run/containerd/io.containerd.runtime.v2.task/moby/13b89612a676908649a9c53efe151781e6267309bfde99934415abd8ec969c83 pid=2520 runtime=io.containerd.runc.v2
Apr 18 16:23:09 homeassistant containerd[349]: time="2023-04-18T16:23:09.585064770Z" level=info msg="loading plugin \"io.containerd.event.v1.publisher\"..." runtime=io.containerd.runc.v2 type=io.containerd.event.v1
Apr 18 16:23:09 homeassistant containerd[349]: time="2023-04-18T16:23:09.585120961Z" level=info msg="loading plugin \"io.containerd.internal.v1.shutdown\"..." runtime=io.containerd.runc.v2 type=io.containerd.internal.v1
Apr 18 16:23:09 homeassistant containerd[349]: time="2023-04-18T16:23:09.585137376Z" level=info msg="loading plugin \"io.containerd.ttrpc.v1.task\"..." runtime=io.containerd.runc.v2 type=io.containerd.ttrpc.v1
Apr 18 16:23:09 homeassistant containerd[349]: time="2023-04-18T16:23:09.585665355Z" level=info msg="starting signal loop" namespace=moby path=/run/containerd/io.containerd.runtime.v2.task/moby/481308628ffcd074969046225f8a9435aea56255ad7b6d1eb690b2e3f270004e pid=2881 runtime=io.containerd.runc.v2`

Last log line was 30 minutes ago

agners, 2023-04-18:
Also logs in general (using  `journalctl -f`) is there some logging going on?

elmr91, 2023-04-18:
Nothing particular:
`# journalctl -f
Apr 18 17:00:40 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:00:51 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:01:01 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:01:12 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:01:22 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:01:33 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:01:43 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:02:23 homeassistant hassio_supervisor[397]:
Apr 18 17:03:06 homeassistant qemu-ga[346]: info: guest-ping called
Apr 18 17:03:06 homeassistant qemu-ga[346]: info: guest-fsfreeze called`

Last line is related to a VM backup I made few minutes ago.

elmr91, 2023-04-18:
System is running a low power but quite efficient CPU: 
4 x Intel(R) Core(TM) i5-7500T CPU @ 2.70GHz (1 Socket)
24 GB RAM (2 GB allocated for HAOS VM)

elmr91, 2023-04-18:
I restored backup taken during last night (HAOS 9.5 / before upgrade)
CPU is back to normal (2%)

No containerd process using CPU

wokkeltje13, 2023-04-18:
See same behavior but not so big, probably a bit more powerful CPU
[image]

The spikes jou see is from the update and corresponding reboot, but after the upgrade you see a slightly higher CPU usage and MEM usage (should be lower after a fresh reboot, but is higher)

dsolva, 2023-04-18:
Im using proxmox and also experienced an increase from 6% to 9% on cpu but i dont think I have a more powerful cpu. containerd using 3-4%. 
Cpu is 2x Intel J4025.

agners, 2023-04-18:
Higher RAM usage is somewhat expected as we tuned VM swapiness to use less swap (since we use zswap now, which is stored on the data disk).

As for CPU load: I can't really reproduce this here on my KVM installation, it seems to be about the same on HAOS 9.5 and 10.0. 

I've also seen that CPU frequency scaling is behaving differently on the new kernel version. Maybe this influences the reported CPU load, not sure. However, since this is virtualization, maybe unlikely :thinking:

evilspoons, 2023-04-18:
Similar results here. Host is Proxmox VE 7.4-3, i5-7500. Went from 3.73 GiB to 4.11 GiB RAM (no big deal), CPU went from 2.4-2.5% before upgrade to 3.5% after upgrade.

Virtual machine configuration is 8 GB RAM, 4 processors with "host" as CPU type setting, OVMF (UEFI), i440fx, virtio scsi and LAN, and one USB device passed through.

wokkeltje13, 2023-04-18:
Not sure if CPU is more powerful
4 x Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz (1 Socket)

Did a rollback
You can see CPU is like before and MEM is a lot lower than before (system had an uptime of 5 days when running 9.5)
So 10.0 after the reboot used 900MB more mem
[image]

agners, 2023-04-18:
> So 10.0 after the reboot used 900MB more mem

As I've said before, this is expected: Home Assistant OS 9.5 used a different swapiness value, meaning it more happily swapped things out (to zram, means part of the memory will get compressed and swapped out). With OS 10.0 it will swap out much later, use more RAM as long as available. Overall, this should not change much, maybe some tasks will be a bit fater (since the system doesn't need to decompress pages swapped out to zram), or a bit slower (since we work with a slightly smaller page cache), YMMV.

pickonedev, 2023-04-18:
CPU and MEMORY 100% usage with the new update...

elmr91, 2023-04-18:
@evilspoons & @wokkeltje13, we have very similar CPU.
I think I know why upgrade from 9.5 to 10 shows a different CPU behavior on my system: I'm using "powersave" CPU governor on host system.

I switched back to "performance" on host : 
`echo "performance" | tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor`

 CPU usage on HAOS 10 decreases to 4.5 % (same increase as other have seen):
[image]

Using powersave host governor magnifies CPU usage difference between HAOS 9.5 & 10

Upgrade from HAOS 9.5 to 10 increases CPU usage. It has a direct impact on energy usage.
My system rose from 6-7 Watt to 8-9 Watt after upgrade

[image]

Less than 10W is still a very good value for a hosting system, but HAOS 10 is less energy efficient than 9.5.
It seems related to the new docker version.

wokkeltje13, 2023-04-18:
mine is also on powersave, did not change during tests with 10.0

pickonedev, 2023-04-18:
[image]

Mine 100% CPU and Memory. Anyone knows how to downgrade the OS version?

dsolva, 2023-04-18:
@elmr91 makes sense why your cpu increase seemed greater due to power safe. I have powersafe turned off and the upgrade caused around 0.8watt increase only (i assume because my processor tdp is just 10W). Almost a watt is a lot for my "power efficient" NUC.

elmr91, 2023-04-18:
> @elmr91 makes sense why your cpu increase seemed greater due to power safe. I have powersafe turned off and the upgrade caused around 0.8watt increase only (i assume because my processor tdp is just 10W). Almost a watt is a lot for my "power efficient" NUC.

mine (i5-7500T) is a 35W TDP processor
@wokkeltje13's (i5-6500) and @evilspoons's (i5-7500) are 65W TDP processors
They should have also few Watt more on HAOS 10

agners, 2023-04-18:
> Mine 100% CPU and Memory. Anyone knows how to downgrade the OS version?

Use this command in the shell:

```
ha os update --version 9.5
```

pickonedev, 2023-04-18:
@agners Thank you!

Any idea why my CPU it is going to full 100 percentage? Memory as well...
I use a PC Dell OptiPlex 3090 MFF, Intel® Core™ i7-10700T @4.50 GHz, 32GB DDR4

agners, 2023-04-18:
> Any idea why my CPU it is going to full 100 percentage? Memory as well...
> I use a PC Dell OptiPlex 3090 MFF, Intel® Core™ i7-10700T @4.50 GHz, 32GB DDR4

Are you using the motionEye add-on?

Taronaga, 2023-04-19:
nuc Intel® Celeron® Processor N3060 after 80% update also problem with mqtt, rolled back to 9.5

pickonedev, 2023-04-19:
> > Any idea why my CPU it is going to full 100 percentage? Memory as well...
> > I use a PC Dell OptiPlex 3090 MFF, Intel® Core™ i7-10700T @4.50 GHz, 32GB DDR4
> 
> Are you using the motionEye add-on?

Yes, I use MotionEye addon as well as other addons.

pickonedev, 2023-04-19:
> nuc Intel® Celeron® Processor N3060 after 80% update also problem with mqtt, rolled back to 9.5, loading became 10-12%, MQTT 6.1.3 - normal flight. p.s. How did the problems start with updating Mosquitto broker to 6.2.0 - I'm already afraid to update, and so I sat without light for half a day [image]

For me, MQTT 6.20 doesn't have any problem. Only the new OS update made my CPU and MEM going like crazy :-)

elmr91, 2023-04-19:
After few hours idling, I can confirm OS10 has a direct impact on host CPU usage (linked to containerd daemon CPU in HAOS VM)

[image]

gfn256, 2023-04-19:
Running Proxmox on i5-6500T 8gb ram with HASSOS vm on 2 cores 4gb ram
Increased VM CPU usage from ver 9.5 to 10.0 more than 3%
Proxmox node itself (above machine - all 4 cores) shows increase of about 1% (sometimes even less).
Not seeing much changed RAM use on either.

One other note, the upgrade was completely seamless, the whole thing took maybe 1-2 minutes.

pickonedev, 2023-04-19:
Maybe the next update, will have some better optimisations...

danielholm, 2023-04-19:
My unit went from 20% to 100%. I disabled some docker containers and now it's a bit lower than before, but still much higher spikes than using 9.5. Very smooth upgrade, though. 

[image]

elmr91, 2023-04-19:
I tried to simply restart containerd service: 
`systemctl restart containerd`

CPU used by containerd was divided by 3 after service restart (from 7-8% to less than 2%).
Host CPU dropped from 6-7 to 4-5 %

agners, 2023-04-19:
@Galapalos @danielholm @pickonedev the very high CPU load is most likely related to motionEye. Frenck is working on an upgrade, it should be available soon. However, this is kinda off-topic here, as this issue is about a a slightly higher CPU usage due to containerd. If the issue persists after the motionEye update, please open a new issue so we can track it separately.

geobogb, 2023-04-20:
I'm not running motionEye but have the same problem with high CPU usage on 10.0. I'm running Home Assistant in a headless Virtual Box machine. On 9.5 it uses around 5% CPU on the host machine (Debian/bullseye). After upgrading to 10.0. the CPU usage rose to over 25%. After downgrading to 9.5, everything is back to normal.

Please let me know if I can provide any help in resolving this issue!

gfn256, 2023-04-20:
Thought I'd share my CPU temps graph since updating to 10.0 - and I don't like the look of it, see my post above - I'm going to get my Tasmota energy measuring socket out to check the wattage usage now.
[image]

Info is sent from Proxmox VE by mqtt every 5 mins (cronjob) to HA
I marked for reference, the 2 spikes - which are the nightly PVE VM HAOS backups.
Checked the ambient temp and its pretty stable for period.

geobogb, 2023-04-20:
> @Galapalos @danielholm @pickonedev @geobogb please make sure to run motionEye 0.19.0,

But I'm not using/running motionEye at all!

andrazek, 2023-04-20:
I experience the same problem. My temps went up by about 5°C with this upgrade. For passive mini PCs that's definitely too much. It will wear out sooner. How can we find the cause of this problem? I'm not using motionEye.

pickonedev, 2023-04-20:
@geobogb I made the new update for motionEye and OS and everything is ok now. The CPU and MEM are the same as before. 
Thank you!

lyricnz, 2023-04-21:
I have the same problem as reported multiple times above.  High CPU usage since 10.0, VM hosted on proxmox, with plenty of CPU + RAM available. Screenshots of HA and proxmox below

[image]
[image]

CPU usage appears to be in homeassistant container (according to "top" and "docker stats").  I'm not running motioneye.

lyricnz, 2023-04-21:
Reverting to 9.5 caused an immediate lowering of CPU usage

[image]

SeByDocKy, 2023-04-21:
> @Galapalos @danielholm @pickonedev the very high CPU load is most likely related to motionEye. Frenck is working on an upgrade, it should be available soon. However, this is kinda off-topic here, as [...]

I even uninstall motioneye and still x 5 more usage.... for me it's not related....

pickonedev, 2023-04-21:
For me it is even better than before...
In the past, CPU was 4%, MEM 8% and TEMP 38 C ...
After the OS update, I had problems with CPU & MEM at 100%, but after I have updated MotionEye and again OS, the CPU is at 3%, mem at 7,2% and temp at 32 C

....

jlunz, 2023-04-22:
I am seeing a similar behavior on a Raspberry Pi Zero 2 W (I know, not recommended ;-))
Load (15m) jumped from almost zero to ~0.5

[image]

geobogb, 2023-04-23:
Still same problem running Home Assistant in Virtual Box. Upgraded to 10.0 around 15:30 and reverted back to 9.5 around 18:30. From the inside, the CPU increase might not be so dramatic:

[image]

But seen from the host it is!

[image]

And, with the risk of repeating myself, I'm not using motionEye.

wokkeltje13, 2023-04-23:
I think there are enough people who showed evidence that version 10 has an impact on resource usage. Increased mem usage is by design, but I do not see any reason why CPU and related to that heat and power consumption should be higher (The MotionEye thing is not relevant to this issue since most people even don't have it installed).

I also see no steps in troubleshooting or remediation appear. Is there something we can do to help since most people just like I did, returned to 9.5. But this is also no long-term solution.

lyricnz, 2023-04-26:
Nothing in current 10.1 beta appears to relate to this issue https://github.com/home-assistant/operating-system/releases/tag/10.1

lyricnz, 2023-04-26:
I tested upgrade, followed by some usage, followed by HAOS revert.  CPU vs versions.

[image]

HA core version, and all integrations, remain unchanged.

jandado, 2023-04-26:
I can confirm on my somewhat new AMD Ryzen 7 5800H, the VM has 2 cores allocated to it. CPU usage doubled, containerd appears to be behind it.
[image]

dsolva, 2023-04-27:
OS 10.1 oficial release is out. I'm curious if the problem persists.

jandado, 2023-04-27:
Yes it does persist @dsolva. I have upgraded over an hour ago, I see no change compared to 10.0

gfn256, 2023-04-28:
Not only does problem persist in 10.1, but I see an increased CPU activity compared with even 10.0 of +/- 2% and a similar average temp increase.
This needs addressing.

TheHolyRoger, 2023-04-28:
Certainly sir, would you like a caesar, a balsamic or a lemon and herb dressing?

agners, 2023-04-28:
FWIW, it is unlikely that 10.1 addresses this as there is no Docker update. It seems that this is related to containerd.

What would be interesting is if a Supervised installation with Debian 11 using Docker 23.0 (compared to Docker 20.10) sees the same increase.

The latest dev build upgrades to Docker 23.0.5, if someone could test that would be interesting:

https://os-builds.home-assistant.io/11.0.dev20230427/

Or to update your system directly (please create and download a backup, since this updates to development builds):

```
ha su options --channel=dev
ha su reload
ha os update
ha su options --channel=stable
```

ArtJames, 2023-04-28:
> Or to update your system directly (please create and download a backup, since this updates to development builds):

Proxmox VE latest, J4105, VM: 2 cores/4GB
9.5 - ~2.1%
11.0 dev - ~6.8%
[image]

Dragonfir3, 2023-04-30:
High CPU consumption with version 10 and 10.1 (containerd: 6-7% all time). Downgraded to 9.5.
[image]

Christophe999s, 2023-05-01:
Add me to the list of affected users. Downgraded to 9.5 and cpu usage went back to normal. Also running on pve 7.4-3

lyricnz, 2023-05-01:
This issue doesn't seem to be getting much traction. How can we help?  (as competent sysadmin/python devs, but not especially familiar with HAOS)

agners, 2023-05-02:
The problem most likely is related to the Docker 23.0 upgrade. To isolate that, it might make sense to build an image using HAOS 9.5, but just upgrade Docker to 23.0, to verify that the problem indeed is related to Docker 23.0. If that is proven, opening an issue in the Docker GitHub project (moby) is probably the next step.

However, in my experience, such an issue entry unlikely will trigger a quick fix. To get it resolved, likely we'd have to track down the actual issue ourselfs. Since it is the `containerd` process which is causing higher CPU load, we'd have to dig into that. One option is using some kind of profiling. I am not familiar with Go profiling, but I am sure there are ways to profile a go process to figure out which operations use (more) CPU than they did before.

During the hole process it could also be that feature x of `containerd` just requires more CPU, and this was expected, in which case the whole endeavor was useless.

I am expecting this to take a significant amount of work to tackle. Since this "only" affects CPU usage, it isn't high on my priority list right now. I also have some hope that some stable package update (e.g. a new Docker patch release) suddenly resolves the issue :crossed_fingers:

elmr91, 2023-05-02:
An old containerd issue provides some hints about debugging/profiling containerd/docker:
[https://github.com/fnproject/fn/issues/700](url)

WouterJN, 2023-05-03:
I'm running on a Proxmox server and after upgrading from 9.x to 10.1 i have the same problem;

[image]

CPU went from 6% to 15%. Supervisor in both cases is 2023.04.1.

tyjtyj, 2023-05-06:
Running 4 HA on 3 different hardware seeing the same issue. RPI4B rev1 for stanby. RPI3B for test, RPI3B in different home with different config and Intel NUC on proxmox. All 4 of them seeing same CPU increase behavior. The issue disappear by going back to 9.5. Like other said, seems containerd is the culprit. 
No motion eye. All 4 system has different addons. The only common addons in terminal.

Impact123, 2023-05-09:
I found this interesting and figured I'll try to replicate this. I used a basic debian 11 VM, cloned it and installed docker/containerd `23.0.6`/`1.6.21` (latest at this time) on one and `20.10.22`/`1.6.8` (What HAOS 9.5 uses) via the new `--version` parameter of docker's installation script.
I used `apt install containerd.io=1.6.8-1` to downgrade `containerd` and rebooted.
I then used [the docker run command from the docs](https://www.home-assistant.io/installation/linux#platform-installation) and let the VMs idle for a bit.
I used proxmox's `metric server` output to gather the stats. [You can also see a snapshot of it here](https://snapshots.raintank.io/dashboard/snapshot/nQdFESeR72MB4i0fq4VZ7vwGEpRs2Ayx).
Maybe my testing is flawed but I can't reproduce this behavior outside of HAOS. At least it doesn't seem to be a general docker/containerd issue.
[image]

agners, 2023-05-09:
@Impact123 these are very interesting findings! I guess this could mean two things:

a) Latest docker/containerd `23.0.6`/`1.6.21` fixes the problem
b) The docker/containerd build config in HAOS is different
c) It is related to some other environment

I'd say c) is most likely, and there are lots of candidates: Go version used, glibc version or the Linux kernel are probable candidates. To rule out a) I can bump to this latest version relatively easy in HAOS 11 dev builds, I'll tackle that today. Then we can test if that fixes the problem. If not, we know it must be caused by something else than containerd itself.

@elmr91 building HAOS from scratch is documented on the [developer website](https://developers.home-assistant.io/docs/operating-system). Of course it requires manually adjusting Docker/containerd package versions. I'll tackle that at one point, but currently the Bluetooth issues have higher priority.

[71 more replies]

closed 2023-10-05 by agners