home-assistant/operating-system #2477: Fix for issue #2390 makes upgrade from 9.5 to 10.0 fail due to missing hassos-data

NateCrawford, 2023-04-18:
### Describe the issue you are experiencing

Upgraded HAOS 9.5 to 10.0 (QEMU 7.2 ARM Virtual Machine running on M1 Mac Mini), which relabeled the "NVMe" virtual disk image to hassos-data-dis. This leaves no filesystem with hassos-data label, and the system stops with the "waiting for home assistant cli to be ready" message.

What is the proper way to re-enable the external disk? Running this in the recovery shell at least got back to a pre-10 system: e2label /dev/nvme0n1p1 hassos-data

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Move datadisk to nvme
2. Upgrade HAOS from 9.5 to 10
3. System hangs on reboot when no FS with "hassos-data" label exists
...

### Anything in the Supervisor logs that might be useful for us?

```txt
None
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

agners, 2023-04-18:
> Upgraded HAOS 9.5 to 10.0 (QEMU 7.2 ARM Virtual Machine running on M1 Mac Mini), which relabeled the "NVMe" virtual disk image to hassos-data-dis. This leaves no filesystem with hassos-data label, and the system stops with the "waiting for home assistant cli to be ready" message.

Hm, this can happen if it takes more than 10s for the data disk device to be ready (we wait up to 10s for data disks to appear, see https://github.com/home-assistant/operating-system/commit/5200096c4e59e37139d097d0e2ec618086160268#diff-2170d95eefeb4f36a11bb3a7a4ea99713dbc1615f39f309b5c4e47222dd720b8R7-R8). This seems a really long time to me.

Do you have a boot log of that boot? You can get the log of the last x boot by using:

```
journalctl -b -x
```

> What is the proper way to re-enable the external disk? Running this in the recovery shell at least got back to a pre-10 system: e2label /dev/nvme0n1p1 hassos-data

Yes this should be relatively safe to do.

NateCrawford, 2023-04-19:
The haos-data-disk-detach script seems to work, as it detects the existing hassos-data file system as external, then relabels it as hassos-data-dis. The problem appears later when the system detects no hassos-data filesystem and gets stuck. Note: I may just be impatient; if I wait longer, does the system re-enable the external data disk automatically?

I don't know where in the upgrade procedure this script gets called, but manually re-labeling hassos-data-dis back to hassos-data results in the old HAOS 9.5 booting. Does the upgrade to 10.0 actually happen on next successful boot?

I can't get at the system logs right now, but will attempt to upgrade again later.

closed 2023-07-25 by github-actions[bot]

cygeus, 2024-02-06:
I do experience this issue every time when doing an upgrade. Up until now, it happened when I upgraded

- Home Assistant OS Generic aarch64 from 11.4 to 11.5
- Home Assistant OS Generic aarch64 from 11.3 to 11.4
- Home Assistant OS Generic aarch64 from 11.2 to 11.3

This install is running on a KVM Hypervisor. The machine has 2 disks: /dev/vda (containing the System partitions) and /dev/vdb (containing the Data partition), each mapped to a qcow2 image.

It is easy to fix by just running `# e2label /dev/vdb1 hassos-disk` and rebooting the VM, but this is definitely not the most ideal long-term solution.
How can I assist to get this issue debugged and fixed?

agners, 2024-02-06:
> How can I assist to get this issue debugged and fixed?

Do you still have a partition called `hassos-disk` on the first disk?

Right after an update, can you check if `systemctl status haos-data-disk-detach.service` did run?

You can probably simulate an update by manually downgrading using `ha os update --version 11.4`.