home-assistant/operating-system #2478: XDO BT802 (RTL8761BU) stopped working on HA Yellow after OS 10 update

CrazyCoder, 2023-04-18:
### Describe the issue you are experiencing

External USB dongle stopped working after installing OS 10 and rebooting. 

Re-plugging the dongle didn't help, using a different USB port didn't help, adding extension cable didn't help, Yellow device power cycling didn't help.

Reverting back to OS 9.5 and re-plugging the dongle fixed the issue.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

- Update to OS 10
- BT USB stops working

### Anything in the Supervisor logs that might be useful for us?

```txt
2023-04-18 10:59:40.755 WARNING (MainThread) [homeassistant.config_entries] Config entry 'USBBT' for bluetooth integration not ready yet: Bluetooth adapter None with address 00:E0:42:C4:2E:03 not found; Retrying in background
```
```

### Anything in the Host logs that might be useful for us?

```txt
dmesg:

[  295.323082] usb 1-1.1: new full-speed USB device number 7 using dwc2
[  295.523061] usb 1-1.1: device descriptor read/64, error -32
[  295.835098] usb 1-1.1: device descriptor read/64, error -32
[  296.147105] usb 1-1.1: new full-speed USB device number 8 using dwc2
[  296.347100] usb 1-1.1: device descriptor read/64, error -32
[  296.663082] usb 1-1.1: device descriptor read/64, error -32
[  296.775219] usb 1-1-port1: attempt power cycle
[  297.479132] usb 1-1.1: new full-speed USB device number 9 using dwc2
[  297.895101] usb 1-1.1: device not accepting address 9, error -32
[  298.099129] usb 1-1.1: new full-speed USB device number 10 using dwc2
[  298.515224] usb 1-1.1: device not accepting address 10, error -32
[  298.522448] usb 1-1-port1: unable to enumerate USB device

```
```

### System information

```
Home Assistant 2023.4.5
Supervisor 2023.04.0
Operating System 10.0
Frontend 20230411.1 - latest
```

### Additional information

Built-in BT is disabled via `/boot/config.txt`:

```
dtoverlay=miniuart-bt
dtoverlay=disable-bt
```

agners, 2023-04-18:
It seems to be some low-level USB issue. Did you try to change the port? Maybe powering off the device completely and power it on again helps?

We do use the new Linux 6.1 kernel from Raspberry Pi, but it shouldn't cause such issues obviously.

CrazyCoder, 2023-07-23:
It works fine in the latest update.

closed 2023-07-23 by CrazyCoder