home-assistant/operating-system #2482: Raspberry PI cannot boot from disk after upgrade to 10

ocalvo, 2023-04-19:
### Describe the issue you are experiencing

After upgrade RPI wont boot:

[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from 9.5 to 10

### Anything in the Supervisor logs that might be useful for us?

```txt
Cannot access logs
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

Cannot access system info

### Additional information

Cannot access system

agners, 2023-04-19:
It doesn't boot beyond this point? This seems very strange, as this is before any of the files getting touched from HAOS (this screen is shown by the EEPROM). It seems as if HAOS vanished from the USB SSD?

Can you double check the content of the USB SSD (e.g. checking partition table using fdisk and check the content of the first partition).

This is essentially another variation of "boot from USB" issue on Rasberry Pi. As I've stated numerous times before, USB boot on Raspberry Pi is unfortunately notoriously unstable. I recommend booting from SD card, and use the SSD as data disk.

agners, 2023-04-19:
Possibly duplicate of #2480.

jurasjo, 2023-04-20:
Maybe updating bootloader can help? It looks like yours is quite old one.
https://pimylifeup.com/raspberry-pi-bootloader/

cbird527, 2023-04-21:
I can confirm this happened to me as well after the update to 10.0. I also tried updating the boot loader to 8ba17717 2023/01/11 and a fresh image of HA OS 10.0 and it would not boot either.  After rolling back to 9.5 there are not issues.

My hardware is a Pi CM4 with 8Gb ram and booting off an NVMe drive.

bcutter, 2023-04-24:
> This is essentially another variation of "boot from USB" issue on Rasberry Pi. As I've stated numerous times before, USB boot on Raspberry Pi is unfortunately notoriously unstable. I recommend booting from SD card, and use the SSD as data disk.

This is very interesting and helpful information.

As I don't want to hijack this issue but I think those questions are important to avoid future issues of this kind (on nearly every HA OS update there are reports of not booting Pi's/boards with SAD boot devices...), where can one find answers on these questions?
1. **What data _exactly_ will remain on the SD card (and therefore what will be outsourced) once the move data disk feature has been applied?** Docker containers, which parts of the OS, what about the Supervisor etc.
2. **Can this be undone, so can an outsourced data disk be sourced back in (everything back on internal SD card)?** Or is this feature a one way road?

ocalvo, 2023-04-26:
If I disconnect the USB system drive and reconnect again, then the boot process continues:
[image]

ocalvo, 2023-04-26:
Maybe I am having a hardware issue, now I see this:
[image]

ocalvo, 2023-04-26:
I am losing my mind, after disconnection everything and flashing 10.1 again, the system was able to boot with every USB connected:
[image]

bschatzow, 2023-04-26:
@ocalvo 
I have had this same issue off and on with different releases.  My system is using a hub for my USB SSD.  It is only on a reboot, not an update and it is not every time.  Like you if I remove my hub and start the boot and then plug in it works every time.  I believe it is a timing issue that a small number of users see.  Now I never update unless I in home to monitor it.

agners, 2023-04-26:
As @bschatzow says, some of the USB SSDs have a timing problem with the detection. The Linux kernel has quirks and hacks to restart detection etc. for devices which violate the USB spec. The Raspberry Pi firmware and the U-Boot USB stack (our bootloader) is not as forgiving. Quite likely that you see some kind of race condition here.

Also, I've seen [in-rush current](https://en.wikipedia.org/wiki/Inrush_current) problems with USB SSDs powered by USB only. Some of them restart again just because they don't get enough juice at first.

> usb 1-1-port4: over current-change #84

So this could be caused by in-rush current issue, or a generally underpowered SSD. To mitigate using a powered hub is most likely helpful.

closed 2023-08-01 by github-actions[bot]