home-assistant/operating-system #2484: Odroid N2+ no more audio output with VLC addon after upgrade to HA OS 10.0

GoSpursGoNL, 2023-04-19:
### Describe the issue you are experiencing

With HA OS 9.5 and earlier this worked fine, but since the upgrade to HA OS 10.0 outputting any sound via the 3.5mm plug of my Odroid N2+ does nothing.

Not sure which logs are relevant. This is what the VLC addon mentions:
```
023/04/19 04:48:13 [notice] 247#247: using the "epoll" event method
2023/04/19 04:48:13 [notice] 247#247: nginx/1.18.0
2023/04/19 04:48:13 [notice] 247#247: OS: Linux 6.1.24
2023/04/19 04:48:13 [notice] 247#247: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2023/04/19 04:48:13 [notice] 247#247: start worker processes
2023/04/19 04:48:13 [notice] 247#247: start worker process 273
[0000ffff92e86570] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0000ffff92e86570] main interface error: no suitable interface module
[0000ffff92fcab50] main libvlc error: interface "dbus,none" initialization failed
[0000ffff92e81360] main interface error: no suitable interface module
[0000ffff92fcab50] main libvlc error: interface "globalhotkeys,none" initialization failed
[0000ffff92ead9d0] main playlist: playlist is empty
[0000ffff92e81750] [http] lua interface: Lua HTTP interface
[0000ffff92e866a0] [telnet] lua interface: Listening on host "telnet://:4212".
[06:48:14] INFO: Successfully send discovery information to Home Assistant.
```

The addon gives 'Default' and 'Built-in audio stereo' as options, I tried both.

When trying to play a sound, in the HA logs I see:
Connection error: Failed to read: 0 bytes read on a total of undefined expected bytes. Partial bytes read: b''

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. With a N2+ use HA OS 9.5, output sound to the media player created by the VLC addon, which is configured to output sound to the board's 3.5mm plug
2. Hear the sound
3. Upgrade to HA OS 10.0
4. Retry step 1 -> silence

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2023.4.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.24
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4996
Installed Version | 1.32.1
Stage | running
Available Repositories | 1264
Downloaded Repositories | 6

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 22

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 113.9 GB
disk_used | 83.1 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.0), Duck DNS (1.15.0), Home Assistant Google Drive Backup (0.110.3), NGINX Home Assistant SSL proxy (3.2.0), Node-RED (14.1.4), Plex Media Server (3.2.4), SSH & Web Terminal (13.1.0), WireGuard (0.8.0), AdGuard Home (4.8.6), VLC (0.1.3), Zigbee2MQTT (1.30.3-1), Mosquitto broker (6.2.0), chrony (2.5.0), ESPHome (2023.3.2), File editor (5.5.0), HomePod Connect (28.6-ls106)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 2
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 18, 2023 at 16:36
-- | --
current_recorder_run | April 18, 2023 at 18:54
estimated_db_size | 94.36 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

AarvdH, 2023-04-19:
I have exactly the same, same system, same log output. Since upgrade to 10.0

agners, 2023-04-19:
This is resolved by this change https://github.com/home-assistant/plugin-audio/pull/134.

With the Audio Plug-in 2023.04.0 audio will work with HAOS 10 again. You can already get it from beta channel using:

```
ha su options --channel=beta
ha su reload
ha audio update 
```

You should now see that 2023.04.0 is installed.
```
ha audio info
```

You may need to restart the VLC add-on after that.

closed 2023-04-19 by agners

AarvdH, 2023-04-19:
Great! Thank you Agners.. Working perfect again. Must I turn the beta off again? Or can I leave it this way?

GoSpursGoNL, 2023-04-19:
Thank you for the quick fix! Works again indeed. I left the beta channel afterwards (@AarvdH via gui -> settings -> system -> updates -> 3 dots -> leave beta channel)