home-assistant/operating-system #2485: Bluetooth failing after upgrade to HAOS 10

Eyco1981, 2023-04-19:
### Describe the issue you are experiencing

I'm running a HA Yellow with a BT/WiFi CM4.
After a non consistant uptime the internal Bluetooth adapter is stopping to work and the integrations fail.
6 devices have been integrated, but even after reducing to 3 devices the problem occurs.
Unlike on HAOS 9.5, Bluetooth doesn't seem to recover from intermittend issues.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to HAOS 10
2. Reboot
3. Wait for failing BT integrations

### Anything in the Supervisor logs that might be useful for us?

```txt
No error messages in Supervisor logs
```

### Anything in the Host logs that might be useful for us?

```txt
Apr 19 09:13:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:15 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:15 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:15 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:15 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:15 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:15 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 19 09:13:35 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:45 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:45 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:45 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:45 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:13:45 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:13:45 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 19 09:14:05 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:14:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:14:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:14:15 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 19 09:14:15 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 19 09:14:15 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
[... 165 lines cut ...]
First occurred: 01:03:47 (1219 occurrences)
Last logged: 11:13:15

hci0 (E4:5F:01:6F:FF:84): Failed to restart Bluetooth scanner: hci0 (E4:5F:01:6F:FF:84): Failed to start Bluetooth: [org.bluez.Error.InProgress] Operation already in progress
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 223, in _async_start
    await self.scanner.start()  # type: ignore[no-untyped-call]
  File "/usr/local/lib/python3.10/site-packages/bleak/__init__.py", line 182, in start
    await self._backend.start()
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/scanner.py", line 191, in start
    self._stop = await manager.active_scan(
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/manager.py", line 368, in active_scan
    assert_reply(reply)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 350, in _async_restart_scanner
    await self._async_start()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 305, in _async_start
    raise ScannerStartError(
homeassistant.components.bluetooth.scanner.ScannerStartError: hci0 (E4:5F:01:6F:FF:84): Failed to start Bluetooth: [org.bluez.Error.InProgress] Operation already in progress`

agners, 2023-04-19:
HAOS 10 uses the same firmware git hash as HAOS 9 is using:

- HAOS 9: https://github.com/home-assistant/buildroot/blob/2022.02.x-haos/package/brcmfmac_sdio-firmware-rpi/brcmfmac_sdio-firmware-rpi.mk
- HAOS 10: https://github.com/home-assistant/buildroot/blob/2023.02.x-haos/package/brcmfmac_sdio-firmware-rpi/brcmfmac_sdio-firmware-rpi.mk

So that would mean that the new BlueZ version (5.66) or the new Kernel 6.1 is causing this problem.

@bdraco have you seen such regressions with newer BlueZ/Kernel versions?

Dillton, 2023-04-19:
Hi.
I am experiencing similar issue with my HP mini PC running HA in Proxmox and using Intel 9560 with iwlwifi firmware.
Currently not home so I can't post any logs or screens from HA VM.

Inside HA I am using Awox community integration for Awox bluetooth lights and since yesterday update its completely unusable. Below at least screen from history of lights mesh and one light where you can see before 7PM (time of updating my system) the connection was stable.
[image]

Heronimonimo, 2023-04-19:
Same issue here after upgrading to 10.0 running on a Proxmox VM on a Intel Nuc from the 8th generation.

cnestande, 2023-04-20:
I experience the same issue on an Intel NUC with HAOS 10, worked fine before the upgrade to HAOS 10.

Heronimonimo, 2023-04-20:
> Same issue here after upgrading to 10.0 running on a Proxmox VM on a Intel Nuc from the 8th generation.

Downgrading HA OS to 9.5 fixed it for me.

240Sidan, 2023-04-20:
> > Same issue here after upgrading to 10.0 running on a Proxmox VM on a Intel Nuc from the 8th generation.
> 
> Downgrading HA OS to 9.5 fixed it for me.

Is it the core you downgraded? Or how do you downgrade to 9.5?

Eyco1981, 2023-04-20:
> Is it the core you downgraded? Or how do you downgrade to 9.5?
 With the command: 
 ha os update --version 9.5

240Sidan, 2023-04-20:
> > > Same issue here after upgrading to 10.0 running on a Proxmox VM on a Intel Nuc from the 8th generation.
> > 
> > 
> > Downgrading HA OS to 9.5 fixed it for me.
> 
> Is it the core you downgraded? Or how do you downgrade to 9.5?

Thanks will try it when I get home

cmangla, 2023-04-20:
I'm experiencing a similar issue on Raspberry Pi 3.

teranex, 2023-04-20:
Same here with HAOS 10 on RPi3. Bluetooth (used for Xiaomi BLE LYWSD03MMC  temperature) works for a few hours, then fails and doesn't come back. In 9.5 this sometimes also happened (once every few days), but after a few minutes it automatically came back. Now only a full system reboot helps

Puma7, 2023-04-20:
Same here on 2 Intel NUC. 
One Intel NUC 11 Essential Kit NUC11ATKC4
and One Intel NUC 12 Pro Kit NUC12WSHi3

on both HA is installed on the internal nvme

[config_entry-bluetooth-68b370091bd9558e25d68e230e7092cd.json.txt](https://github.com/home-assistant/operating-system/files/11287356/config_entry-bluetooth-68b370091bd9558e25d68e230e7092cd.json.txt)

[config_entry-bluetooth-394b387a8cee694d2e66bb4f34072697.json.txt](https://github.com/home-assistant/operating-system/files/11287355/config_entry-bluetooth-394b387a8cee694d2e66bb4f34072697.json.txt)

240Sidan, 2023-04-20:
Have reverted back to HAOS 9.5 now, and it is much more reliable but sometimes it keeps disconnecting. 

Getting this message and then have to reload the Bluetooth integration. 2023-04-20 19:21:15.348 ERROR (MainThread) [homeassistant.components.bluetooth.scanner] hci0 (XX:XX:XX:XX:XX:XX): Error stopping scanner: [org.bluez.Error.InProgress] Operation already in progress

gcarrarom, 2023-04-20:
Same on RPi4. Reboot doesn't help. I'll try to rollback to 9.5 as it used to "kinda" work there - only failing every hour but recovering.

gcarrarom, 2023-04-20:
Smells like BLE issue for RPi4:
```
[core-ssh ~]$ dmesg | grep Blue
89.173721] Bluetooth: Core ver 2.22
89.173853] Bluetooth: HI device and connection manager initialized
89.173873] Bluetooth: HI socket layer initialized
89.173883] Bluetooth: L2CAP socket layer initialized
89.173900] Bluetooth: SCO socket layer initialized
89.184081] Bluetooth: HI UART driver ver 2.3
89.184104] Bluetooth: HI UART protocol H4 registered
89.184176] Bluetooth: HI UART protocol Three-wire (H5) registered
89.184342] Bluetooth: HI UART protocol Broadcom registered
89.660433] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
89.660453] Bluetooth: BNEP filters: protocol multicast
89.660470] Bluetooth: BNEP socket layer initialized
89.665767] Bluetooth: MGMT ver 1.22
137.963015] Bluetooth: RFCOMM TTY layer initialized
137.963052] Bluetooth: RFCOMM socket layer initialized
137.963081] Bluetooth: RFCOM ver 1.11
[22987.140431] Bluetooth: hi0: Opcode 0x200d failed: -110
[22987.140494] Bluetooth: hi: request failed to create LE connection: err -110
```

megaz555, 2023-04-21:
Same issue here, running on Intel NUC 7th gen with Proxmox (latest version).

Based on the above feedback, it looks like something is very seriously broken with Bluetooth drivers in HAOS 10.

obol2, 2023-04-21:
Same with my Intel NUC (HA running on Proxmox). Reverting to version 9.5 fixed the problem.

osaether, 2023-04-21:
Issues here as well. Running a VM in Proxmox. 
I have 15 Xiaomi sensors and two Inkbird. A restart of HA bring some back but looses others. I have not managed to get all back. 
Reverting to OS v9.5 fixes the problems.

mcc05, 2023-04-21:
Same here my bluetooth dongle is not longer recognised 
Ok on 9.5

mariusnauj, 2023-04-21:
I experience the same issue on an Intel NUC with HAOS 10, worked fine before the upgrade to HAOS 10.

240Sidan, 2023-04-21:
> I experience the same issue on an Intel NUC with HAOS 10, worked fine before the upgrade to HAOS 10.

Try revert back to 9.5 it worked for me 
With the command:
ha os update --version 9.5

mariusnauj, 2023-04-21:
Yes we can revert to 9.5, but need that HA fix this issue.

agners, 2023-04-21:
The latest stable Linux kernel 6.1.25 has some changes which might address this problem:
https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.25

In particular:
- Bluetooth: Fix printing errors if LE Connection times out
- Bluetooth: hci_conn: Fix not cleaning up on LE Connection failure

So far I was not able to reproduce the failures described here, hence I cannot test the new kernel myself. The last nightly development build comes with the 6.1.25. Can someone who can easily reproduce the problem give this a try maybe? Ideally you should take a full backup and download it, just to be safe :smile: 

Updating to the latest development builds would involve:
```
ha supervisor options --channel=dev
ha supervisor reload
ha supervisor update
ha os info
```

At this point there should be a OS update offered (11.0.dev20230420). You can install it from the CLI as well

```
ha os update
```

After the update, switch back to the stable or beta channel:

```
ha supervisor options --channel=stable
ha supervisor reload
```

Downgrading back to HAOS 10 (or 9.5) is fairly safe and can be done using:

```
ha os update --version=9.5
```

240Sidan, 2023-04-21:
Will try later tonight if I got the time

Eyco1981, 2023-04-21:
@agners 
Have been running HAOS 11.0.dev20230420 for 1h 20mins.
Has been looking promising first, but Bluetooth failed again and didn't recover (have been waiting for 30 additional mins.).
Log entries are as follows:

**Host**
`Apr 21 15:31:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:10 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:10 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:10 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 21 15:31:30 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:30 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:40 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:40 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:31:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:31:40 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 21 15:32:00 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:00 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:10 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:10 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:10 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:10 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 21 15:32:30 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:30 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:35 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:41 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:41 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:41 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:41 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:32:41 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:32:41 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
Apr 21 15:33:00 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:33:00 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:33:05 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
Apr 21 15:33:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:33:05 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
Apr 21 15:33:05 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
[... cut ...]
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 350, in _async_restart_scanner
    await self._async_start()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/scanner.py", line 305, in _async_start
    raise ScannerStartError(
homeassistant.components.bluetooth.scanner.ScannerStartError: hci0 (E4:5F:01:6F:FF:84): Failed to start Bluetooth: [org.bluez.Error.InProgress] Operation already in progress`

Eyco1981, 2023-04-21:
It hasn't recovered within the last 5 hours.
So I checked if a HA core restart would help. It didn't, so I suppose a host reboot would be needed with the HAOS dev version too.

icthus86, 2023-04-22:
I'm experiencing the same problem with the Bluetooth on a Yellow with HAOS 10. I've tried the steps above and they are doing the same thing. no recovery from a failed Bluetooth initiation.

nocodeman55, 2023-04-22:
My bluetooth was completely dead after 10 update but after 5x reboots it just works now. Intel based wlan combo. I did nothing else.

240Sidan, 2023-04-23:
Upgraded to the core version 2023.4.6 and I'm on HA OS 9.5 and it is working great. Losing connection sometimes but only for a minute or so. Will try HAOS 10 now and see

greg520820, 2023-04-24:
Started having Bluetooth issue after updating to HA OS 10.0.  Running HA on virtualbox. Entering "bluetoothctl" at cli returns a  "segmentation fault". Host log attached: 

```
Apr 23 17:23:06 homeassistant systemd[1]: Stopped Home Assistant CLI.
Apr 23 17:23:06 homeassistant systemd[1]: Started Home Assistant CLI.
Apr 23 17:23:06 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-4fc6ff2d765687a79caa2d8f297fc983ffaa2b20cc522fd4215534ae18e7729d-runc.fOuyJQ.mount: Deactivated successfully.
Apr 23 17:25:34 homeassistant kernel: Bluetooth: hci0: Opcode 0x2043 failed: -110
Apr 23 17:25:34 homeassistant kernel: Bluetooth: hci0: request failed to create LE connection: err -110
Apr 23 17:26:11 homeassistant kernel: bluetoothctl[25263]: segfault at 0 ip 00007f1405faf7f6 sp 00007ffd94eca5f8 error 4 in libc.so.6[7f1405f32000+149000] likely on CPU 0 (core 0, socket 0)
Apr 23 17:26:11 homeassistant kernel: Code: 00 00 66 90 66 0f ef c0 66 0f ef c9 66 0f ef d2 66 0f ef db 48 89 f8 48 89 f9 48 81 e1 ff 0f 00 00 48 81 f9 cf 0f 00 00 77 6a <f3> 0f 6f 20 66 0f 74 e0 66 0f d7 d4 85 d2 74 04 0f bc c2 c3 48 83
Apr 23 17:26:11 homeassistant kernel: audit: type=1701 audit(1682270771.406:383): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=unconfined pid=25263 comm="bluetoothctl" exe="/usr/bin/bluetoothctl" sig=11 res=1
Apr 23 17:26:11 homeassistant systemd[1]: Created slice Slice /system/systemd-coredump.
Apr 23 17:26:11 homeassistant kernel: audit: type=1334 audit(1682270771.416:384): prog-id=127 op=LOAD
Apr 23 17:26:11 homeassistant kernel: audit: type=1334 audit(1682270771.417:385): prog-id=128 op=LOAD
Apr 23 17:26:11 homeassistant kernel: audit: type=1334 audit(1682270771.417:386): prog-id=129 op=LOAD
Apr 23 17:26:11 homeassistant systemd[1]: Started Process Core Dump (PID 25264/UID 0).
Apr 23 17:26:11 homeassistant systemd-coredump[25265]: Process 25263 (bluetoothctl) of user 0 dumped core.
Apr 23 17:26:11 homeassistant systemd[1]: systemd-coredump@0-25264-0.service: Deactivated successfully.
Apr 23 17:26:11 homeassistant kernel: audit: type=1334 audit(1682270771.550:387): prog-id=129 op=UNLOAD
Apr 23 17:26:11 homeassistant kernel: audit: type=1334 audit(1682270771.550:388): prog-id=128 op=UNLOAD
Apr 23 17:26:11 homeassistant kernel: audit: type=1334 audit(1682270771.550:389): prog-id=127 op=UNLOAD
Apr 23 17:26:19 homeassistant kernel: bluetoothctl[25274]: segfault at 0 ip 00007f4eb9c937f6 sp 00007ffc2e463848 error 4 in libc.so.6[7f4eb9c16000+149000] likely on CPU 0 (core 0, socket 0)
Apr 23 17:26:19 homeassistant kernel: Code: 00 00 66 90 66 0f ef c0 66 0f ef c9 66 0f ef d2 66 0f ef db 48 89 f8 48 89 f9 48 81 e1 ff 0f 00 00 48 81 f9 cf 0f 00 00 77 6a <f3> 0f 6f 20 66 0f 74 e0 66 0f d7 d4 85 d2 74 04 0f bc c2 c3 48 83
Apr 23 17:26:19 homeassistant kernel: audit: type=1701 audit(1682270779.545:390): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=unconfined pid=25274 comm="bluetoothctl" exe="/usr/bin/bluetoothctl" sig=11 res=1
Apr 23 17:26:19 homeassistant kernel: audit: type=1334 audit(1682270779.549:391): prog-id=130 op=LOAD
Apr 23 17:26:19 homeassistant kernel: audit: type=1334 audit(1682270779.549:392): prog-id=131 op=LOAD
Apr 23 17:26:19 homeassistant kernel: audit: type=1334 audit(1682270779.549:393): prog-id=132 op=LOAD
```

Sorien, 2023-04-24:
same problem with update of ubuntu 20.04 to 23.04 all bluetooth le devices are fucked up. it looks like Home Assistant is not receiving every le packet, reverting to 20.03 with bluez 5.64 fixes problem

agners, 2023-04-24:
> Started having Bluetooth issues after updating to HA OS 10.0. Running HA on virtualbox. Entering "bluetoothctl" at HA cli returns a "segmentation fault". Host log attached:

I've noticed that segfault too, but it seems to be limited to `bluetoothctl`, which isn't really used by Home Assistant Core.  In any case, I've addressed that problem with PR #2509.

gcarrarom, 2023-04-25:
Oh my! My Avantree DG40S USB Bluetooth Adapter stopped working as well! Now that's concerning. That was my salvation as it is "supported" from the HA page. Both onboard and the dongle are timing out and not coming back after restarts:

```
2023-04-25 00:26:22.334 WARNING (MainThread) [homeassistant.config_entries] Config entry 'DC:A6:32:38:C8:65' for bluetooth integration not ready yet: hci0 (DC:A6:32:38:C8:65): Timed out starting Bluetooth after 15 seconds; Retrying in background
2023-04-25 00:26:22.371 WARNING (MainThread) [homeassistant.config_entries] Config entry '00:FB:F9:61:D0:D3' for bluetooth integration not ready yet: hci1 (00:FB:F9:61:D0:D3): Timed out starting Bluetooth after 15 seconds; Retrying in background
2023-04-25 00:28:27.244 WARNING (MainThread) [homeassistant.config_entries] Config entry 'DC:A6:32:38:C8:65' for bluetooth integration not ready yet: hci0 (DC:A6:32:38:C8:65): Timed out starting Bluetooth after 15 seconds; Retrying in background
2023-04-25 00:28:27.471 WARNING (MainThread) [homeassistant.config_entries] Config entry '00:FB:F9:61:D0:D3' for bluetooth integration not ready yet: hci1 (00:FB:F9:61:D0:D3): Timed out starting Bluetooth after 15 seconds; Retrying in background
```

greg520820, 2023-04-25:
Installed HA OS 10.1 seven hours ago.  It fixed the the segfault error, so you can use bluetoothctl to troubleshoot bluetooth issues. Also, there have been no bluetooth errors or dropped data from bluetooth sensors.  On OS 10.0 was getting BT errors every hour.

GlenWi, 2023-04-26:
Hi Greg, where did you get version 10.1?
Still working OK for you?
Thanks.

theelog, 2023-04-26:
Same issue running on an Intel NUC. I use Xiaomi sensors via BTHome and saw intermittent connection drops after updating to 10.0. I was able to fix it by downgrading to 9.5 using this command: `ha os update --version=9.5`

Eyco1981, 2023-04-26:
Just tested HAOS 10.1.
Bluetooth is still failing and not recovering by itself.

spiek81, 2023-04-26:
Same issue here after update, running rpi3 with 3 Xiaomi BLE Temp./hum. gauges. Downgrading now back to 9.5 hopely resolving the problem for now.

johnnybegood, 2023-04-26:
I have the same issue, setup is as follows:

- RPI 4-64 with built-in Bluetooth
- Airthings Wave+ (125131)

mariusnauj, 2023-04-26:
People it's realated with HAOS 10. Because downgrade to 9.5 version fix this issue.

greg520820, 2023-04-26:
At least for me, there were two problems.

OS 10.0 contained Linux kernel (6.1.21) that had bluetooth issues.  OS 10.1 has Linux kernel 6.1.25 that contains some bluetooth fixes. 
Second problem was running HA on Virtualbox version 6.1.22, that wasn't tested on Linux kernel 6.1.XX.  After updating Virtualbox to version 6.1.44 and installing OS 10.1 the bluetooth issues seem to be resolved.  Been running for 24 hours with no bluetooth errors and only one missed data point from a bluetooth sensor.    

Path to get Bluetooth to work:
BT worked with OS 9.5 and VirtualBox 6.1.22
BT did not work with OS 10.0 and VirtualBox 6.1.22
BT did not work with OS 10.0 and VirtualBox 6.1.44
BT worked with OS 10.1 and VirtualBox 6.1.44

reindruu, 2023-04-26:
Here same problem.
I'm using 11 pieces of  Xiaomi Mijia Bluetooth Thermometer LYWSD03MMC (BLE).
Some connected via ESP32 Bluetooth Proxy. Those are not affected.
Only the pieces which are connected directly via the bluetooth from the intel NUC are affected.
No connection for those thermometers.
After downgrading HAOS from 10 to 9.5 the problem is gone.
Downgrading with: ha os update --version=9.5

GlenWi, 2023-04-26:
Thanks for sharing.  I am on Virtualbox 7.0.8 so I don't think that is the issue.  By process of elimination, it's clearly HA OS 10 and maybe 10.1 also.  Everyone who has had this issue I believe has resolved it when reverting to HA OS 9.5 - the problem lies within OS 10.
I have both the older round Mi sensors and the new smaller square ones.  Same issues - Bluetooth connections broken.

greg520820, 2023-04-26:
[GlenWi](https://github.com/GlenWi) Have you tried OS 10.1? If your are running Supervised it only takes a minute. Go to updates, 3-dots upper right, join beta, refresh and should give you an option to install OS 10.1.

Mouseskowitz, 2023-04-27:
I was also having the bluetooth issues described above. I have and Intel NUC 11 Essentials. My devices are Inkbird TH2s. I was able to reconnect the devices by hitting refresh on the bluetooth device in devices, I never had to restart any higher level component. The time between crashes was quite random, anywhere from minutes to days. 
I've downgraded to 9.5, but I'm willing to test if there is something really promising.

greg520820, 2023-04-27:
[asperkins](https://github.com/asperkins) may want to reboot your host device if you didn't?

asperkins, 2023-04-27:
> 

I did. Completely shut down, turned off and back on. It's an older optiplex 7050 micro running HA OS.

DorjeDorf, 2023-04-27:
> People it's realated with HAOS 10. Because downgrade to 9.5 version fix this issue.

Not for everyone. In my case, downgrading did not fix the problem and bluetooth is dead even now on 10.1.

SrBurner, 2023-04-27:
I have an Intel NUC 7 J4005 with a pair of Xiaomi bluetooth humidity and temperature sensors. Bluetooth is failling since OS 10 update, works for a few minutes and them stops updating information.
I have just update to OS 10.1 and issue persist. Thing I tried without luck:
- reboot the whole computer.
- Reload bluetooth integration
- Use terminal command bluetoothctl power cycle.
- Use terminal command bluetoothctl scan on

After doing any of this bluetooth works for a few minutes and stops responding.
Didn’t see anithing in logs.
With OS 9.5 was working flawlessly.

ZetaWaves, 2023-04-27:
v10.0 had bluetooth failing every 10-30 seconds.
v10.1 has bluetooth failing every 30 minutes or less.

The problem with bluetooth still exists and has not been fixed in 10.1.

ZetaWaves, 2023-04-27:
> [asperkins](https://github.com/asperkins) may want to reboot your host device if you didn't?
> 
> I've had no Bluetooth errors for over 48 hrs. on OS 10.1. Maybe the Linux kernel in 10.1 is interacting differently with the base systems .

Nope, Mine died in about an hour. It will eventually die, even on v10.1. The problem still exists.
You will also nuke your NVMe drive after awhile if you continue running these borked releases. 
At this point I'd skip v10.x altogether, seems like they're spending too much time testing in production.

Geotig, 2023-04-28:
Had to downgrade to 9.5 after trying 10 and 10.1 for a few days each. Restarting the Bluetooth integration and rebooting the machine did nothing to me.

HA OS is running in an old laptop (i5-6200u, intel wifi/bt card, 8gb ram) and I’m using the internal card for the Bluetooth connections. In 9.5 works flawlessly.

spiek81, 2023-04-28:
> There is no change which could affect Bluetooth and could address this issue (in theory the stable Linux kernel bump could, but we did not expect that to change things). So it is not surprising that HAOS 10.1 does not change the behavior.
> 
> I've not been able too reproduce it so far, my Raspberry Pi's happily report sensor values from my BTHome enabled devices.
> 
> Is there a known simple setup (with one/few Bluetooth devices) which is known to trigger this problem?
> 
> @spiek81 what kind of Xiaomi sensors do you have? The square Xiaomi Mijia? I have those but with the BTHome firmware. Do you use the original firmware on those?

No, i have the old round one's. After downgrading to 9,5 2 days  ago no issues has ben occurred en everything is working fine now.

SrBurner, 2023-04-28:
> No, i have the old round one's. After downgrading to 9,5 2 days ago no issues has been occurred en everything is working fine now.
@spiek81 Are you running OS 10.1 right now, or just staying in 9.5? Just to know.

theelog, 2023-04-28:
I spoke too soon. Bluetooth integration is now broken in it's entirety.

```Config entry '[mac-redacted]' for bluetooth integration not ready yet: Bluetooth adapter None with address [mac-redacted] not found; Retrying in background```

gcarrarom, 2023-04-28:
Reporting back: RaspberryPi4 B + Avantree DG40S USB Bluetooth Adapter working after upgrading to 10.1. Will report back in this comment if it stops working.

mariusnauj, 2023-04-29:
I Have INTEL NUC with HAOS. After update to HAOS to 10v, my xiaomi bluetooth devices was broken. After update to HAOS 10.1v bluetooth broken too. It runs fine after I downgraded HAOS to v9.5.

quemart, 2023-04-29:
Same Problem:
My INKBIRD IBS_TH temperature/humidity sensors have all failed after recent updates.

Dell Wyse 5070 Thin Client
Home Assistant 2023.4.6
Supervisor 2023.04.1
Operating System 10.1
Frontend 20230411.1 - latest

Plausible0Bug, 2023-04-29:
Same, Raspberry Pi 4 CM + Home assistant Yellow:
[image]
[image]
```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.4.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "bluetooth",
    "name": "Bluetooth",
    "after_dependencies": [
      "hassio"
    ],
    "codeowners": [
      "@bdraco"
    ],
    "config_flow": true,
    "dependencies": [
      "usb"
    ],
    "documentation": "https://www.home-assistant.io/integrations/bluetooth",
    "iot_class": "local_push",
    "loggers": [
      "btsocket",
      "bleak_retry_connector",
      "bluetooth_adapters",
      "bluetooth_auto_recovery"
    ],
    "quality_scale": "internal",
    "requirements": [
      "bleak==0.20.1",
      "bleak-retry-connector==3.0.2",
      "bluetooth-adapters==0.15.3",
      "bluetooth-auto-recovery==1.0.3",
      "bluetooth-data-tools==0.3.1",
      "dbus-fast==1.84.2"
    ],
    "is_built_in": true
  },
  "data": {
    "manager": {
      "adapters": {
        "hci0": {
          "address": "E4:5F:01:DF:00:00",
          "sw_version": "",
          "hw_version": "usb:v1D6Bp0246d0542",
          "passive_scan": true,
          "manufacturer": "Raspberry Pi Trading Ltd",
          "product": null,
          "vendor_id": null,
          "product_id": null
        }
      },
      "slot_manager": {
        "manager": true,
        "adapter_slots": {
          "hci0": 5
        },
        "allocations_by_adapter": {
          "hci0": []
        }
      },
      "scanners": [
        {
          "name": "hci0 (E4:5F:01:DF:00:00)",
          "start_time": 95.064099602,
          "source": "E4:5F:01:DF:00:00",
          "scanning": false,
          "type": "HaScanner",
          "last_detection": 76578.779684603,
          "monotonic_time": 142357.682742187,
          "discovered_devices_and_advertisement_data": [],
          "adapter": "hci0"
        }
      ],
      "connectable_history": [],
      "all_history": [],
      "advertisement_tracker": {
        "intervals": {},
        "sources": {},
        "timings": {}
      }
    },
    "adapters": {
      "hci0": {
        "address": "E4:5F:01:DF:00:00",
        "sw_version": "",
        "hw_version": "usb:v1D6Bp0246d0542",
        "passive_scan": true,
        "manufacturer": "Raspberry Pi Trading Ltd",
        "product": null,
        "vendor_id": null,
        "product_id": null
      }
    },
    "dbus": {
      "/org/bluez": {
        "org.freedesktop.DBus.Introspectable": {},
        "org.bluez.AgentManager1": {},
        "org.bluez.ProfileManager1": {}
      },
      "/org/bluez/hci0": {
        "org.freedesktop.DBus.Introspectable": {},
        "org.bluez.Adapter1": {
          "Address": "E4:5F:01:DF:00:00",
          "AddressType": "public",
          "Name": "",
          "Alias": "",
          "Class": 2883584,
          "Powered": true,
          "PowerState": "on",
          "Discoverable": false,
          "DiscoverableTimeout": 180,
          "Pairable": false,
          "PairableTimeout": 0,
          "Discovering": false,
          "UUIDs": [
            "0000110e-0000-1000-8000-00805f9b34fb",
            "00001200-0000-1000-8000-00805f9b34fb",
            "0000110a-0000-1000-8000-00805f9b34fb",
            "0000110b-0000-1000-8000-00805f9b34fb",
            "00001108-0000-1000-8000-00805f9b34fb",
            "0000110c-0000-1000-8000-00805f9b34fb",
            "00001800-0000-1000-8000-00805f9b34fb",
            "00001801-0000-1000-8000-00805f9b34fb",
            "00001844-0000-1000-8000-00805f9b34fb",
            "0000180a-0000-1000-8000-00805f9b34fb",
            "00001112-0000-1000-8000-00805f9b34fb"
          ],
          "Modalias": "usb:v1D6Bp0246d0542",
          "Roles": [
            "central",
            "peripheral"
          ]
        },
        "org.freedesktop.DBus.Properties": {},
        "org.bluez.BatteryProviderManager1": {},
        "org.bluez.GattManager1": {},
        "org.bluez.AdvertisementMonitorManager1": {
          "SupportedMonitorTypes": [
            "or_patterns"
          ],
          "SupportedFeatures": []
        },
        "org.bluez.Media1": {
          "SupportedUUIDs": [
            "0000110a-0000-1000-8000-00805f9b34fb",
            "0000110b-0000-1000-8000-00805f9b34fb"
          ]
        },
        "org.bluez.NetworkServer1": {},
        "org.bluez.LEAdvertisingManager1": {
          "ActiveInstances": 0,
          "SupportedInstances": 5,
          "SupportedIncludes": [
            "tx-power",
            "appearance",
            "local-name"
          ],
          "SupportedFeatures": [],
          "SupportedCapabilities": {
            "MaxAdvLen": 31,
            "MaxScnRspLen": 31
          }
        }
      }
    }
  }
}
```
Reboot the device(complete hard reset instead of restart home assistant system) temporarily restore Bluetooth functionality then crash after a day or two.

gertk64, 2023-05-01:
Hear hear, HA 10.0 on RPi3 and (onboard) bluethooth cuts out after some time. Trying to revert to 9.5 which worked flawless for months.

agners, 2023-05-02:
@theelog 

> I spoke too soon. Bluetooth integration is now broken in it's entirety.

In that situation, did you see a different adapter discovered or just none?

[91 more replies]

closed 2023-06-22 by agners