home-assistant/operating-system #2486: Odroid N2+ could not connect to RTL-SDR device after upgrade to HA OS 10

SenMorgan, 2023-04-19:
### Describe the issue you are experiencing

Since the upgrade to HA OS 10, the addon "ADS-B Multi-Portal Feeder" has been unable to connect to the RTL-SDR receiver, even after unplugging and plugging it back in, as well as rebooting the system. There were no issues with earlier versions of HA OS.

Addon log output (sensitive data hide under «************»):

`s6-rc: info: service adsbexchange-mlat: starting
s6-rc: info: service adsbexchange-stats successfully started
s6-rc: info: service adsbexchange-mlat successfully started
[fr24feed] 2023-04-19 11:48:23 | [feed][i]configuring decoder
[fr24feed] 2023-04-19 11:48:23 | [feed][c]Max range AIR: 350.0nm
[fr24feed] 2023-04-19 11:48:23 | [feed][c]Max range GND: 100.0nm
[fr24feed] 2023-04-19 11:48:23 | [feed][i]configuration changed
[fr24feed] 2023-04-19 11:48:23 | [feed][i]defined 3 servers
[fr24feed] 2023-04-19 11:48:23 | [feed][c]Timestamps: optional
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.822 [E][:] Local time: 2023-04-19 11:48:23 +0200
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.822 [E][:] GMT+0 time: 2023-04-19 10:48:23 +0200
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.822 [E][:] Your machine should be set as GMT+0 time zone!
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.830 [W][:] Time zone is not set to GMT+0
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.830 [I][crxstats.cpp:588] [stats]Stats thread started
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.838 [E][:] Local time: 2023-04-19 11:48:23 +0200
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.838 [E][:] GMT+0 time: 2023-04-19 10:48:23 +0200
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.838 [E][:] Your machine should be set as GMT+0 time zone!
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.838 [W][:] Time zone is not set to GMT+0
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.838 [I][receiver_ac_sender.cpp:137] Stopping ReceiverACSender threads for feed 
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.838 [D][receiver_ac_sender.cpp:141] Stop called on non-running network thread for feed 
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.838 [I][receiver_ac_sender.cpp:96] Configured ReceiverACSender: 185.218.24.22:8099,185.218.24.23:8099,185.218.24.24:8099, feed: LKTB205, send_interval: 5s/1s, max age: 15s, send metadata: true, mode: 1, filtering: true
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.846 [E][:] Local time: 2023-04-19 11:48:23 +0200
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.846 [E][:] GMT+0 time: 2023-04-19 10:48:23 +0200
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.846 [E][:] Your machine should be set as GMT+0 time zone!
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.846 [W][:] Time zone is not set to GMT+0
[fr24feed] 2023-04-19 11:48:23 | 23-04-19 11:48:23.846 [I][receiver_ac_sender.cpp:36] Network thread connecting to 185.218.24.22:8099 for feed LKTB205
[fr24feed] 2023-04-19 11:48:23 | [feed][n]LKTB205@185.218.24.22:8099/UDP
[fr24feed] 2023-04-19 11:48:23 | [feed][n]connecting
[fr24feed] 2023-04-19 11:48:23 | [feed][n]connected via UDP (fd 23)
[fr24feed] 2023-04-19 11:48:23 | [feed][n]working
[dump1090] Wed Apr 19 11:48:23 2023 CEST  dump1090-fa v8.2 starting up.
[dump1090] rtlsdr: no supported devices found.
 System: Home Assistant OS 10.0  (aarch64 / odroid-n2)
[adsbexchange-feed] invoked by: /usr/local/share/adsbexchange/readsb --net-only --debug=n --quiet --net-beast-reduce-interval 0.5 --net-connector feed1.adsbexchange.com,30004,beast_reduce_out,feed2.adsbexchange.com,64004 --write-json /run/adsbexchange-feed --write-json-every 1 --write-prom /run/adsbexchange-feed/stats.prom --net-heartbeat 60 --net-ro-size 1280 --net-ro-interval 0.2 --net-bi-port 30154 --net-connector 127.0.0.1,30005,beast_in --lat ************ --lon ************
[adsbexchange-feed] Wed Apr 19 11:48:24 2023 CEST  readsb starting up.
[adsbexchange-feed] readsb version: 3.14.1595 compiled on 230207
[adsbexchange-feed] Using lat:   ************, lon:  ************
[adsbexchange-feed] 30154: Beast TCP input port           
[adsbexchange-feed] BeastReduce TCP output: Attempting connection to feed1.adsbexchange.com port 30004 ...
[adsbexchange-feed] Beast TCP input: Attempting connection to 127.0.0.1 port 30005 ...
[adsbexchange-feed] Beast TCP input: Connection to 127.0.0.1 port 30005 failed: 111 (Connection refused)
[adsbexchange-stats] Found local resolver in resolv.conf, disabling DNS Cache
[adsbexchange-stats] host command or perl not available, disabling DNS Cache
[adsbexchange-stats] Using UUID [************] for stats uploads
[adsbexchange-stats] Using JSON directory [/run/adsbexchange-feed] for source data
[adsbexchange-stats] NOT using script's DNS cache
 Home Assistant Core: 2023.4.5
[adsbexchange-feed] BeastReduce TCP output: Connection established: feed1.adsbexchange.com (52.24.158.239) port 30004
 Home Assistant Supervisor: 2023.04.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------`

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install ADS-B Multi-Portal Feeder addon
2. Connect corresponding RTL-SDR receiver
3. Configure addon as required per default configuration
4. Upgrade to HA OS  10.0
5. Start addon and see the logs
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

### System information

## System Information

version | core-2023.4.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.24
arch | aarch64
timezone | Europe/Prague
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.31.0
Stage | running
Available Repositories | 1278
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 113.9 GB
disk_used | 65.2 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | chrony (2.5.0), File editor (5.5.0), Home Assistant Google Drive Backup (0.108.2), Samba share (10.0.0), RPC Shutdown (2.2), Zigbee2MQTT (1.30.3-1), Let's Encrypt (4.12.8), NGINX Home Assistant SSL proxy (3.2.0), MariaDB (2.5.2), Mosquitto broker (6.2.0), Terminal & SSH (9.6.1), ADS-B Multi-Portal Feeder (1.20.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 8
views | 15
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 9, 2023 at 07:17
-- | --
current_recorder_run | April 19, 2023 at 11:01
estimated_db_size | 1694.05 MiB
database_engine | mysql
database_version | 10.6.10

</details>

SenMorgan, 2023-04-19:
I forgot to mention that the Odroid N2+ is the main board of my **Home Assistant Blue**, on which I am running HA OS.

agners, 2023-04-19:
Can you check the kernel logs? You can read the full log since the last boot using the SSH/Web terminal:

```
ha host logs --identifier kernel -b 0 -n 10000
```

SenMorgan, 2023-04-20:
Unfortunately, neither of those arguments after `ha host logs` are accepted.

```text
[core-ssh ~]$ ha host logs --identifier kernel -b 0 -n 10000
Error: unknown flag: --identifier
Usage:
  ha host logs [flags]

Aliases:
  logs, log, lg

Examples:

  ha host logs

Flags:
  -h, --help   help for logs

Global Flags:
      --api-token string   Home Assistant Supervisor API token
      --config string      Optional config file (default is $HOME/.homeassistant.yaml)
      --endpoint string    Endpoint for Home Assistant Supervisor (default is 'supervisor')
      --log-level string   Log level (defaults to Warn)
      --no-progress        Disable the progress spinner
      --raw-json           Output raw JSON from the API

FATA[0000] Error while executing rootCmd: unknown flag: --identifier
```

SenMorgan, 2023-04-20:
I disconnected and reconnected the device, and here are the corresponding logs:

```text
Apr 20 05:45:44 homeassistantblue kernel: usb 1-1.3: USB disconnect, device number 17
Apr 20 05:45:51 homeassistantblue kernel: usb 1-1.3: new high-speed USB device number 19 using xhci-hcd
```

SenMorgan, 2023-04-20:
I have discovered that the "ADS-B Multi-Portal Feeder" can detect the RTL-SDR receiver, but unfortunately, it fails to start, which was not a problem before upgrading to HA OS 10. Here is the complete log that I tried to copy and paste from the UI logs window:

```text
s6-rc: info: service dump1090: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service dump1090 successfully started
s6-rc: info: service netcat: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service init-upintheair: starting
s6-rc: info: service confd-piaware: starting
s6-rc: info: service confd-layers: starting
s6-rc: info: service confd-html: starting
s6-rc: info: service confd-fr24feed: starting
s6-rc: info: service conf-planefinder: starting
s6-rc: info: service conf-adsbexchange: starting
s6-rc: info: service banner: starting
s6-rc: info: service netcat successfully started
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
-----------------------------------------------------------
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[280]: INFO Backend set to env
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[280]: INFO Starting confd
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[280]: INFO Backend source(s) set to 
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[280]: INFO Target config /usr/lib/fr24/public_html/layers.js out of sync
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[280]: INFO Target config /usr/lib/fr24/public_html/layers.js has been updated
s6-rc: info: service confd-layers successfully started
[dump1090] Thu Apr 20 07:53:13 2023 CEST  dump1090-fa v8.2 starting up.
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[338]: INFO Backend set to env
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[338]: INFO Starting confd
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[338]: INFO Backend source(s) set to 
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[338]: INFO Target config /usr/lib/fr24/public_html/config.js out of sync
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[338]: INFO Target config /usr/lib/fr24/public_html/config.js has been updated
s6-rc: info: service confd-html successfully started
[dump1090] rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
s6-rc: info: service conf-planefinder successfully started
s6-rc: info: service planefinder: starting
s6-rc: info: service planefinder successfully started
s6-rc: info: service planefinder-log: starting
s6-rc: info: service planefinder-log successfully started
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[387]: INFO Backend set to env
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[387]: INFO Starting confd
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[387]: INFO Backend source(s) set to 
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[387]: INFO Target config /etc/fr24feed.ini out of sync
2023-04-20T07:53:13+02:00 f1c878cb-adsb-multi-portal-feeder /opt/confd/bin/confd[387]: INFO Target config /etc/fr24feed.ini has been updated
s6-rc: info: service confd-fr24feed successfully started
s6-rc: info: service fr24feed: starting
s6-rc: info: service fr24feed successfully started
s6-rc: info: service confd-piaware successfully started
s6-rc: info: service piaware: starting
s6-rc: info: service piaware successfully started
s6-rc: info: service init-upintheair successfully started
s6-rc: info: service http: starting
s6-rc: info: service http successfully started
[dump1090] Found Rafael Micro R820T tuner
 Add-on: ADS-B Multi-Portal Feeder
[dump1090] rtlsdr: tuner gain set to 49.6 dB (gain step 28)
[dump1090] Allocating 4 zero-copy buffers
 Dump1090 based feeder for FlightRadar24, FlightAware and more
-----------------------------------------------------------
 Add-on version: 1.20.0
s6-rc: info: service conf-adsbexchange successfully started
s6-rc: info: service adsbexchange-feed: starting
s6-rc: info: service adsbexchange-feed successfully started
s6-rc: info: service adsbexchange-stats: starting
s6-rc: info: service adsbexchange-mlat: starting
s6-rc: info: service adsbexchange-stats successfully started
s6-rc: info: service adsbexchange-mlat successfully started
 You are running the latest version of this add-on.
[... 109 lines cut ...]
s6-rc: info: service piaware successfully stopped
s6-rc: info: service confd-piaware: stopping
s6-rc: info: service netcat successfully stopped
s6-rc: info: service adsbexchange-mlat successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service init-upintheair successfully stopped
s6-rc: info: service adsbexchange-stats successfully stopped
s6-rc: info: service adsbexchange-feed: stopping
s6-rc: info: service confd-layers successfully stopped
s6-rc: info: service fr24feed successfully stopped
s6-rc: info: service confd-fr24feed: stopping
s6-rc: info: service confd-html successfully stopped
[... cut ...]
s6-rc: info: service conf-planefinder: stopping
s6-rc: info: service confd-fr24feed successfully stopped
s6-rc: info: service conf-planefinder successfully stopped
s6-rc: info: service adsbexchange-feed successfully stopped
s6-rc: info: service conf-adsbexchange: stopping
s6-rc: info: service dump1090: stopping
s6-rc: info: service conf-adsbexchange successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service dump1090 successfully stopped
```

SenMorgan, 2023-04-20:
Thanks for the tip on downgrading. The SDR is working again!
Also, the logs look different when I unplug and plug in the device with HA OS 9.5:

<details><summary>LOG</summary>

```text
Apr 20 10:45:50 homeassistantblue kernel: usb 1-1: clear tt 4 (9052) error -71
Apr 20 10:45:50 homeassistantblue kernel: usb 1-1.3: USB disconnect, device number 7
Apr 20 10:45:54 homeassistantblue systemd[1]: docker-0f44e275049e6b41b2ba107b245095f1d36cf76efb556323e99def3469364197.scope: Deactivated successfully.
Apr 20 10:45:54 homeassistantblue systemd[1]: docker-0f44e275049e6b41b2ba107b245095f1d36cf76efb556323e99def3469364197.scope: Consumed 37.370s CPU time.
Apr 20 10:45:54 homeassistantblue kernel: kauditd_printk_skb: 70 callbacks suppressed
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1334 audit(1681987554.169:401): prog-id=97 op=UNLOAD
Apr 20 10:45:54 homeassistantblue dockerd[1067]: time="2023-04-20T10:45:54.195056843Z" level=info msg="shim disconnected" id=0f44e275049e6b41b2ba107b245095f1d36cf76efb556323e99def3469364197
Apr 20 10:45:54 homeassistantblue dockerd[1059]: time="2023-04-20T10:45:54.195143453Z" level=info msg="ignoring event" container=0f44e275049e6b41b2ba107b245095f1d36cf76efb556323e99def3469364197 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Apr 20 10:45:54 homeassistantblue dockerd[1067]: time="2023-04-20T10:45:54.197302523Z" level=warning msg="cleaning up after shim disconnected" id=0f44e275049e6b41b2ba107b245095f1d36cf76efb556323e99def3469364197 namespace=moby
Apr 20 10:45:54 homeassistantblue dockerd[1067]: time="2023-04-20T10:45:54.197376683Z" level=info msg="cleaning up dead shim"
Apr 20 10:45:54 homeassistantblue dockerd[1067]: time="2023-04-20T10:45:54.209969638Z" level=warning msg="cleanup warnings time=\"2023-04-20T10:45:54Z\" level=info msg=\"starting signal loop\" namespace=moby pid=9211 runtime=io.containerd.runc.v2\n"
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1325 audit(1681987554.213:402): table=nat:114 family=2 entries=1 op=nft_unregister_rule pid=9224 subj=unconfined comm="iptables"
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1300 audit(1681987554.213:402): arch=c00000b7 syscall=211 success=yes exit=472 a0=3 a1=ffffd37e8168 a2=0 a3=ffff98ad6760 items=0 ppid=1059 pid=9224 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1327 audit(1681987554.213:402): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4400444F434B4552002D7000746370002D6400302F30002D2D64706F72740038373534002D6A00444E4154002D2D746F2D64657374696E6174696F6E003137322E33302E33332E383A383735340000002D690068617373696F
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1325 audit(1681987554.221:403): table=filter:115 family=2 entries=1 op=nft_unregister_rule pid=9226 subj=unconfined comm="iptables"
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1300 audit(1681987554.221:403): arch=c00000b7 syscall=211 success=yes exit=612 a0=3 a1=ffffcce8ba78 a2=0 a3=ffffab090760 items=0 ppid=1059 pid=9226 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1327 audit(1681987554.221:403): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4400444F434B45520000002D690068617373696F002D6F0068617373696F002D7000746370002D64003137322E33302E33332E38002D2D64706F72740038373534002D6A00414343455054
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1325 audit(1681987554.257:404): table=nat:116 family=2 entries=1 op=nft_unregister_rule pid=9228 subj=unconfined comm="iptables"
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1300 audit(1681987554.257:404): arch=c00000b7 syscall=211 success=yes exit=564 a0=3 a1=fffff00dba98 a2=0 a3=ffff989b0760 items=0 ppid=1059 pid=9228 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
Apr 20 10:45:54 homeassistantblue kernel: audit: type=1327 audit(1681987554.257:404): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4400504F5354524F5554494E47002D7000746370002D73003137322E33302E33332E38002D64003137322E33302E33332E38002D2D64706F72740038373534002D6A004D415351554552414445
Apr 20 10:45:54 homeassistantblue dockerd[1059]: time="2023-04-20T10:45:54.384589080Z" level=warning msg="Failed to delete conntrack state for 172.30.33.8: invalid argument"
Apr 20 10:45:54 homeassistantblue kernel: hassio: port 14(veth7c3250b) entered disabled state
Apr 20 10:45:54 homeassistantblue kernel: vethc25ee68: renamed from eth0
Apr 20 10:45:54 homeassistantblue systemd-udevd[9236]: Using default interface naming scheme 'v250'.
Apr 20 10:45:54 homeassistantblue NetworkManager[401]: <info>  [1681987554.4528] manager: (vethc25ee68): new Veth device (/org/freedesktop/NetworkManager/Devices/38)
Apr 20 10:45:54 homeassistantblue kernel: hassio: port 14(veth7c3250b) entered disabled state
Apr 20 10:45:54 homeassistantblue kernel: device veth7c3250b left promiscuous mode
Apr 20 10:45:54 homeassistantblue kernel: hassio: port 14(veth7c3250b) entered disabled state
[... cut ...]
Apr 20 10:45:55 homeassistantblue kernel: eth0: renamed from veth39a794d
Apr 20 10:45:55 homeassistantblue kernel: IPv6: ADDRCONF(NETDEV_CHANGE): vethd7a2fe2: link becomes ready
Apr 20 10:45:55 homeassistantblue NetworkManager[401]: <info>  [1681987555.2296] device (vethd7a2fe2): carrier: link connected
Apr 20 10:45:55 homeassistantblue kernel: usb 1-1.3: new high-speed USB device number 8 using xhci-hcd
Apr 20 10:45:56 homeassistantblue kernel: usb 1-1.3: device descriptor read/64, error -71
```

agners, 2023-04-20:
Hm, that is interesting, it seems that HAOS 9.5 actually reports problems:

```
Apr 20 10:45:50 homeassistantblue kernel: usb 1-1.3: USB disconnect, device number 7
...
Apr 20 10:45:55 homeassistantblue kernel: usb 1-1.3: new high-speed USB device number 8 using xhci-hcd
Apr 20 10:45:56 homeassistantblue kernel: usb 1-1.3: device descriptor read/64, error -71
```

Whereas HAOS 10 doesn't report any:

```
Apr 20 05:45:44 homeassistantblue kernel: usb 1-1.3: USB disconnect, device number 17
Apr 20 05:45:51 homeassistantblue kernel: usb 1-1.3: new high-speed USB device number 19 using xhci-hcd
```

It could be related that things don't get accessed in HAOS 10 :thinking:

SenMorgan, 2023-04-21:
The issue has been resolved in version 1.20.3 of the Home Assistant add-on ADS-B Multi Portal Feeder. https://github.com/MaxWinterstein/homeassistant-addons

closed 2023-04-21 by SenMorgan