home-assistant/operating-system #2487: Data migration not working - concludes but still runs from SD

Stooovie, 2023-04-19:
### Describe the issue you are experiencing

“Move data disk” concludes (I am able to select my 120GB SSD fine) --and nothing happens. Everything is still on the SD card. Storage is still displaying 80% full, same as before (it’s a 32GB SD card). The migration control panel continues to say "you are currently using SR32G-0x8506d35c" (the SD card).

Home Assistant 2023.4.5, Supervisor 2023.04.0, Operating System 10.0, Frontend 20230411.1 - latest running on Rpi4.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. migrate data to USB SSD
2. after the procedure (taking roughly 10 minutes) it continues to run from SD 

### Anything in the Supervisor logs that might be useful for us?

```txt
23-04-19 15:36:54 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:36:54 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:36:54 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:36:54 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:42:00 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-04-19 15:45:16 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-04-19 15:45:17 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-04-19 15:45:17 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-04-19 15:46:31 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-04-19 15:47:06 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2MQTT (45df7312_zigbee2mqtt)
[... 425 lines cut ...]
ID_REVISION: SBFM61.3
ID_SERIAL: Patriot_Burst_E0CD0799032200919475
ID_SERIAL_SHORT: E0CD0799032200919475
ID_TYPE: disk
ID_USB_DRIVER: usb-storage
ID_USB_INSTANCE: '0:0'
ID_USB_INTERFACES: ':080650:080662:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: '2115'
ID_USB_MODEL_ENC: 2115\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20
ID_USB_MODEL_ID: 55aa
ID_USB_REVISION: '0'
ID_USB_SERIAL: ASMT_2115_1234567923CE-0:0
ID_USB_SERIAL_SHORT: 1234567923CE
ID_USB_TYPE: disk
ID_USB_VENDOR: ASMT
ID_USB_VENDOR_ENC: ASMT\x20\x20\x20\x20
ID_USB_VENDOR_ID: 174c
MAJOR: '8'
MINOR: '1'
PARTN: '1'
PARTNAME: hassos-data-external
SUBSYSTEM: block
TAGS: ':systemd:'
USEC_INITIALIZED: '4924616'

agners, 2023-04-20:
Did the system reboot after you selected the disk?

Can you check the logs from the terminal?

```
ha host logs -b -1 -n 10000
```

Stooovie, 2023-04-20:
It did reboot, yes. The files are there, I checked by connecting the SSD to a Linux system.

I cannot run the command:
`[core-ssh ~]$ ha host logs -b -1 -n 10000
Error: unknown shorthand flag: 'b' in -b
Usage:
  ha host logs [flags]

Aliases:
  logs, log, lg

Examples:

  ha host logs

Flags:
  -h, --help   help for logs

Global Flags:
      --api-token string   Home Assistant Supervisor API token
      --config string      Optional config file (default is $HOME/.homeassistant.yaml)
      --endpoint string    Endpoint for Home Assistant Supervisor (default is 'supervisor')
      --log-level string   Log level (defaults to Warn)
      --no-progress        Disable the progress spinner
      --raw-json           Output raw JSON from the API

FATA[0000] Error while executing rootCmd: unknown shorthand flag: 'b' in -b `

Stooovie, 2023-04-20:
Pastebin and Hastebin refuse to work, so here's the result log:

https://transfer.sh/M3Qh35/log.txt

Thanks!

agners, 2023-04-20:
@Stooovie it seems that this is not the boot where a move attempt has been made:

```
Dec 20 18:15:46 homeassistant systemd[1]: Home Assistant OS data partition migration was skipped because of an unmet condition check (ConditionPathExists=/mnt/overlay/move-data).
```

Can you check different boots (-2/-3) or attempt to use the data disk again and then check again?

If "Home Assistant OS data partition migration" is never being run, than that means that the Supervisor doesn't even trigger a data disk move. In that cases though the system should not restart and there should be an error message (or at least something in the Supervisor logs).

Stooovie, 2023-04-20:
You're right. 

So, I wiped the SSD clean, reinserted, rebooted the Pi, started the migration (SSD is correctly detected), picked the correct drive in the migration dialogue, it rebooted after a couple of minutes, then it took couple more minutes to come back up... and it's still running off the SD card.

This is a fresh log taken right after HA came back up. No message, no notification:

https://transfer.sh/pbh8Qy/migration.log

This is from System/Hardware/All hardware:

```
DEVLINKS: >-
  /dev/disk/by-id/ata-Patriot_Burst_E0CD0799032200919475-part1
  /dev/disk/by-id/usb-ASMT_2115_1234567923CE-0:0-part1
  /dev/disk/by-label/hassos-data /dev/disk/by-partlabel/hassos-data-external
  /dev/disk/by-partuuid/4ade87e0-9c2f-4588-b18b-605fce0d4b99
  /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1:1.0-scsi-0:0:0:0-part1
  /dev/disk/by-uuid/34eda77e-f37e-4ae6-a2df-049cb858f8e3
DEVNAME: /dev/sda1
```

I see the "hassos-data" label so the data has been carried over. This is the SSD, manufactured by Patriot. Alas, the system still runs off SD which is "SR32G-0x8506d35c'" (Kingston 32GB micro SD card).

This is a fourth attempt, same result.

agners, 2023-04-20:
This could be a problem:
```
hwmon hwmon1: Undervoltage detected!
```

Maybe writing to the SSD needs more power, which then causes the data disk move command to abort.

Stooovie, 2023-04-21:
With powered hub, HA doesn't even come back. [Log](https://transfer.sh/wkwg9q/migration.log)

bcutter, 2023-06-24:
So you sorted this out by...? (I read in another issue you (?) left RPi4 platform and went ahead with a Proxmox setup, right

Stooovie, 2023-06-25:
> So you sorted this out by...? _(I read in another issue you (?) left RPi4 platform and went ahead with a Proxmox setup, right)_

Hi, I didn't solve it. It may be my SATA-USB cable but I don't think so as I did have this very setup working in the past with older HA OS. It was however a final nail in the coffin. The second-hand i5 fanless PC draws about the same amount of energy as the Pi, and with a NVMe drive, it's incomparably more stable. I can recommend running HA this way, definitely.

closed 2023-10-01 by github-actions[bot]