home-assistant/operating-system #2488: After HA OS 10 Update no sound and configuration.yaml in config folder

Johnnii360, 2023-04-19:
### Describe the issue you are experiencing

Hi!

I updated to HA OS 10 and Core 2023.4.5 today. Now I can't use build in Stereo Jack anymore and when I connect via SSH to the docker image shell the config folder is empty. I think that's why there is no TTS and audio anymore. I think due to the empty config folder MPD can't work anymore and get the auth error because there are config parameters in the configuration.yaml.

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

10

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Using an ODROID N2+.
2. Upgrade to HA OS 10.
3. Try to use TTS or MPD or just look into the config folder via SSH.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
File "/usr/local/lib/python3.10/site-packages/sentry_sdk/_compat.py", line 60, in reraise
    raise value
  File "/usr/local/lib/python3.10/site-packages/sentry_sdk/integrations/aiohttp.py", line 116, in sentry_app_handle
    response = await old_handle(self, request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 185, in block_bad_requests
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 201, in system_validation
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 268, in token_validation
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 280, in core_proxy
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/utils.py", line 62, in wrap_api
    answer = await method(api, *args, **kwargs)
  File "/usr/src/supervisor/supervisor/api/store.py", line 178, in store_info
    ATTR_ADDONS: [
  File "/usr/src/supervisor/supervisor/api/store.py", line 179, in <listcomp>
    self._generate_addon_information(self.sys_addons.store[addon])
  File "/usr/src/supervisor/supervisor/api/store.py", line 114, in _generate_addon_information
    ATTR_ADVANCED: addon.advanced,
  File "/usr/src/supervisor/supervisor/addons/model.py", line 227, in advanced
    return self.data[ATTR_ADVANCED]
  File "/usr/src/supervisor/supervisor/store/addon.py", line 19, in data
    return self.sys_store.data.addons[self.slug]
KeyError: '243ffc37_pixelflut'
23-04-19 18:13:26 INFO (MainThread) [supervisor.store] Loading add-ons from store: 98 all - 0 new - 0 remove
23-04-19 18:13:26 INFO (MainThread) [supervisor.store] Loading add-ons from store: 98 all - 0 new - 0 remove
23-04-19 18:13:58 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
23-04-19 18:13:58 INFO (MainThread) [supervisor.homeassistant.core] Successfully started Home Assistant 2023.4.5
23-04-19 18:13:58 INFO (SyncWorker_3) [supervisor.docker.interface] Cleanup images: ['ghcr.io/home-assistant/odroid-n2-homeassistant:2023.4.4']
23-04-19 18:30:57 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-04-19 18:36:34 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
23-04-19 18:36:34 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
23-04-19 18:36:34 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_ssh
23-04-19 18:36:34 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
23-04-19 18:37:25 INFO (MainThread) [supervisor.api.middleware.security] /audio/info access from core_ssh
23-04-19 18:40:06 INFO (MainThread) [supervisor.backups.manager] Creating new partial backup with slug 8d1d6e2d
23-04-19 18:40:06 INFO (MainThread) [supervisor.backups.manager] Backing up 8d1d6e2d store Add-ons
[... 254 lines cut ...]
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/alsa-mixer.sh
[18:00:15] INFO: Adjust ALSA mixer settings for /dev/snd/controlC0
[18:00:16] INFO: Adjust ALSA mixer settings for /dev/snd/controlC1
cont-init: info: /etc/cont-init.d/alsa-mixer.sh exited 0
cont-init: info: running /etc/cont-init.d/filesystem.sh
cont-init: info: /etc/cont-init.d/filesystem.sh exited 0
cont-init: info: running /etc/cont-init.d/pulse-config.sh
cont-init: info: /etc/cont-init.d/pulse-config.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun alsa (no readiness notification)
services-up: info: copying legacy longrun pulseaudio (no readiness notification)
s6-rc: info: service legacy-services successfully started
W: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set.
W: [pulseaudio] main.c: Home directory of user 'root' is not '/var/run/pulse', ignoring.
W: [pulseaudio] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.
W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please make sure that you actually do want to do that.
W: [pulseaudio] main.c: Please read http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/ for an explanation why system mode is usually a bad idea.
W: [pulseaudio] module-udev-detect.c: Failed to open /proc/asound/card1: No such file or directory
W: [pulseaudio] module-udev-detect.c: Failed to open /proc/asound/card0: No such file or directory
E: [pulseaudio] module-rescue-streams.c: module-rescue-stream is obsolete and should no longer be loaded. Please remove it from your configuration.
E: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.DBus.Error.NameHasNoOwner: Could not activate remote peer: unit failed.
```

agners, 2023-04-19:
Duplicate of #2484.

agners, 2023-04-19:
Not sure about the missing `configuration.yaml`. Can you still access Home Assistant? Then configuration.yaml should be still there.

Johnnii360, 2023-04-20:
> Not sure about the missing `configuration.yaml`. Can you still access Home Assistant? Then configuration.yaml should be still there.

Access is still possible.

But:

`[core-ssh ~]$ cd config
[core-ssh config]$ ls
[core-ssh config]$ ls -la
total 0
[core-ssh config]$ 
`

agners, 2023-04-20:
Audio issue should be resolved with the new Audio plug-in release. If audio is still not working, make sure it is up-to-date:

```
ha audio update
```

A restart might be required.

closed 2023-04-20 by agners