home-assistant/operating-system #2489: Home Assistant OS is not booting

sim-san, 2023-04-19:
### Describe the issue you are experiencing

I can't not boot to the Home Assistant OS after a fresh SSD flashing.

Error message: Booting Slot A (OK=0, TRY=0)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Download Home Assistent OS x86 Image file
2. Flash a new SSD with the image
3. Install the SSD in an Intel NUC7
4. Disable Secure Boot and enable UEFI in the BIOS
5. Start the NUC connected to an Screen with HDMI
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs available
```

### Anything in the Host logs that might be useful for us?

```txt
No logs available
```

### System information

No system information available 

### Additional information

[image]

agners, 2023-04-20:
It looks like the bootloader crashes very late, or the Linux kernel very early. Can you try HAOS 9.5 to see if that works on the same hardware?

https://github.com/home-assistant/operating-system/releases/tag/9.5

sim-san, 2023-04-20:
Same behavior.

But I solved the issue. I have to update my bios to be able to boot Home Assistant OS.

closed 2023-04-20 by sim-san