home-assistant/operating-system #2490: Since upgrading to HAOS 10.0, Gmail notify having Problems SMTP

gfn256, 2023-04-20:
### Describe the issue you are experiencing

Since upgrading yesterday from 9.5 to 10.0, I'm experiencing Gmail notify errors where about half of my Email notifications are not sent, while throwing (SMTP timing?) errors, (logs attached).

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.   Upgrade from 9.5 to 10.0
2.   Have Automations send a Gmail Notify service.

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
Logger: homeassistant.components.smtp.notify
Source: components/smtp/notify.py:217
Integration: smtp (documentation, issues)
First occurred: 7:30:48 AM (2 occurrences)
Last logged: 7:30:58 AM

SMTPServerDisconnected sending mail: retrying connection

Log Details (ERROR)
Logger: homeassistant.core
Source: components/smtp/notify.py:230
First occurred: 7:30:48 AM (2 occurrences)
Last logged: 7:30:58 AM

Error executing service: <ServiceCall notify.gmail [REDACTED:..........xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx..........]
Error executing service: <ServiceCall notify.gmail [REDACTED:..........xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx..........]

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/smtplib.py", line 398, in getreply
    line = self.file.readline(_MAXLINE + 1)
  File "/usr/local/lib/python3.10/socket.py", line 705, in readinto
    return self._sock.recv_into(b)
  File "/usr/local/lib/python3.10/ssl.py", line 1274, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/local/lib/python3.10/ssl.py", line 1130, in read
    return self._sslobj.read(len, buffer)
TimeoutError: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/smtp/notify.py", line 224, in _send_email
    mail.sendmail(self._sender, recipients, msg.as_string())
  File "/usr/local/lib/python3.10/smtplib.py", line 902, in sendmail
    (code, resp) = self.data(msg)
  File "/usr/local/lib/python3.10/smtplib.py", line 567, in data
    (code, repl) = self.getreply()
  File "/usr/local/lib/python3.10/smtplib.py", line 401, in getreply
    raise SMTPServerDisconnected("Connection unexpectedly closed: "
smtplib.SMTPServerDisconnected: Connection unexpectedly closed: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1838, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1857, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 264, in _async_notify_message_service
    await self.async_send_message(**kwargs)
  File "/usr/src/homeassistant/homeassistant/components/notify/legacy.py", line 240, in async_send_message
    await self.hass.async_add_executor_job(
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/smtp/notify.py", line 217, in send_message
    return self._send_email(msg, recipients)
  File "/usr/src/homeassistant/homeassistant/components/smtp/notify.py", line 230, in _send_email
    mail.quit()
  File "/usr/local/lib/python3.10/smtplib.py", line 1004, in quit
    res = self.docmd("quit")
  File "/usr/local/lib/python3.10/smtplib.py", line 431, in docmd
    self.putcmd(cmd, args)
  File "/usr/local/lib/python3.10/smtplib.py", line 378, in putcmd
    self.send(f'{s}{CRLF}')
  File "/usr/local/lib/python3.10/smtplib.py", line 365, in send
    raise SMTPServerDisconnected('please run connect() first')
smtplib.SMTPServerDisconnected: please run connect() first
```

### System information

## System Information

version | core-2023.4.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.24
arch | x86_64
timezone | Asia/Jerusalem
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | August 11, 2023 at 3:00 AM
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-8.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 30.8 GB
disk_used | 8.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.2.0), File editor (5.5.0), Home Assistant Google Drive Backup (0.110.3), RPC Shutdown (2.2), Samba share (10.0.0), Terminal & SSH (9.6.2), eWeLink Smart Home (1.4.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 10, 2023 at 6:47 AM
-- | --
current_recorder_run | April 19, 2023 at 4:52 PM
estimated_db_size | 157.98 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

Something to note: If I run the automation by pressing run, I have not seen any error. Peculiar?

agners, 2023-04-20:
This looks more like some type of TCP/IP connectivity issue.

Can you downgrade to 9.5 again to see if this indeed fixes the issue?

```
ha os update --version 9.5
```

gfn256, 2023-04-20:
In trying to establish how often the Gmail Notify has errors, I have been running an automation every so often to send a Gmail. So far I've had no errors with 4/4 being sent! When the doctor comes - illness disappears!
I'm going to report back if I have further errors.
I'm hesitant at this stage to revert to 9.5, until I can know something more concrete about current error.

agners, 2023-04-20:
Yeah probably a Google hiccup with some bad timing :sweat_smile: 

Thanks for the update! I am closing the issue then.

gfn256, 2023-04-20:
Just to convey my experience; immediately on the installation update of 10.0 when the machine was self-shutingdown  I didn't receive my Gmail Shutdown Notification (my automation) although I did subsequentially receive the Gmail Startup Notification (my automation) on startup seconds later. After that, my Gmail Notifications were hit and miss (about 50%), until I the doctor showed up!