home-assistant/operating-system #2494: hassos 10.0 causes reboot loop UNLESS a screen is plugged in

MidasLamb, 2023-04-20:
### Describe the issue you are experiencing

I've been running home assistant on an intel nuc, everything was working fine until I updated to hassos 10.0.

To debug it I moved it to a different room where I plugged in keyboard, mouse and a screen through the HDMI port.
It then started perfectly fine.
Thinking it might have been a hiccup I moved it back to its original place (without any keyboard, mouse or screen attached), and immediately it started rebooting constantly.

After some further trial and error I've determined that having the screen connected through HDMI will make it so that it boots properly and stays on, while having no screen connected it will startup, but shortly after having started everything it will shut itself down again and reboot, starting the cycle over and over again.

Even unplugging the screen after it's been running stable for multiple hours will cause it to shutdown a bit later and start the bootcycle.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

From what I could tell, I upgraded from the latest 9 version, and have no screen connected to the intel nuc.

### Anything in the Supervisor logs that might be useful for us?

```txt
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/a17fefae/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/752f0981/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/ce37752d/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/15d03859/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/504b6591/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/faeeab8d/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/dd25d3f9/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/40b654ec/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/77543c1b/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/2628ef5f/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/ada886d8/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/40ff27c2/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/73e02a78/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/b2fdc36e/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/3cdf5052/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/585bb753/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/c1b6f216/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/605955c8/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/1ca11e33/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/b6b1fb75/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/64abf074/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/5c080bf6/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/48331cc7/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/b885777c/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/bac6da99/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/29c7c2e9/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/b7d128bb/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/2ff30552/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/6e87948c/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/4649221b/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/8d7a9aea/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/41eee31b/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/6baa6d84/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/3de28090/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/8d617c64/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/046db368/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/bc051cb0/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/efcabbb6/info access from cebe7a76_hassio_google_drive_backup
23-04-20 20:18:35 INFO (MainThread) [supervisor.api.middleware.security] /backups/6309824a/info access from cebe7a76_hassio_google_drive_backup
[... 225 lines cut ...]
version_api | ok
installed_addons | Cloudflared (4.1.2), ESPHome (2023.3.2), Mosquitto broker (6.2.0), Studio Code Server (5.5.6), Matter Server (4.3.1), SQLite Web (3.7.1), chrony (2.5.0), Home Assistant Google Drive Backup (0.110.3), Terminal & SSH (9.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 6
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 13, 2023 at 4:06 PM
-- | --
current_recorder_run | April 20, 2023 at 8:18 PM
estimated_db_size | 804.76 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2023-04-25:
> After some further trial and error I've determined that having the screen connected through HDMI will make it so that it boots properly and stays on, while having no screen connected it will startup (and home assistant is briefly available), but shortly after having started everything it will shut itself down again and reboot, starting the cycle over and over again.

If a the bootup doesn't complete, the system should roll-back after 3 attempts. I assume that is not happening then? (you let it go through more than three reboots?)

Can you get the logs of the last reboot after attaching the screen?

```
ha host logs -b -1 -n 10000
```

(you can redirect the output to the config directory using `ha host logs -b -1 -n 10000 > /config/last-boot.log`).

MidasLamb, 2023-04-25:
I've downgraded to 9.5, once I have some time I'll upgrade again and get these logs.
I think the bootups do complete entirely, home assistant seems to be available through the web interface and accessible (nothing loading more in the bottom left) for like 10-30 seconds before it starts to shutdown. So that's why it probably doesn't do the rollback. It was rebooting for multiple hours before I noticed it so it definitely did more than 3 in that timeframe.

Klumpke, 2023-04-27:
I have exactly the same problem, also with the recently released 10.1 version.
Running on a Intel NUC 8i3BEH.

Klumpke, 2023-08-07:
Issue still exists. The workaround that @Thorry84 suggested does not work on my Intel NUC 8i3BEH.

agners, 2023-11-07:
@Klumpke the logs don't look like they are from one of the crashes. It seems that the crash happens so early, that nothing is stored in the log yet (before user-space/systemd gets started).

Is this still reproducible for you with HAOS 11.1?

Thorry84, 2023-11-07:
I've been able to reproduce it on HAOS 11.1. It seems like it's really early in the boot process, maybe in the bootloader?

I can't figure out what the issue is. Sometimes plugging in a screen helps, sometimes also plugging in a keyboard does the trick. Sometimes changing BIOS settings like Fast Boot helps, sometimes it doesn't. Sometimes removing all USB devices helps.

It for sure started with 10.0, before that no issues at all with booting.

closed 2024-02-15 by github-actions[bot]